nyanmisaka/ffmpeg-rockchip #107: Support for RK3568?

danergo, 2024-08-31:
Hello!

I know that this project's aim is to use hardware acceleration on RK3588.

But in my PCB, I have a Rockchip RK3568B2 CPU.

I cloned this repo, and built both mpp and rga libs successfully (based on your example), but then during ffmpeg compilation I'm getting many errors regarding rga.

Is it possible that my CPU is not capable of using this library?

nyanmisaka, 2024-09-02:
rk356x is also supported. But it seems that the librga you compiled is outdated, please purge the existing librga header files and recompile according to the Wiki.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation

closed 2024-09-02 by nyanmisaka

danergo, 2024-09-02:
Thanks for confirming.

What is the best way to purge old rga? Isn't a new build and install supposed to overwrite everything?

Thank you!

nyanmisaka, 2024-09-02:
Generally speaking there is no best way. Any path in `/usr/*/include/*` may contain header files, and their installation path is determined at build time. It is best to uninstall rga-related packages directly, or build in a clean environment, such as docker.

danergo, 2024-09-02:
Hi, again, I have succeeded to build this ffmpeg (I have removed all old rga folders from /usr directory).

So, thank you!

Now, however I keep running into the below problem, which I couldn't really understand. If you could have some time to look on it, I'd appreciate it:

````
$ sudo ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i DJI_0181.MP4 -c:a copy -strict -2 -vf scale_rkrga=w=1920:h=1080:format=nv12:afbc=1 -c:v h264_rkmpp -rc_mode VBR -b:v 6M -maxrate 6M -bufsize 12M -profile:v high -g:v 120 -y DJI_0181_resized.mp4
ffmpeg version 8164ff7 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5598968aa0] stream 0, timescale not set
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'DJI_0181.MP4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2mp41
    creation_time   : 2023-04-08T14:53:45.000000Z
    encoder         : DJIAction2
  Duration: 00:09:26.74, start: 0.000000, bitrate: 53480 kb/s
  Stream #0:0[0x1](und): Video: hevc (Main) (hvc1 / 0x31637668), yuv420p(tv, bt709), 3840x2160, 51156 kb/s, 23.98 fps, 23.98 tbr, 24k tbn (default)
    Metadata:
      creation_time   : 2023-04-08T14:53:45.000000Z
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
    Side data:
      displaymatrix: rotation of -180.00 degrees
  Stream #0:1[0x2](und): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 317 kb/s (default)
    Metadata:
      creation_time   : 2023-04-08T14:53:45.000000Z
      handler_name    : SoundHandler
      vendor_id       : [0][0][0][0]
  Stream #0:2[0x3](und): Data: none (djmd / 0x646D6A64), 9 kb/s
    Metadata:
      creation_time   : 2023-04-08T14:53:45.000000Z
      handler_name    : DJI meta
  Stream #0:3[0x4](und): Data: none (dbgi / 0x69676264), 1976 kb/s
    Metadata:
      creation_time   : 2023-04-08T14:53:45.000000Z
      handler_name    : DJI dbgi
  Stream #0:4[0x0]: Video: mjpeg (Baseline), yuvj420p(pc, bt470bg/unknown/unknown), 1280x720 [SAR 1:1 DAR 16:9], 90k tbr, 90k tbn (attached pic)
rga_api version 1.3.1_[11] (1e2f0db build: 2023-04-20 13:33:00 base: )
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> h264 (h264_rkmpp))
  Stream #0:1 -> #0:1 (copy)
Press [q] to stop, [?] for help
[Parsed_scale_rkrga_0 @ 0x55989ce310] Output format 'nv12' with AFBC modifier is not supported by RGA2
 RgaBlit(1356) RGA_BLIT fail: Invalid argument RgaBlit(1357) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[18, (nil), (nil), (nil), 2560]
rect[0, 0, 3840, 2160, 3840, 2160, 2560, 0]
f-blend-size-rotation-col-log-mmu[2560, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[31, (nil), (nil), (nil), 2560]
rect[0, 0, 1920, 1080, 1920, 1080, 2560, 0]
f-blend-size-rotation-col-log-mmu[2560, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
[Parsed_scale_rkrga_0 @ 0x55989ce310] RGA blit failed: -22
[vf#0:0 @ 0x55989d0900] Error while filtering: Generic error in an external library
Failed to inject frame into filter network: Generic error in an external library
Error while filtering: Generic error in an external library
[out#0/mp4 @ 0x559897e8b0] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=00:00:01.15 bitrate=   0.0kbits/s speed=38.3x
Conversion failed!
```

Note: I tried removing the "format" from scale_rkrga option, but the result was the same as above.

Do you have any suggestion by any chance on what would worth checking?

Thank you very much!

nyanmisaka, 2024-09-03:
Outdated librga. Recompile according to the instructions in the wiki and upgrade the kernel to 5.10 or later.

danergo, 2024-09-03:
Perfect, thank you very much! 4K->1080 (24fps) transcoded at almost 2x speed! I love it :)

One more simple question though: although I modified all required resources to be "root:video" owner, and my user is a member of "video" group, I can't run it without sudo:

```
[hevc_rkmpp @ 0x559d754d60] Failed to init MPP context: -1
[vist#0:0/hevc @ 0x559d7508d0] Error while opening decoder: Generic error in an external library
[vost#0:0/h264_rkmpp @ 0x559d7703f0] Error initializing a simple filtergraph
Error opening output file DJI_0181_resized2.mp4.
Error opening output files: Generic error in an external library
```

Should I change the owner of the required resource files (ex. /dev/dri) to "video:video", or "user:user" to be able to use this without sudo?