nyanmisaka/ffmpeg-rockchip #108: Is MPP hardware encoding being utilized for dual H.264 1080p streams on rk3399pro?

2457205703, 2024-09-11:
Hi, I have a question about using MPP hardware encoding on my armboard with RK3399Pro SoC.

I've followed the wiki instructions to make and install the project. Then I tested streaming from  CSI cameras to a ZLMediaKit([https://github.com/ZLMediaKit/ZLMediaKit)](https://github.com/ZLMediaKit/ZLMediaKit) streaming server running locally.

To adapt to my development board's official initial image, I modified the configuration command in the final step of compiling ffmpeg-rockchip:
```shell
./configure --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
make -j $(nproc)
```
to
```shell
sudo ./configure --prefix=/usr --extra-cflags="-I/usr/include/rga -I/usr/include/rockchip -I/usr/include/libdrm" --extra-ldflags="-L/usr/lib  -L. -lrga" --arch=arm64  --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-shared --enable-pic
```

My streaming commands and outputs are:
```shell
linaro@linaro-alip:~$ ffmpeg -f v4l2 -video_size 1920x1080 -i /dev/video5 -c:v h264_rkmpp -b:v 1M -maxrate 1.5M -bufsize 2M -g 60 -r 30 -f rtsp rtsp://127.0.0.1/live/test
librga:RGA_GET_VERSION:3.02,3.020000
ctx=0x55b80bd2d0,ctx->rgaFd=3
Rga built version:7bf9abf
ffmpeg version 6.1 Copyright (c) 2000-2023 the FFmpeg developers
built with gcc 8 (Debian 8.3.0-6)
configuration: --prefix=/usr --extra-cflags='-I/usr/include/rga -I/usr/include/rockchip -I/usr/include/libdrm' --extra-ldflags='-L/usr/lib -L. -lrga' --arch=arm64 --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-shared --enable-pic
libavutil 58. 29.100 / 58. 29.100
libavcodec 60. 31.102 / 60. 31.102
libavformat 60. 16.100 / 60. 16.100
libavdevice 60. 3.100 / 60. 3.100
libavfilter 9. 12.100 / 9. 12.100
libswscale 7. 5.100 / 7. 5.100
libswresample 4. 12.100 / 4. 12.100
libpostproc 57. 3.100 / 57. 3.100
[video4linux2,v4l2 @ 0x55b80bdbc0] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x55b80bdbc0] ioctl(VIDIOC_G_PARM): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x55b80bdbc0] Time per frame unknown
[video4linux2,v4l2 @ 0x55b80bdbc0] Stream #0: not enough frames to estimate rate; consider increasing probesize
Input #0, video4linux2,v4l2, from '/dev/video5':
Duration: N/A, start: 221.796818, bitrate: N/A
Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 1920x1080, 1000k tbr, 1000k tbn
Stream mapping:
Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
Output #0, rtsp, to 'rtsp://127.0.0.1/live/test':
Metadata:
encoder : Lavf60.16.100
Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, q=2-31, 1000 kb/s, 30 fps, 90k tbn
Metadata:
encoder : Lavc60.31.102 h264_rkmpp
frame= 1672 fps= 30 q=-0.0 size=N/A time=00:00:55.70 bitrate=N/A dup=3 drop=1 speed= 1x
```

This seems to be streaming in real-time already. When I use VLC player on my own Windows PC, I can play the video normally, albeit with a slight delay.

Then I tried streaming from my second CSI camera. Here are the commands and outputs from my two terminals:

```shell
linaro@linaro-alip:~$ ffmpeg -f v4l2 -video_size 1920x1080 -i /dev/video5 -c:v h264_rkmpp -b:v 1M -maxrate 1.5M -bufsize 2M -g 60 -r 30 -f rtsp rtsp://127.0.0.1/live/test
librga:RGA_GET_VERSION:3.02,3.020000
ctx=0x55a69ac2d0,ctx->rgaFd=3
Rga built version:7bf9abf
ffmpeg version 6.1 Copyright (c) 2000-2023 the FFmpeg developers
built with gcc 8 (Debian 8.3.0-6)
configuration: --prefix=/usr --extra-cflags='-I/usr/include/rga -I/usr/include/rockchip -I/usr/include/libdrm' --extra-ldflags='-L/usr/lib -L. -lrga' --arch=arm64 --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-shared --enable-pic
libavutil 58. 29.100 / 58. 29.100
libavcodec 60. 31.102 / 60. 31.102
libavformat 60. 16.100 / 60. 16.100
libavdevice 60. 3.100 / 60. 3.100
libavfilter 9. 12.100 / 9. 12.100
libswscale 7. 5.100 / 7. 5.100
libswresample 4. 12.100 / 4. 12.100
libpostproc 57. 3.100 / 57. 3.100
[video4linux2,v4l2 @ 0x55a69acbc0] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x55a69acbc0] ioctl(VIDIOC_G_PARM): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x55a69acbc0] Time per frame unknown
[video4linux2,v4l2 @ 0x55a69acbc0] Stream #0: not enough frames to estimate rate; consider increasing probesize
Input #0, video4linux2,v4l2, from '/dev/video5':
Duration: N/A, start: 602.406418, bitrate: N/A
Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 1920x1080, 1000k tbr, 1000k tbn
Stream mapping:
Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
Output #0, rtsp, to 'rtsp://127.0.0.1/live/test':
Metadata:
encoder : Lavf60.16.100
Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, q=2-31, 1000 kb/s, 30 fps, 90k tbn
Metadata:
encoder : Lavc60.31.102 h264_rkmpp
[vost#0:0/h264_rkmpp @ 0x55a69b21e0] More than 1000 frames duplicated991 drop=0 speed=0.805x
frame= 2076 fps= 24 q=-0.0 size=N/A time=00:01:09.20 bitrate=N/A dup=1289 drop=0 speed=0.798x
```
```shell
linaro@linaro-alip:~$ ffmpeg -f v4l2 -video_size 1920x1080 -i /dev/video0 -c:v h264_rkmpp -b:v 1M -maxrate 1.5M -bufsize 2M -g 60 -r 30 -f rtsp rtsp://127.0.0.1/live/test1
librga:RGA_GET_VERSION:3.02,3.020000
ctx=0x559713a2d0,ctx->rgaFd=3
Rga built version:7bf9abf
ffmpeg version 6.1 Copyright (c) 2000-2023 the FFmpeg developers
built with gcc 8 (Debian 8.3.0-6)
configuration: --prefix=/usr --extra-cflags='-I/usr/include/rga -I/usr/include/rockchip -I/usr/include/libdrm' --extra-ldflags='-L/usr/lib -L. -lrga' --arch=arm64 --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-shared --enable-pic
libavutil 58. 29.100 / 58. 29.100
libavcodec 60. 31.102 / 60. 31.102
libavformat 60. 16.100 / 60. 16.100
libavdevice 60. 3.100 / 60. 3.100
libavfilter 9. 12.100 / 9. 12.100
libswscale 7. 5.100 / 7. 5.100
libswresample 4. 12.100 / 4. 12.100
libpostproc 57. 3.100 / 57. 3.100
[video4linux2,v4l2 @ 0x559713abc0] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x559713abc0] ioctl(VIDIOC_G_PARM): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x559713abc0] Time per frame unknown
[video4linux2,v4l2 @ 0x559713abc0] Stream #0: not enough frames to estimate rate; consider increasing probesize
Input #0, video4linux2,v4l2, from '/dev/video0':
Duration: N/A, start: 613.401995, bitrate: N/A
Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 1920x1080, 1000k tbr, 1000k tbn
Stream mapping:
Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
Output #0, rtsp, to 'rtsp://127.0.0.1/live/test1':
Metadata:
encoder : Lavf60.16.100
Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, q=2-31, 1000 kb/s, 30 fps, 90k tbn
Metadata:
encoder : Lavc60.31.102 h264_rkmpp
[vost#0:0/h264_rkmpp @ 0x55971401e0] More than 1000 frames duplicated996 drop=0 speed=0.767x
frame= 2122 fps= 23 q=-0.0 size=N/A time=00:01:10.70 bitrate=N/A dup=1651 drop=0 speed=0.767x
```

After I started playing the streams with VLC player, the delay and stuttering became quite significant, and it seemed that the processing was not happening in real-time. Subsequently, I used the 'top' command to check the resource usage:
```shell
top - 15:04:13 up 13 min, 10 users, load average: 3.34, 2.70, 1.56
Tasks: 247 total, 3 running, 244 sleeping, 0 stopped, 0 zombie
%Cpu(s): 11.2 us, 8.7 sy, 0.0 ni, 79.8 id, 0.0 wa, 0.0 hi, 0.2 si, 0.0 st
MiB Mem : 1960.4 total, 972.8 free, 519.3 used, 468.4 buff/cache
MiB Swap: 2048.0 total, 2048.0 free, 0.0 used. 1281.5 avail Mem

PID USER PR NI VIRT RES SHR S %CPU %MEM TIME+ COMMAND
23866 linaro 20 0 501400 127916 113744 S 15.1 6.4 0:40.29 ffmpeg
23348 linaro 20 0 501400 130624 113548 S 14.5 6.5 0:51.10 ffmpeg
512 root 20 0 379680 8368 6168 S 11.5 0.4 0:27.15 rkisp_3A_server
514 root 20 0 314144 8440 6080 S 11.5 0.4 1:06.75 rkisp_3A_server
1697 root 20 0 1059960 21872 13064 S 9.9 1.1 0:51.29 MediaServer
2164 linaro 20 0 13900 5340 3944 S 2.3 0.3 0:12.15 sshd
1506 linaro 20 0 13908 5292 3896 S 2.0 0.3 0:12.84 sshd
1758 linaro 20 0 13908 5288 3892 S 2.0 0.3 0:13.13 sshd
3455 linaro 20 0 13900 5404 4008 S 2.0 0.3 0:12.58 sshd
3582 linaro 20 0 7000 3864 2320 S 1.6 0.2 0:07.54 top
1 root 20 0 165380 9500 7028 R 1.3 0.5 0:10.65 systemd
1813 linaro 20 0 6228 3072 2348 S 1.3 0.2 0:06.00 top
438 message+ 20 0 7380 4236 3088 S 1.0 0.2 0:06.45 dbus-daemon
2232 linaro 20 0 6228 2972 2244 S 1.0 0.1 0:05.48 top
```
The CPU usage doesn't seem particularly high.

So, my question is: According to the official documentation, the VPU should be capable of simultaneously encoding 2 streams of H.264 1080p/30fps, along with additional encoding/decoding tasks such as 6 streams of 1080P/30FPS H264/H265, or 2 streams of 4K/30FPS (supporting HDR 10bits), or other equivalent computational combinations.

Is it possible that my streaming setup is not utilizing the VPU's MPP library for encoding?

nyanmisaka, 2024-09-11:
> According to the official documentation, the VPU should be capable of simultaneously encoding 2 streams of H.264 1080p/30fps, along with additional encoding/decoding tasks such as 6 streams of [...]

[image]

https://rockchip.fr/RK3399Pro%20datasheet%20V1.1.pdf

The H.264 encoder of RK3399Pro can only encode a **1080p@30fps** in real time, accroding to the datasheet from Rockchip.

2457205703, 2024-09-11:
So, I have already been using the hardware encoder, correct? Is it that the VPU component of the RK3399Pro can only support the encoding task for one camera at 1080p/30fps?

nyanmisaka, 2024-09-11:
> So, I have already been using the hardware encoder, correct? Is it that the VPU component of the RK3399Pro can only support the encoding task for one camera at 1080p/30fps?

Yes. It can only do only 1x 1080p@30fps encode. Any more and the encoder will be overloaded.

> Does the performance of the VPU component vary across different chip models?

Yes. The performance of VPU depends on the SoC model. For example, RK3566/3568 can do 1080p@60fps, RK3576 can do 4k@60fps, and RK3588 can do 8k@30fps. You can calculate the equivalent performance of the latter two at 1080p by yourself. These data can be found on the Rockchip official website, or you can search "RK3588 datasheet" on Google/Bing to get more hardware details.

2457205703, 2024-09-11:
Thank you very much for your prompt reply. It has been a great help to me.

2457205703, 2024-09-11:
I have one more question: Would it be possible for me to perform a decoding task simultaneously with encoding? For example, could I do 1080p/30fps encoding while also decoding a 320x240/30fps stream?

nyanmisaka, 2024-09-11:
> I have one more question: Would it be possible for me to perform a decoding task simultaneously with encoding? For example, could I do 1080p/30fps encoding while also decoding a 320x240/30fps stream?

Yes. The decoder and encoder are two separate fixed-function modules, but they share memory bandwidth.
You'd better take a look at the Wiki page. https://github.com/nyanmisaka/ffmpeg-rockchip/wiki

closed 2024-09-12 by nyanmisaka