nyanmisaka/ffmpeg-rockchip #109: [BUG] Initializing texture for hardware decoding failed

defencedog, 2024-09-18:
My `video.mp4` is an HEVC 10bit. When I pay it via `mpv --vo=gpu --hwdec=auto video.mp4` it shows a blue screen (if it is not 10bit the H265 video plays properly) 
> [vo/gpu] Initializing texture for hardware decoding failed.

However I experimented with `ffplay -vcodec hevc_rkmpp video.mp4` it playes properly! My SBC is OPi3b v2.1 RK3566. 

```
$ffmpeg -decoders | grep rkmpp

 V..... av1_rkmpp            Rockchip MPP (Media Process Platform) AV1 decoder (codec av1)
 V..... h263_rkmpp           Rockchip MPP (Media Process Platform) H263 decoder (codec h263)
 V..... h264_rkmpp           Rockchip MPP (Media Process Platform) H264 decoder (codec h264)
 V..... hevc_rkmpp           Rockchip MPP (Media Process Platform) HEVC decoder (codec hevc)
 V..... mpeg1_rkmpp          Rockchip MPP (Media Process Platform) MPEG1VIDEO decoder (codec mpeg1video)
 V..... mpeg2_rkmpp          Rockchip MPP (Media Process Platform) MPEG2VIDEO decoder (codec mpeg2video)
 V..... mpeg4_rkmpp          Rockchip MPP (Media Process Platform) MPEG4 decoder (codec mpeg4)
 V..... vp8_rkmpp            Rockchip MPP (Media Process Platform) VP8 decoder (codec vp8)
 V..... vp9_rkmpp            Rockchip MPP (Media Process Platform) VP9 decoder (codec vp9)
```

Any opinions on this?

nyanmisaka, 2024-09-18:
https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Rendering

The Wiki already explains what's needed.

defencedog, 2024-09-18:
Extremely thanks `--vf=scale_rkrga=force_yuv=8bit` this option removed this issue

closed 2024-09-18 by defencedog