nyanmisaka/ffmpeg-rockchip #112: ffmpeg PIPE is hung up when encoding 1080P with h264_rkmpp

sxfman, 2024-09-22:
cmd_ffmpeg = [
            'ffmpeg_core_' + self.camType,
            '-hwaccel', 'rkmpp',
            '-f', 'mjpeg',
            '-r', '30',
            '-i', '-',
            '-c:v', 'h264_rkmpp',
            '-flags', '+low_delay',
            '-video_size', '1920x1080',
            '-r', '30',
            '-vbsf', 'h264_mp4toannexb',
            '-b:v', '2M',
            '-profile:v', "high",
            '-level', '4.1',
            '-g:v', '30',
            '-f', 'h264',
            '-'
        ]
pipe_vd = subprocess.Popen(cmd_ffmpeg, stdin=PIPE, stdout=PIPE, bufsize=1024 * 20, shell=False)
while True:
    ......
    _jpg = .........
    pipe_vd.stdin.write(_jpg)
    pipe_vd.stdin.flush()
    _h264 = pipe_vd.stdout.read()
    pipe_vd.stdout.flush()
    ......
    
I use a ffmpeg PIPE to encode MJPEG streaming to h264 streaming, it works well with 720p and lower resolutions, but if i change the src MJPEG resolution to 1080p('-qp_init' set to be bigger than 5), it doesn't work, and hung up at the point of PIPE input(the statement of : pipe_vd.stdin.write(_jpg)).

I have tried some, and found some clues:
    1. if the upstream MJPEG streaming resolution was set to 720p or lower, the PIPE would work well; 
    2. if the upstream MJPEG streaming encoding param of '-qp_init' was set below 5(very poor quality and very little of bitrate), even though the MJPEG resolution was 1080p, the PIPE would work well;
    3. if I use "libx264" instead of "h264_rkmpp" in the ffmpeg PIPE, the PIPE would work well, no matter what the upstream MJPEG resolution was selected(1080p with '-qp_init' 80 was also ok); 
    4. if I change the param '-bufsize' of ffmpeg encoding PIPE to a very little value, 1080p encoding still not ok;
    
I looked up for the PIPE size of linux, and found that the max PIPE size is about 64KB, no matter what the PIPI buffersize is configured. So the most fact could be: 
    h264_rkmpp encoder differs from x264 encoder, and it need more input/output buffer. When starting encode h264 stream of 1080p, 64KB of PIPE buffer exhausts, but still no encoded h264 frame is output to PIPE.stdout, so the PIPE buffer can not be released by PIPE.stdout.read(), so the pipe.stdin.write() is hung up.

nyanmisaka, 2024-09-22:
I can't reproduce it in the most common linux shell pipes.
```
./ffmpeg -v quiet -f lavfi -i testsrc2=s=1920x1080:r=30,format=nv12 \
-c:v h264_rkmpp -flags +low_delay -b:v 2M -profile:v high -g:v 30 -r 30 -f h264 - \
| ./ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -r 30 -i - -f null -
```

```
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_rkmpp) -> wrapped_avframe (native))
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf61.1.100
  Stream #0:0: Video: wrapped_avframe, drm_prime(tv, progressive), 1920x1080, q=2-31, 200 kb/s, 30 fps, 30 tbn
      Metadata:
        encoder         : Lavc61.3.100 wrapped_avframe
frame=97985 fps=203 q=-0.0 size=N/A time=00:54:26.16 bitrate=N/A speed=6.78x
```

sxfman, 2024-09-22:
yes, I have tried also with redirected cmd pipe, it really works!
ffmpeg -loglevel quiet -hwaccel rkmpp -f v4l2 -video_size 1920x1080 -framerate 15 -i /dev/video11 -f mjpeg -c:v mjpeg_rkmpp -flags +low_delay -qp_init 80 -r 15 - | ffmpeg -hwaccel rkmpp -f mjpeg -r 15 -i - -c:v h264_rkmpp -flags:v +global_header -video_size 1920x1080 -r 15 -vbsf h264_mp4toannexb -b:v 2M -profile:v high -level 4.1 -g:v 30 -f h264 - | ffmpeg -loglevel quiet -i - -c:v copy -segment_time 600 -reset_timestamps 1 -f segment _%03d.mp4

I am now quite confused, only have to check my codes, perhaps there is some bug in subprocess.PIPE pkg of python.

closed 2024-10-09 by nyanmisaka