nyanmisaka/ffmpeg-rockchip #113: 使用ffmpeg-rockchip对H264数据进行硬件解码失败

CETC-AIoT, 2024-09-23:
硬件平台：RK3568
内核：4.19.192/5.10.198
mpp版本：2023-12-14版本及最新版本

使用avcodec_send_packet方法（h264_rkmpp）进行解码，但是这个方法返回报错，先是返回-11，然后返回 -542398533（Generic error in an external library.）
而使用AV_CODEC_ID_H264进行软解则没有问题，请问一下是什么问题？是mpp的兼容性问题么？
另外使用软件的时候，有时解码之后的格式是NV12，有时候是YUV420P，请问这个是怎么设置的？

nyanmisaka, 2024-09-23:
The attachment link you provided is invalid.

CETC-AIoT, 2024-09-23:
您好！已经重新上传，麻烦帮忙看一下是什么问题，十分感谢！

nyanmisaka, 2024-09-23:
MPP can't tolerate this bitstream. There are too many errors in it. So it's not ffmpeg's fault.

`mpp[150250]: h264d_sei: seq_parameter_set_id 0 may be invalid`

```
[h264 @ 0xaaaad2d01310] Format h264 detected only with low score of 1, misdetection possible!
[h264 @ 0xaaaad2d020f0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xaaaad2d020f0] decode_slice_header error
[h264 @ 0xaaaad2d020f0] no frame!
[h264 @ 0xaaaad2d020f0] SEI type 1 size 180 truncated at 146
[h264 @ 0xaaaad2d020f0] non-existing PPS 0 referenced
[h264 @ 0xaaaad2d020f0] SEI type 1 size 180 truncated at 144
[h264 @ 0xaaaad2d020f0] non-existing PPS 0 referenced
[h264 @ 0xaaaad2d020f0] decode_slice_header error
[h264 @ 0xaaaad2d020f0] no frame!
...
```

nyanmisaka, 2024-09-24:
On the other hand, the fact that `mpi_dec_test` cannot decode most frames already points to the problem.

Unlike software decoders, hardware decoders are usually more strict about the bitstream.

closed 2024-09-24 by nyanmisaka

CETC-AIoT, 2024-09-25:
Thanks for your reply. 
I found a strange thing, I skipped the first 60 frames of data, and then I started decoding, and there was a high probability of errors, and after about 140 EAGAIN errors, I would report "Generic error in an extenal library", but I started decoding from the first frame, The probability of success is much higher, and once the data behind the success can be correctly decoded. 
In addition, I saved the data twice to a file, and then decoded it using mpi_dec_test. The result was also random, and not every frame was decoded correctly. 
GOP=30 in video stream, AUD information at the end of each frame - fixed 6 bytes: 0x00 0x00 0x00 0x01 0x09 0x10
The file is too big. I sent it to your email.
the error log:
[image]