nyanmisaka/ffmpeg-rockchip #115: Moonlight-qt 6.1.0 no hardware acceleration

woiza, 2024-10-05:
Hi,

I built fffmpeg-rockchip following your Wiki on my Rock 5B with Radxa's latest [Debian image](https://github.com/radxa-build/rock-5b/releases).

I can watch/decode h265 and other files:

```
rock@rock-5b:~/Downloads$ ffmpeg -stream_loop -1 -hwaccel rkmpp -i  jellyfish-110-mbps-hd-hevc.mkv -an -sn -vframes 5000 -f null -
ffmpeg version 5ddd3d4 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, matroska,webm, from 'jellyfish-110-mbps-hd-hevc.mkv':
  Metadata:
    COMPATIBLE_BRANDS: iso4hvc1iso6
    MAJOR_BRAND     : iso4
    MINOR_VERSION   : 1
    ENCODER         : Lavf56.3.100
  Duration: 00:00:30.10, start: 0.067000, bitrate: 109636 kb/s
  Stream #0:0(und): Video: hevc (Main), yuv420p(tv), 1920x1080 [SAR 1:1 DAR 16:9], 29.97 fps, 29.97 tbr, 1k tbn (default)
    Metadata:
      CREATION_TIME   : 2016-02-05 12:45:40
      LANGUAGE        : und
      HANDLER_NAME    : hevc@GPAC0.5.2-DEV-rev565-g71748d7-ab-suite
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    COMPATIBLE_BRANDS: iso4hvc1iso6
    MAJOR_BRAND     : iso4
    MINOR_VERSION   : 1
    encoder         : Lavf60.16.100
  Stream #0:0(und): Video: wrapped_avframe, nv12(tv, progressive), 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 200 kb/s, 29.97 fps, 29.97 tbn (default)
    Metadata:
      CREATION_TIME   : 2016-02-05 12:45:40
      LANGUAGE        : und
      HANDLER_NAME    : hevc@GPAC0.5.2-DEV-rev565-g71748d7-ab-suite
      encoder         : Lavc60.31.102 wrapped_avframe
[out#0/null @ 0xaaab1d284470] video:2344kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
frame= 5000 fps=215 q=-0.0 Lsize=N/A time=00:02:46.80 bitrate=N/A speed=7.18x
```

After that I built the latest moonlight-qt (6.1.0) following the instructions in their [Github readme ](https://github.com/moonlight-stream/moonlight-qt). Moonlight can't make use of hardware decoders, but Chrome, etc. can. I read in your Wiki that a patch has to be applied. However, it looks like this is not the case anymore [source](https://github.com/moonlight-stream/moonlight-qt/blob/master/app/streaming/video/ffmpeg-renderers/drm.cpp#L237).
Do you have an idea why? Should moonlight-qt be built differently with some other or additional flags/commands?

nyanmisaka, 2024-10-05:
I rarely use moonlight-qt, this section was added by @hbiyik. Maybe he has some ideas.

But before that, you can bisect the relevant commits in moonlight-qt yourself if it worked in an older version.

hbiyik, 2024-10-05:
@woiza i have just tested the git version of of moonlight-qt and it is working, you should see something like 
`FFMpeg: [h264_rkmpp @  .....]`
in the logs.

If hw acceleration is not working, which codec is moonlight using?

woiza, 2024-10-05:
@hbiyik moonlight uses h264 software. Here is my log. Do you mind providing all required steps?

```
00:00:00 - SDL Info (0): Compiled with SDL 2.26.5
00:00:00 - SDL Info (0): Running with SDL 2.26.5
00:00:00 - Qt Info: No translation available for "en_US"
00:00:00 - Qt Warning: qrc:/gui/main.qml:13:1: QML ApplicationWindow: ToolTip must be attached to an Item
00:00:00 - Qt Info: Found "gamecontrollerdb.txt" at "/home/rock/.cache/Moonlight Game Streaming Project/Moonlight/gamecontrollerdb.txt"
00:00:00 - SDL Info (0): Loaded 328 new gamepad mappings
00:00:00 - SDL Info (0): V-sync disabled
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/rockchip_drv_video.so
libva info: va_openDriver() returns -1
00:00:00 - SDL Info (0): Trying fallback VAAPI driver names
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/iHD_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'radeonsi'
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/radeonsi_drv_video.so
libva error: /usr/lib/aarch64-linux-gnu/dri/radeonsi_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
00:00:00 - SDL Error (0): Failed to initialize VAAPI: -1
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/rockchip_drv_video.so
libva info: va_openDriver() returns -1
00:00:00 - SDL Info (0): Trying fallback VAAPI driver names
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/iHD_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'radeonsi'
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/radeonsi_drv_video.so
libva error: /usr/lib/aarch64-linux-gnu/dri/radeonsi_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
00:00:00 - SDL Error (0): Failed to initialize VAAPI: -1
00:00:00 - SDL Info (0): Trying SdlRenderer for codec hevc_cuvid due to preferred pixel format: 0x9f
00:00:00 - SDL Info (0): Trying DrmRenderer for codec hevc_cuvid due to compatible pixel format: 0x9f
00:00:00 - SDL Info (0): Opened DRM render node: /dev/dri/renderD128
00:00:00 - SDL Info (0): GPU driver: rockchip
00:00:00 - SDL Info (0): Direct rendering via DRM is disabled
00:00:00 - SDL Info (0): DRM backend supports exporting EGLImage
00:00:00 - SDL Info (0): EGLRenderer: EGL doesn't support HDR rendering
00:00:00 - SDL Info (0): Opened DRM render node: /dev/dri/renderD128
00:00:00 - SDL Info (0): GPU driver: rockchip
00:00:00 - SDL Info (0): Direct rendering via DRM is disabled
00:00:00 - SDL Warn (0): No renderer can handle output from decoder: hevc_cuvid
00:00:00 - SDL Info (0): Trying SdlRenderer for codec hevc_cuvid due to preferred pixel format: 0x9f
00:00:00 - SDL Info (0): Trying DrmRenderer for codec hevc_cuvid due to compatible pixel format: 0x9f
00:00:00 - SDL Info (0): Opened DRM render node: /dev/dri/renderD128
00:00:00 - SDL Info (0): GPU driver: rockchip
00:00:00 - SDL Info (0): Direct rendering via DRM is disabled
00:00:00 - SDL Info (0): DRM backend supports exporting EGLImage
00:00:00 - SDL Info (0): EGLRenderer: EGL doesn't support HDR rendering
00:00:00 - SDL Info (0): Opened DRM render node: /dev/dri/renderD128
00:00:00 - SDL Info (0): GPU driver: rockchip
00:00:00 - SDL Info (0): Direct rendering via DRM is disabled
00:00:00 - SDL Warn (0): No renderer can handle output from decoder: hevc_cuvid
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/rockchip_drv_video.so
[... 503 lines cut ...]
00:00:01 - FFmpeg: [h264 @ 0xffff98ae1510] Reinit context to 1280x720, pix_fmt: yuv420p
00:00:01 - SDL Info (0): FFmpeg-based video decoder chosen
00:00:01 - Qt Info: Found "gamecontrollerdb.txt" at "/home/rock/.cache/Moonlight Game Streaming Project/Moonlight/gamecontrollerdb.txt"
00:00:01 - SDL Info (0): Loaded 328 new gamepad mappings
00:00:01 - Qt Debug: Current Moonlight version: "6.1.0"
00:00:01 - Qt Info: Gamepad mappings are up to date
00:00:01 - Qt Info: Discovered mDNS host: "microshit-pc.local."
00:00:01 - Qt Info: Resolved "microshit-pc.local." to QHostAddress("192.168.2.5")
00:00:01 - Qt Info: Resolved "microshit-pc.local." to QHostAddress("fd17:150a:cce5:f248:7818:5849:a39f:f116")
00:00:01 - Qt Info: Resolved "microshit-pc.local." to QHostAddress("fd17:150a:cce5:f248:4056:da38:9f6e:3bd2")
00:00:01 - Qt Info: Resolved "microshit-pc.local." to QHostAddress("fe80::80c8:f97d:1f83:e431")
00:00:01 - Qt Info: Resolved "microshit-pc.local." to QHostAddress("192.168.2.5")
00:00:01 - Qt Info: Resolved "microshit-pc.local." to QHostAddress("fd17:150a:cce5:f248:7818:5849:a39f:f116")
[... cut ...]
00:00:03 - Qt Info: Ignoring ULA: QHostAddress("fd17:150a:cce5:f248:7818:5849:a39f:f116")
00:00:03 - Qt Info: Ignoring ULA: QHostAddress("fd17:150a:cce5:f248:4056:da38:9f6e:3bd2")
00:00:03 - Qt Info: Processing new PC at "192.168.2.5:47989" from mDNS with IPv6 address "<NULL>"
00:00:03 - Qt Info: Executing request: "http://192.168.2.5:47989/serverinfo?uniqueid=0123456789ABCDEF&uuid=48a3813aac1941029b45eefda43b9d8b"
00:00:03 - Qt Info: Executing request: "https://192.168.2.5:47984/serverinfo?uniqueid=0123456789ABCDEF&uuid=2d4ecc7ccbb34e0b887c832c8e0a67dd"
```

hbiyik, 2024-10-05:
is it possible that youa re using the system ffmpeg rather than you built? You can manipulate your paths with LD_LIBRARY_PATH

> all required steps?

there is specific step, moonlight should traverse the codecs which provides DRM_PRIME output, and rkmpp codecs should be found.

woiza, 2024-10-05:
@hbiyik thank you for pointing me in the right direction!

closed 2024-10-05 by woiza