nyanmisaka/ffmpeg-rockchip #118: Why using rkmpp need to enable gpl version 3?

tongjiang, 2024-10-15:
MPP is a free project of rockchip, why add gpl license for rkmpp code?
This prevented me from using the project warehouse in our commercial project.

nyanmisaka, 2024-10-15:
MPP is licensed in [Apache-2.0](https://github.com/rockchip-linux/mpp/blob/develop/LICENSES/Apache-2.0), which is only compatible with GPL v3.

https://www.apache.org/licenses/GPL-compatibility.html

> Apache License v2.0 and GPL Compatibility
The [Free Software Foundation](http://www.fsf.org/) [considers](http://www.fsf.org/licensing/licenses/#apache2) the [Apache License, Version 2.0](https://www.apache.org/licenses/LICENSE-2.0) to be a [free software](http://www.gnu.org/philosophy/free-sw.html) license, [compatible](http://www.gnu.org/licenses/license-list.html#GPLCompatibleLicenses) with version 3 of the [GPL](http://www.fsf.org/licensing/licenses/gpl.html). The [Software Freedom Law Center](http://www.softwarefreedom.org/) provides [practical advice](http://www.softwarefreedom.org/resources/2007/gpl-non-gpl-collaboration.html) for developers about including permissively licensed source.

> Apache 2 software can therefore be included in GPLv3 projects, because the GPLv3 license accepts our software into GPLv3 works. However, GPLv3 software cannot be included in Apache projects. The licenses are incompatible in one direction only, and it is a result of ASF's licensing philosophy and the GPLv3 authors' interpretation of copyright law.

> Despite our best efforts, the FSF has never considered the Apache License to be compatible with GPL version 2, citing the patent termination and indemnification provisions as restrictions not present in the older GPL license. The Apache Software Foundation believes that you should always try to obey the constraints expressed by the copyright holder when redistributing their work.

tongjiang, 2024-10-15:
Why just using LGPL instead of GPLv3 in ffmpeg project when include rkmpp? cause the FSF doesn't allow this?

nyanmisaka, 2024-10-15:
> so can we just add ‘’--enable-version3" to enable rkmpp? Which only make the ffmpeg project using the LGPLv3.

Yes, `--enable-version3` should be enough for your use case.
Apache 2.0 licensed MPP can be included in LGPLv3 licenced FFmpeg.

https://github.com/nyanmisaka/ffmpeg-rockchip/blob/master/configure#L1827-L1828

closed 2024-10-15 by nyanmisaka