nyanmisaka/ffmpeg-rockchip #121: Stream Artefacts with Live rtmp

YannickPerret, 2024-10-21:
Hello,
With the rockchip version I have a problem with artefacts in Stream, I don't know if it's my fault on one of my configurations, but it looks like a cache or buffer error. I'm not sure if it's my fault on one of my configurations, but it looks like a cache or buffer error. As if the stream was keeping images and overlaying new content... Well, it's hard to explain, but I've filmed a capture of my stream doing VOD replays. The VODs are stored on a remote S3, so the videos don't have any basic artefact problems. 

[https://youtu.be/DFSKX3m9O5I](video with artefact)
[https://youtu.be/nd07oX10TCA](other video with artefact)

my rockchip installation : 
```bash
sudo apt install -y autoconf automake build-essential cmake git libtool pkg-config yasm nasm libx264-dev libx265-dev libvpx-dev libfdk-aac-dev libmp3lame-dev libopus-dev libass-dev libfreetype6-dev zlib1g-dev libdrm-dev libssl-dev

git clone -b jellyfin-mpp --depth=1 https://github.com/nyanmisaka/mpp.git rkmpp
cd rkmpp
mkdir rkmpp_build && cd rkmpp_build
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DBUILD_TEST=OFF ..
make -j$(nproc)
sudo make install

git clone -b jellyfin-rga --depth=1 https://github.com/nyanmisaka/rk-mirrors.git rkrga
meson setup rkrga rkrga_build --prefix=/usr --libdir=lib --buildtype=release --default-library=shared -Dcpp_args=-fpermissive -Dlibdrm=false -Dlibrga_demo=false
meson configure rkrga_build
ninja -C rkrga_build install

cd ~/dev
git clone --depth=1 https://github.com/nyanmisaka/ffmpeg-rockchip.git ffmpeg
cd ffmpeg
./configure --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-openssl --enable-libx264 --enable-libx265 --enable-libvpx --enable-libfdk-aac --enable-libmp3lame --enable-libopus --enable-libass --enable-libfreetype
make -j$(nproc)
sudo make install
```

and my final ffmpeg commande : 
```js
 '-re',
  '-hwaccel',
  'rkmpp',
  '-protocol_whitelist',
  'file,concat,http,https,tcp,tls,crypto',
  '-stream_loop',
  '-1',
  '-f',
  'concat',
  '-safe',
  '0',
  '-i',
  'concat:https://s3.***.cloud/object/v1/AUTH_e99c1****f00-bffa-42*****05f3.m3u8',
  '-r',
  '60',
  '-filter_complex',
  '[0:v]fps=fps=60[vout]',
  '-map',
  '[vout]',
  '-map',
  '0:a?',
  '-s',
  '1920x1080',
  '-c:a',
  'aac',
  '-c:v',
  'h264_rkmpp',
  '-b:v',
  '6000k',
  '-maxrate',
  '6000k',
  '-bufsize',
  '12000k',
  '-flags',
  'low_delay',
  '-f',
  'flv',
  'RTMP_KEY_LVE'
  ```
  Do you have any idea what I did wrong to get her into trouble? I doubt it's the storage on the S3, I'm with a known provider and the servers are located in the same country as me.
 Thanks in advances

nyanmisaka, 2024-10-28:
```
  '-stream_loop',
  '-1',
  '-f',
  'concat',
  '-safe',
  '0',
  '-i',
  'concat:https://s3.***.cloud/object/v1/AUTH_e99c1****f00-bffa-42*****05f3.m3u8',
...
  '-f',
  'flv',
```

I suspect the periodic artifacts may be related to these commands, and are more likely to occur in ffmpeg's own demuxer and muxer rather than the decoder and encoder. Use `-v verbose` and/or `-v debug` to get more information.

nyanmisaka, 2024-11-11:
@YannickPerret The recent commit 57b7952f6ae8497fc8091eb1a0cf33c2a238b50d may have fixed this issue. You can try again.

nyanmisaka, 2024-11-17:
Is this still an issue after the recent changes?

nyanmisaka, 2024-12-04:
Closing as similar errors have been fixed.

closed 2024-12-04 by nyanmisaka