nyanmisaka/ffmpeg-rockchip #127: compile error

gopower, 2024-11-11:
[image]

libavcodec/rkmppdec.c: In function ‘rkmpp_get_coding_type’:
libavcodec/rkmppdec.c:45:44: error: ‘MPP_VIDEO_CodingAV1’ undeclared (first use in this function); did you mean ‘MPP_VIDEO_CodingAVS’?
   45 |     case AV_CODEC_ID_AV1:           return MPP_VIDEO_CodingAV1;
      |                                            ^~~~~~~~~~~~~~~~~~~
      |                                            MPP_VIDEO_CodingAVS
libavcodec/rkmppdec.c:45:44: note: each undeclared identifier is reported only once for each function it appears in
libavcodec/rkmppenc.c: In function ‘rkmpp_get_mpp_fmt_h26x’:
libavcodec/rkmppenc.c:46:37: error: ‘MPP_FMT_YUV444P’ undeclared (first use in this function); did you mean ‘MPP_FMT_YUV444SP’?
   46 |     case AV_PIX_FMT_YUV444P: return MPP_FMT_YUV444P;
      |                                     ^~~~~~~~~~~~~~~
      |                                     MPP_FMT_YUV444SP
libavcodec/rkmppenc.c:46:37: note: each undeclared identifier is reported only once for each function it appears in
libavcodec/rkmppenc.c: In function ‘rkmpp_submit_frame’:
libavcodec/rkmppenc.c:683:9: error: implicit declaration of function ‘mpp_frame_set_fbc_hdr_stride’; did you mean ‘mpp_frame_set_hor_stride’? [-Werror=implicit-function-declaration]
  683 |         mpp_frame_set_fbc_hdr_stride(mpp_frame, hor_stride);

nyanmisaka, 2024-11-11:
You are using an outdated MPP library, please check the tutorials in the Wiki and update your MPP.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation

```
git clone -b jellyfin-mpp --depth=1 https://github.com/nyanmisaka/mpp.git
```

closed 2024-11-11 by nyanmisaka

sathishNexion, 2025-01-14:
[image]
[image]
[image]

I exactly Followed this --> https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation
But i encountered some errors, Friend

I am using rv1126