nyanmisaka/ffmpeg-rockchip #128: sws_scale是否底层是否支持rga转化格式？

tickmoo, 2024-11-14:


nyanmisaka, 2024-11-14:
RGA in ffmpeg-rockchip is implemented as an independent video filter `scale_rkrga`, which is independent of SwScale, similar to `scale_cuda` and `scale_vaapi`.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Filter

tickmoo, 2024-11-14:
使用ffmpeg 打开 mipi yuv流，这步不会调用解码，但是需要使用sws_scale 将 AvFrame转换成 rgb ，这里如何使用 rga 模块？十分感谢🙏

nyanmisaka, 2024-11-14:
I can only provide you with ffmpeg commands. If this does not meet your needs, you have to write your own code to call the libavfilter interface.

```
ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -i INPUT \
-vf hwupload,scale_cuda=format=rgba,hwmap,format=rgba -f rawvideo -y ~/dst.rgba
```

closed 2024-11-17 by nyanmisaka

qaz624824554, 2025-01-01:
我已经解决了，等我写篇文章出来就发出来