nyanmisaka/ffmpeg-rockchip #129: 会支持 AVS+ 的编码嘛

Wickscc, 2024-11-17:
mpp 是支持 avs+ 的, 但是编译后没有发现有, 是不支持嘛?

```
mpp[16426]: mpi_dec_utils: usage: mpi_dec_test [options]
mpp[16426]: utils: -i  input_file       input bitstream file
mpp[16426]: utils: -o  output_file      output bitstream file,
mpp[16426]: utils: -c  ops_file         input operation config file
mpp[16426]: utils: -w  width            the width of input bitstream
mpp[16426]: utils: -h  height           the height of input bitstream
mpp[16426]: utils: -t  type             input stream coding type
mpp[16426]: utils: -f  format           output frame format type
mpp[16426]: utils: -d  debug            debug flag
mpp[16426]: utils: -x  timeout          output timeout interval
mpp[16426]: utils: -n  frame_number     max output frame number
mpp[16426]: utils: -s  instance_nb      number of instances
mpp[16426]: utils: -(null)  (null)              (null)
mpp[16426]: mpi: mpp coding type support list:
mpp[16426]: mpi: type: dec id 0 coding: mpeg2            id 2
mpp[16426]: mpi: type: dec id 0 coding: mpeg4            id 4
mpp[16426]: mpi: type: dec id 0 coding: h.263            id 3
mpp[16426]: mpi: type: dec id 0 coding: h.264/AVC        id 7
mpp[16426]: mpi: type: dec id 0 coding: h.265/HEVC       id 16777220
mpp[16426]: mpi: type: dec id 0 coding: vp8              id 9
mpp[16426]: mpi: type: dec id 0 coding: VP9              id 10
mpp[16426]: mpi: type: dec id 0 coding: avs+             id 16777221
mpp[16426]: mpi: type: dec id 0 coding: jpeg             id 8
mpp[16426]: mpi: type: enc id 1 coding: h.264/AVC        id 7
mpp[16426]: mpi: type: enc id 1 coding: jpeg             id 8
mpp[16426]: mpi: type: enc id 1 coding: h265             id 16777220
mpp[16426]: mpi: type: enc id 1 coding: vp8              id 9
```

```
ffmpeg -decoders  | grep rkmpp
ffmpeg version 4.1.4-1firefly3 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --prefix=/usr --extra-version=1firefly3 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --disable-libaom --disable-libmysofa --disable-libvidstab --enable-libdrm --enable-librga --enable-rkmpp --enable-version3 --disable-libopenh264 --disable-vaapi --disable-vdpau --disable-decoder=h264_v4l2m2m --disable-decoder=vp8_v4l2m2m --disable-decoder=mpeg2_v4l2m2m --disable-decoder=mpeg4_v4l2m2m --enable-shared --disable-doc
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
 V..... h264_rkmpp           h264 (rkmpp) (codec h264)
 V..... hevc_rkmpp           hevc (rkmpp) (codec hevc)
 V..... vp8_rkmpp            vp8 (rkmpp) (codec vp8)
 V..... vp9_rkmpp            vp9 (rkmpp) (codec vp9)
```

nyanmisaka, 2024-11-17:
I tested it before but it is not stable, so I have no plans to add it. You can add it to `rkmppdec.c` yourself and try again.

https://github.com/nyanmisaka/ffmpeg-rockchip/blob/9dbaf5a69d8f2ecf01d4a09daa5989987b1aede2/libavcodec/rkmppdec.c#L39

```c
    MPP_VIDEO_CodingAVSPLUS,            /**< AVS+ */
    MPP_VIDEO_CodingAVS,                /**< AVS profile=0x20 */
    MPP_VIDEO_CodingAVS2,               /**< AVS2 */
```

```c
    AV_CODEC_ID_AVS,
    AV_CODEC_ID_AVS2,
```

Run `./ffmpeg` instead of `ffmpeg`

closed 2024-11-17 by nyanmisaka

Wickscc, 2024-11-18:
Thanks for the reply. I tried adding the following code in libavcodec/rkmppdec.c, but the FFmpeg decoder didn't add anything, and decoding AVS+ content still results in an error. Could it be due to instability factors?

```
static MppCodingType rkmpp_get_coding_type(AVCodecContext *avctx)
{
    switch (avctx->codec_id) {
    case AV_CODEC_ID_H263:          return MPP_VIDEO_CodingH263;
    case AV_CODEC_ID_H264:          return MPP_VIDEO_CodingAVC;
    case AV_CODEC_ID_HEVC:          return MPP_VIDEO_CodingHEVC;
    case AV_CODEC_ID_AV1:           return MPP_VIDEO_CodingAV1;
    case AV_CODEC_ID_VP8:           return MPP_VIDEO_CodingVP8;
    case AV_CODEC_ID_VP9:           return MPP_VIDEO_CodingVP9;
    case AV_CODEC_ID_MPEG1VIDEO:    /* fallthrough */
    case AV_CODEC_ID_MPEG2VIDEO:    return MPP_VIDEO_CodingMPEG2;
    case AV_CODEC_ID_MPEG4:         return MPP_VIDEO_CodingMPEG4;
    case AV_CODEC_ID_AVS:           return MPP_VIDEO_CodingAVS;         // 添加 AVS 编码类型
    case AV_CODEC_ID_AVS2:          return MPP_VIDEO_CodingAVSPLUS;        // 添加 AVS2 编码类型

//    case AV_CODEC_ID_AVSPLUS:       return MPP_VIDEO_CodingAVSPLUS;     // 添加 AVS+ 编码类型
    default:                        return MPP_VIDEO_CodingUnused;
    }
}
```

```
root@firefly:/home/wicks/ffmpeg-rockchip# ./ffmpeg -decoders | grep avs
ffmpeg version af480a2e1d Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp
  WARNING: library configuration mismatch
  avcodec     configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-libx264
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
 V....D avs                  AVS (Audio Video Standard) video
 V....D cavs                 Chinese AVS (Audio Video Standard) (AVS1-P2, JiZhun profile)
root@firefly:/home/wicks/ffmpeg-rockchip# ./ffmpeg -decoders | grep rkmpp
ffmpeg version af480a2e1d Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp
  WARNING: library configuration mismatch
  avcodec     configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-libx264
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
 V..... av1_rkmpp            Rockchip MPP (Media Process Platform) AV1 decoder (codec av1)
 V..... h263_rkmpp           Rockchip MPP (Media Process Platform) H263 decoder (codec h263)
 V..... h264_rkmpp           Rockchip MPP (Media Process Platform) H264 decoder (codec h264)
 V..... hevc_rkmpp           Rockchip MPP (Media Process Platform) HEVC decoder (codec hevc)
 V..... mpeg1_rkmpp          Rockchip MPP (Media Process Platform) MPEG1VIDEO decoder (codec mpeg1video)
 V..... mpeg2_rkmpp          Rockchip MPP (Media Process Platform) MPEG2VIDEO decoder (codec mpeg2video)
 V..... mpeg4_rkmpp          Rockchip MPP (Media Process Platform) MPEG4 decoder (codec mpeg4)
 V..... vp8_rkmpp            Rockchip MPP (Media Process Platform) VP8 decoder (codec vp8)
 V..... vp9_rkmpp            Rockchip MPP (Media Process Platform) VP9 decoder (codec vp9)
root@firefly:/home/wicks/ffmpeg-rockchip#
```

nyanmisaka, 2024-11-18:
https://github.com/nyanmisaka/ffmpeg-rockchip/blob/9dbaf5a69d8f2ecf01d4a09daa5989987b1aede2/libavcodec/rkmppdec.c#L1019-L1045

It also needs to be registered in many files.

```
configure
libavcodec/allcodecs.c
libavcodec/Makefile
```