nyanmisaka/ffmpeg-rockchip #130: 'buffer_hwupload': Invalid argument

tickmoo, 2024-11-18:
下面是我初始化filter的部分代码，看起来是没问题的，device_ref 也有设置，为什么最终跑起来时候，会爆这个错误？
[image]
[image]

以下是错误代码：
[image]

nyanmisaka, 2024-11-18:
It should be the lack of setting the hardware context for the filter.

Search `hw` in these codes:

https://github.com/FFmpeg/FFmpeg/blob/master/fftools/ffmpeg_hw.c

https://github.com/FFmpeg/FFmpeg/blob/master/fftools/ffmpeg_filter.c

tickmoo, 2024-11-18:
> It should be the lack of setting the hardware context for the filter.
> 
> Search `hw` or `filter_hw_device` or `hw_device_for_filter` in these codes:
> 
> https://github.com/FFmpeg/FFmpeg/blob/master/fftools/ffmpeg_hw.c
> 
> https://github.com/FFmpeg/FFmpeg/blob/master/fftools/ffmpeg_filter.c

不使用hwupload 过滤器，仅仅使用vpp_rkrga过滤器可以实现crop和yuv->rgb24的格式转换吗？

nyanmisaka, 2024-11-18:
Nope. `hwupload` is a must have filter for memory input.

memory ->(hwupload)-> hardware DMA-BUF -> RGA -> ...

closed 2024-11-19 by nyanmisaka

JianJingGao, 2024-12-03:
我跟你一样的情况，你后来怎么弄的？