nyanmisaka/ffmpeg-rockchip #131: scale_rkmpp: 'nv24' and 'nv42' are only supported by RGA2-Pro

Tryanks, 2024-11-18:
我正在使用 HDMI RX 作为输入源，ffmpeg 将视频流编码为 mjpeg。
通过 v4l2-ctl 命令查看 /dev/video0 设备信息，当插入我的电脑的信号后，注意到输入的像素格式是 NV24（像素堆叠方式为 YYYYUVUVUVUV）。
这导致必须要转换像素格式，才能送入 mjpeg_rkmpp 编码器。我希望使用 rga 加速这个过程，于是我构造了这样的命令：
```shell
ffmpeg \
    -init_hw_device rkmpp=hw \
    -filter_hw_device hw \
    -f v4l2 \
    -i /dev/video0 \
    -vf hwupload,scale_rkrga=format=nv24,hwmap,format=bgra \
    -c:v mjpeg_rkmpp \
    -qp_init 80 \
    -vframes 10 -y \
    rga/tmp%d.jpg
```
遗憾的是，这项命令无法执行，报错为 `'nv24' and 'nv42' are only supported by RGA2-Pro`。
我想这应该没办法解决了，我的疑问是，已经没有其它任何方法加速像素格式转换的过程了吗？
因为纯靠 CPU 计算这个无法维持长时间运行，我的硬件已经因此炸掉了一个。

nyanmisaka, 2024-11-18:
Only H264 and HEVC encoders on RK3588 can handle NV24 format. Neither RGA3 nor RGA2-Enhance support NV24. RGA2-Pro in RK3576 is the first RGA hardware to support NV24.

So you have to prevent HDMI RX from outputting NV24. The code here may be relevant, but you have to try it on your own.

https://github.com/friendlyarm/kernel-rockchip/commit/ee391fb229f90af6866cba9e514fd96676dbf843

Tryanks, 2024-11-18:
遗憾的是，HDMI RX 的输入格式取决于插在上面的设备，它本身并不参与像素格式转换的过程。看起来这个问题缺失无法解决，我必须考虑改用 h264/hevc 编码器来重整视频流，或者采用为 CPU 散热等辅助方案来强行做像素转换。
感谢您的指示，非常有帮助，因为我在此之前是不知道 h26x_rkmpp 可以直接处理 NV24 的。

closed 2024-11-18 by Tryanks

nyanmisaka, 2024-11-18:
HDMI RX is similar to a display and reports via EDID what formats it supports. If it only reports YCRCB420 support then the HDMI output device should not output YCRCB444.

Tryanks, 2024-11-18:
> So you have to prevent HDMI RX from outputting NV24. The code here may be relevant.
> 
> [friendlyarm/kernel-rockchip@ee391fb](https://github.com/friendlyarm/kernel-rockchip/commit/ee391fb229f90af6866cba9e514fd96676dbf843)

难以表达我对您指导的感谢之情。
通过您提供的这份代码参考，我成功将其移植到我的设备 OrangePi5Plus 中。
（这份回复只是为了其它可能看到这个 issue 的人而写的，我为偏离仓库主题而道歉）
以下是根据上述提供的 commit 参考编写的，适用于 opi5+ 的内核 patch：
https://gist.github.com/Tryanks/54cb0100a43d6362ac76df09665f1054

应用这个 patch 并编译安装内核，随后执行 `echo 3 > /sys/class/hdmirx/hdmirx/edid`，再接入 HDMIRX 设备，便可以惊讶地发现：
```
root@orangepi5plus:~# v4l2-ctl -d /dev/video0  -V
Format Video Capture Multiplanar:
        Width/Height      : 1920/1080
        Pixel Format      : 'BGR3' (24-bit BGR 8-8-8)
        Field             : None
        Number of planes  : 1
        Flags             : premultiplied-alpha, 0x000000fe
        Colorspace        : sRGB
        Transfer Function : Unknown (0x000000b8)
        YCbCr/HSV Encoding: Unknown (0x000000ff)
        Quantization      : Limited Range
        Plane 0           :
           Bytes per Line : 5760
           Size Image     : 6220800
```
It works!
Magic.