nyanmisaka/ffmpeg-rockchip #133: [h264_rkmpp @ 0xffff585e4c60] Failed to init MPP context: -1

ageevp, 2024-11-20:
ffplay -i '/home/petr/Downloads/bbb_sunflower_2160p_60fps_normal.mp4' -codec:v h264_rkmpp
ffplay version abfc47f Copyright (c) 2003-2023 the FFmpeg developers
  built with gcc 14 (Ubuntu 14.2.0-4ubuntu2)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/petr/Downloads/bbb_sunflower_2160p_60fps_normal.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 1
    compatible_brands: isomavc1
    creation_time   : 2013-12-17T16:40:26.000000Z
    title           : Big Buck Bunny, Sunflower version
    artist          : Blender Foundation 2008, Janus Bager Kristensen 2013
    comment         : Creative Commons Attribution 3.0 - http://bbb3d.renderfarming.net
    genre           : Animation
    composer        : Sacha Goedegebure
  Duration: 00:10:34.57, start: 0.000000, bitrate: 8487 kb/s
  Stream #0:0[0x1](und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(progressive), 3840x2160 [SAR 1:1 DAR 16:9], 8002 kb/s, 60 fps, 60 tbr, 60k tbn (default)
    Metadata:
      creation_time   : 2013-12-17T16:40:26.000000Z
      handler_name    : GPAC ISO Video Handler
      vendor_id       : [0][0][0][0]
  Stream #0:1[0x2](und): Audio: mp3 (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 160 kb/s (default)
    Metadata:
      creation_time   : 2013-12-17T16:40:28.000000Z
      handler_name    : GPAC ISO Audio Handler
      vendor_id       : [0][0][0][0]
  Stream #0:2[0x3](und): Audio: ac3 (ac-3 / 0x332D6361), 48000 Hz, 5.1(side), fltp, 320 kb/s (default)
    Metadata:
      creation_time   : 2013-12-17T16:40:28.000000Z
      handler_name    : GPAC ISO Audio Handler
      vendor_id       : [0][0][0][0]
    Side data:
      audio service type: main
[h264_rkmpp @ 0xffff585e4c60] Failed to init MPP context: -1
   6.41 M-A:  0.000 fd=   0 aq=   20KB vq=    0KB sq=    0B f=0/0

ageevp, 2024-11-20:
uname -r
6.11.0-1006-rockchip

Tryanks, 2024-11-20:
Try 6.1.43 BSP Kernel

nyanmisaka, 2024-11-20:
Mainline linux doesn't support RKMPP. Use rockchip BSP kernel 5.10 or 6.1 LTS instead.

closed 2024-11-20 by nyanmisaka