nyanmisaka/ffmpeg-rockchip #139: build ffmpeg demo link failed

JianJingGao, 2024-11-28:
Please help me look this
`proc] Executing command: /usr/bin/cmake --build /home/teamhd/workspace/rk-ffmpeg/build --config Debug --target all --
[build] [1/1 100% :: 0.978] Linking CXX executable bin/RK_FFMPEG
[build] FAILED: bin/RK_FFMPEG 
[build] : && /usr/bin/aarch64-linux-gnu-g++  -std=c++11 -Wall -O0 -Wfatal-errors   -pthread -w -g -g  -rdynamic CMakeFiles/RK_FFMPEG.dir/CMakeFiles/3.16.3/CompilerIdCXX/CMakeCXXCompilerId.cpp.o CMakeFiles/RK_FFMPEG.dir/main.cpp.o  -o bin/RK_FFMPEG  -Wl,-Bstatic  -lavcodec  -lavformat  -lavutil  -lswscale  -lswresample  -Wl,-Bdynamic  -lz && :
[build] /usr/bin/ld: CMakeFiles/RK_FFMPEG.dir/main.cpp.o: in function `main':
[build] /home/teamhd/workspace/rk-ffmpeg/build/../main.cpp:116: multiple definition of `main'; CMakeFiles/RK_FFMPEG.dir/CMakeFiles/3.16.3/CompilerIdCXX/CMakeCXXCompilerId.cpp.o:/home/teamhd/workspac 
[build] /usr/bin/ld: (.text.unlikely+0xe0): undefined reference to `aom_codec_av1_dx_algo'
[build] /usr/bin/ld: (.text.unlikely+0xf0): undefined reference to `aom_codec_av1_dx_algo'
[build] /usr/bin/ld: (.text.unlikely+0xf8): undefined reference to `aom_codec_dec_init_ver'
[build] /usr/bin/ld: (.text.unlikely+0x104): undefined reference to `aom_codec_error'
[build] /usr/bin/ld: (.text+0xbb8): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: (.text+0xbd8): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0xc2c): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavformat.a(dashdec.o): in function `get_content_url.constprop.0':
[build] (.text+0x1508): undefined reference to `xmlNodeGetContent'
[build] /usr/bin/ld: (.text+0x1558): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x1560): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x1570): undefined reference to `xmlNodeGetContent'
[build] /usr/bin/ld: (.text+0x15c0): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x15c8): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x15d8): undefined reference to `xmlNodeGetContent'
[build] /usr/bin/ld: (.text+0x1628): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x1630): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x1640): undefined reference to `xmlNodeGetContent'
[build] /usr/bin/ld: (.text+0x1690): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x1698): undefined reference to `xmlFree'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavformat.a(dashdec.o): in function `parse_manifest':
[build] (.text+0x214c): undefined reference to `xmlCheckVersion'
[build] /usr/bin/ld: (.text+0x2164): undefined reference to `xmlReadMemory'
[build] /usr/bin/ld: (.text+0x216c): undefined reference to `xmlDocGetRootElement'
[build] /usr/bin/ld: (.text+0x21d0): undefined reference to `xmlFreeDoc'
[build] /usr/bin/ld: (.text+0x21d4): undefined reference to `xmlCleanupParser'
[build] /usr/bin/ld: (.text+0x21dc): undefined reference to `xmlFreeNode'
[build] /usr/bin/ld: (.text+0x22e0): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: (.text+0x2300): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x230c): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2360): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2368): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2384): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: (.text+0x2494): undefined reference to `xmlFirstElementChild'
[build] /usr/bin/ld: (.text+0x24b0): undefined reference to `xmlNextElementSibling'
[build] /usr/bin/ld: (.text+0x24d4): undefined reference to `xmlCopyNode'
[build] /usr/bin/ld: (.text+0x24e0): undefined reference to `xmlFirstElementChild'
[build] /usr/bin/ld: (.text+0x2528): undefined reference to `xmlNextElementSibling'
[build] /usr/bin/ld: (.text+0x2550): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2560): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x25a0): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: (.text+0x2618): undefined reference to `xmlFirstElementChild'
[build] /usr/bin/ld: (.text+0x2640): undefined reference to `xmlNodeGetContent'
[build] /usr/bin/ld: (.text+0x2664): undefined reference to `xmlNextElementSibling'
[build] /usr/bin/ld: (.text+0x2668): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2674): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x26a8): undefined reference to `xmlNodeGetContent'
[build] /usr/bin/ld: (.text+0x26e0): undefined reference to `xmlNodeGetContent'
[build] /usr/bin/ld: (.text+0x2710): undefined reference to `xmlFirstElementChild'
[build] /usr/bin/ld: (.text+0x275c): undefined reference to `xmlNextElementSibling'
[build] /usr/bin/ld: (.text+0x2810): undefined reference to `xmlFirstElementChild'
[build] /usr/bin/ld: (.text+0x2858): undefined reference to `xmlNextElementSibling'
[build] /usr/bin/ld: (.text+0x2888): undefined reference to `xmlNewNode'
[build] /usr/bin/ld: (.text+0x2a4c): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: (.text+0x2a60): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: (.text+0x2a74): undefined reference to `xmlGetProp'
[build] /usr/bin/ld: (.text+0x2aac): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2ab0): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2ac8): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2acc): undefined reference to `xmlFree'
[build] /usr/bin/ld: (.text+0x2ae0): undefined reference to `xmlFree'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavformat.a(dashdec.o):(.text+0x2ae4): more undefined references to `xmlFree' follow
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavformat.a(dashdec.o): in function `parse_manifest':
[... 198 lines cut ...]
[build] /usr/bin/ld: (.text+0x30ec): undefined reference to `clReleaseEvent'
[build] /usr/bin/ld: (.text+0x31a4): undefined reference to `clEnqueueUnmapMemObject'
[build] /usr/bin/ld: (.text+0x31bc): undefined reference to `clWaitForEvents'
[build] /usr/bin/ld: (.text+0x31e8): undefined reference to `clReleaseEvent'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libswresample.a(soxr_resample.o): in function `get_out_samples':
[build] (.text+0x84): undefined reference to `soxr_delay'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libswresample.a(soxr_resample.o): in function `get_delay':
[build] (.text+0x10c): undefined reference to `soxr_delay'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libswresample.a(soxr_resample.o): in function `flush':
[build] (.text+0x194): undefined reference to `soxr_delay'
[build] /usr/bin/ld: (.text+0x1b8): undefined reference to `soxr_process'
[build] /usr/bin/ld: (.text+0x1d8): undefined reference to `soxr_process'
[build] /usr/bin/ld: (.text+0x1e0): undefined reference to `soxr_delay'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libswresample.a(soxr_resample.o): in function `process':
[build] (.text+0x258): undefined reference to `soxr_set_num_channels'
[build] /usr/bin/ld: (.text+0x264): undefined reference to `soxr_set_error'
[build] /usr/bin/ld: (.text+0x2bc): undefined reference to `soxr_process'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libswresample.a(soxr_resample.o): in function `destroy':
[build] (.text+0x304): undefined reference to `soxr_delete'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libswresample.a(soxr_resample.o): in function `create':
[build] (.text+0x3dc): undefined reference to `soxr_io_spec'
[build] /usr/bin/ld: (.text+0x40c): undefined reference to `soxr_quality_spec'
[build] /usr/bin/ld: (.text+0x440): undefined reference to `soxr_delete'
[build] /usr/bin/ld: (.text+0x460): undefined reference to `soxr_create'
[build] collect2: error: ld returned 1 exit status`

JianJingGao, 2024-11-28:
I added some dependencies, I reduced some errors, and I still can't find these

`[build] /home/teamhd/workspace/rk-ffmpeg/build/../main.cpp:116: multiple definition of `main'; CMakeFiles/RK_FFMPEG.dir/CMakeFiles/3.16.3/CompilerIdCXX/CMakeCXXCompilerId.cpp.o:/home/teamhd/workspace/rk-ffmpeg/build/CMakeFiles/3.16.3/CompilerIdCXX/CMakeCXXCompilerId.cpp:638: first defined here
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavcodec.a(rkmppdec.o): in function `rkmpp_decode_init':
[build] /home/teamhd/dev/ffmpeg/libavcodec/rkmppdec.c:283: undefined reference to `querystring'
[build] collect2: error: ld returned 1 exit status
[build] ninja: build stopped: subcommand failed.`

JianJingGao, 2024-11-28:
# 链接 FFmpeg 库到可执行文件
target_link_libraries(RK_FFMPEG

/usr/lib/aarch64-linux-gnu/libavcodec.a
/usr/lib/aarch64-linux-gnu/libavformat.a
/usr/lib/aarch64-linux-gnu/libavutil.a
/usr/lib/aarch64-linux-gnu/libswscale.a
/usr/lib/aarch64-linux-gnu/libswresample.a
${ZLIB_LIBRARIES}  # 链接 zlib 库
${ROCKCHIP_MPP_LIBRARIES}
${LZMA_LIBRARIES}
${LIBDRM_LIBRARIES}
)

nyanmisaka, 2024-11-28:
These errors are clearly NOT caused by ffmpeg-rockchip. Please learn how to build ffmpeg and fix your build scripts.

JianJingGao, 2024-11-28:
I solved these problems by adding mpp, rkrgb and other dependencies。
# 链接 FFmpeg 库到可执行文件
target_link_libraries(RK_FFMPEG

/usr/lib/aarch64-linux-gnu/libavcodec.a
/usr/lib/aarch64-linux-gnu/libavformat.a
/usr/lib/aarch64-linux-gnu/libavutil.a
/usr/lib/aarch64-linux-gnu/libswscale.a
/usr/lib/aarch64-linux-gnu/libswresample.a
/home/teamhd/dev/rkrga_build/librga.so
${ZLIB_LIBRARIES}  # 链接 zlib 库
${ROCKCHIP_MPP_LIBRARIES}
${LZMA_LIBRARIES}
${LIBDRM_LIBRARIES}
)

closed 2024-11-28 by JianJingGao