nyanmisaka/ffmpeg-rockchip #140: Added libfilter.a to implement drm2BGR24, but failed to compile.

JianJingGao, 2024-11-29:
Added libfilter.a to implement drm2BGR24, but failed to compile. Please help me check what is the problem?

ERROR Message
```
[build] : && /usr/bin/aarch64-linux-gnu-g++  -std=c++11 -Wall -O0 -Wfatal-errors   -pthread -w -g -g  -rdynamic CMakeFiles/YWPLAY_PLUG.dir/src/YWMuPlayer.cpp.o CMakeFiles/YWPLAY_PLUG.dir/src/YWPlayerQSV.cpp.o CMakeFiles/YWPLAY_PLUG.dir/src/common/file_system.cpp.o CMakeFiles/YWPLAY_PLUG.dir/main.cpp.o  -o /home/teamhd/workspace/workspace/bin/YWPLAY_PLUG  -Wl,-rpath,/home/teamhd/dev/rkrga_build  -Wl,-Bstatic  -lavcodec  -lavformat  -lavutil  -lavfilter  -lswscale  -lpostproc  -lswresample  -Wl,-Bdynamic  /home/teamhd/dev/rkrga_build/librga.so  -lz  -lrockchip_mpp  -llzma  -ldrm  /usr/lib/aarch64-linux-gnu/libboost_thread.so.1.71.0  /usr/lib/aarch64-linux-gnu/libboost_atomic.so.1.71.0  -lpthread && :
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(f_sendcmd.o): in function `init':
[build] /home/teamhd/dev/ffmpeg/libavfilter/f_sendcmd.c:425: undefined reference to `av_file_map'
[build] /usr/bin/ld: /home/teamhd/dev/ffmpeg/libavfilter/f_sendcmd.c:433: undefined reference to `av_file_unmap'
[build] /usr/bin/ld: /home/teamhd/dev/ffmpeg/libavfilter/f_sendcmd.c:438: undefined reference to `av_file_unmap'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_curves.o): in function `parse_psfile':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_curves.c:594: undefined reference to `av_file_map'
[build] /usr/bin/ld: /home/teamhd/dev/ffmpeg/libavfilter/vf_curves.c:635: undefined reference to `av_file_unmap'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_deshake.o): in function `filter_frame':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_deshake.c:429: undefined reference to `av_pixelutils_get_sad_fn'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_dnn_detect.o): in function `dnn_detect_post_proc_ov':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_dnn_detect.c:93: undefined reference to `av_detection_bbox_create_side_data'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_dnn_detect.o): in function `dnn_detect_post_proc_tf':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_dnn_detect.c:171: undefined reference to `av_detection_bbox_create_side_data'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_mpdecimate.o): in function `init':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_mpdecimate.c:159: undefined reference to `av_pixelutils_get_sad_fn'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_spp.o): in function `child_class_iterate':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_spp.c:52: undefined reference to `avcodec_dct_get_class'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_spp.o): in function `config_input':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_spp.c:346: undefined reference to `avcodec_dct_init'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_spp.o): in function `preinit':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_spp.c:463: undefined reference to `avcodec_dct_alloc'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vsrc_cellauto.o): in function `uninit':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vsrc_cellauto.c:211: undefined reference to `av_file_unmap'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vsrc_cellauto.o): in function `init_pattern_from_file':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vsrc_cellauto.c:146: undefined reference to `av_file_map'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vsrc_life.o): in function `uninit':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vsrc_life.c:274: undefined reference to `av_file_unmap'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vsrc_life.o): in function `init_pattern_from_file':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vsrc_life.c:169: undefined reference to `av_file_map'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_selectivecolor.o): in function `parse_psfile':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_selectivecolor.c:192: undefined reference to `av_file_map'
[build] /usr/bin/ld: /home/teamhd/dev/ffmpeg/libavfilter/vf_selectivecolor.c:233: undefined reference to `av_file_unmap'
[build] /usr/bin/ld: /home/teamhd/dev/ffmpeg/libavfilter/vf_selectivecolor.c:233: undefined reference to `av_file_unmap'
[build] /usr/bin/ld: /home/teamhd/dev/ffmpeg/libavfilter/vf_selectivecolor.c:233: undefined reference to `av_file_unmap'
[build] collect2: error: ld returned 1 exit status

Find that a reference to this function exists.

```
nm /usr/lib/aarch64-linux-gnu/libavutil.a | grep av_file_unmap
00000000000001b0 T av_file_unmap
```

```
CMAKE LINK:

target_link_libraries(YWPLAY_PLUG
/usr/lib/aarch64-linux-gnu/libavcodec.a
/usr/lib/aarch64-linux-gnu/libavformat.a
/usr/lib/aarch64-linux-gnu/libavutil.a
/usr/lib/aarch64-linux-gnu/libavfilter.a
/usr/lib/aarch64-linux-gnu/libswscale.a
/usr/lib/aarch64-linux-gnu/libpostproc.a
/usr/lib/aarch64-linux-gnu/libswresample.a

/home/teamhd/dev/rkrga_build/librga.so
${ZLIB_LIBRARIES}  # 链接 zlib 库
${ROCKCHIP_MPP_LIBRARIES}
${LZMA_LIBRARIES}
${LIBDRM_LIBRARIES}
${Boost_LIBRARIES}
 readerwriterqueue  nlohmann_json::nlohmann_json

  )

nyanmisaka, 2024-11-29:
link order between lavu and lavfi

https://stackoverflow.com/questions/45135/why-does-the-order-in-which-libraries-are-linked-sometimes-cause-errors-in-gcc

JianJingGao, 2024-11-29:
> link order between lavu and lavfi
> 
> https://stackoverflow.com/questions/45135/why-does-the-order-in-which-libraries-are-linked-sometimes-cause-errors-in-gcc

halo，I changed the dependency order and this error occurred, which I do not understand。

[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_spp.c:52: undefined reference to `avcodec_dct_get_class'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_spp.o): in function `config_input':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_spp.c:346: undefined reference to `avcodec_dct_init'
[build] /usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libavfilter.a(vf_spp.o): in function `preinit':
[build] /home/teamhd/dev/ffmpeg/libavfilter/vf_spp.c:463: undefined reference to `avcodec_dct_alloc'

JianJingGao, 2024-11-29:
target_link_libraries(YWPLAY_PLUG
/usr/lib/aarch64-linux-gnu/libavcodec.a
/usr/lib/aarch64-linux-gnu/libavformat.a
/usr/lib/aarch64-linux-gnu/libavfilter.a
/usr/lib/aarch64-linux-gnu/libavutil.a
/usr/lib/aarch64-linux-gnu/libswscale.a
/usr/lib/aarch64-linux-gnu/libswresample.a
/usr/lib/aarch64-linux-gnu/libpostproc.a

/home/teamhd/dev/rkrga_build/librga.so
${ZLIB_LIBRARIES}  # 链接 zlib 库
${ROCKCHIP_MPP_LIBRARIES}
${LZMA_LIBRARIES}
${LIBDRM_LIBRARIES}
${Boost_LIBRARIES}
 readerwriterqueue  nlohmann_json::nlohmann_json

  )

nyanmisaka, 2024-11-29:
I am not familiar with your build script. So can't provide more help beyond the wiki.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation

JianJingGao, 2024-11-29:
> I am not familiar with your build script. So can't provide more help beyond the wiki.
> 
> https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation

I tried to compile example, the first few demos can be compiled successfully, until the filter demo reported an error, I suspect that there is a problem with this component. Would you be interested in taking a look for me?

`teamhd@teamhd:~/dev/ffmpeg/doc/examples$ make -f Makefile.example
cc   decode_filter_video.o  -lavdevice -lm -latomic -ldrm -lxcb -lasound -lavfilter -pthread -lm -latomic -lrga -lpostproc -lm -latomic -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lrga -llzma -lrockchip_mpp -lz -lswresample -lm -latomic -lswscale -lm -latomic -lavutil -pthread -lm -ldrm -lrockchip_mpp -latomic -lX11  -o decode_filter_video
collect2: error: ld returned 1 exit status
make: *** [<builtin>: decode_filter_video] Error 1`

JianJingGao, 2024-11-29:
teamhd@teamhd:~/dev/ffmpeg/doc/examples$ make -f Makefile.example
cc  -Wall -g   -c -o avio_http_serve_files.o avio_http_serve_files.c
avio_http_serve_files.c: In function ‘process_client’:
avio_http_serve_files.c:45:32: warning: pointer targets in passing argument 1 of ‘strlen’ differ in signedness [-Wpointer-sign]
   45 |         if (resource && strlen(resource))
      |                                ^~~~~~~~
      |                                |
      |                                uint8_t * {aka unsigned char *}
In file included from /usr/include/libavutil/common.h:40,
                 from /usr/include/libavutil/avutil.h:301,
                 from /usr/include/libavcodec/codec_par.h:26,
                 from /usr/include/libavformat/avformat.h:312,
                 from avio_http_serve_files.c:31:
/usr/include/string.h:385:35: note: expected ‘const char *’ but argument is of type ‘uint8_t *’ {aka ‘unsigned char *’}
  385 | extern size_t strlen (const char *__s)
      |                       ~~~~~~~~~~~~^~~
avio_http_serve_files.c:52:61: warning: pointer targets in passing argument 1 of ‘strcmp’ differ in signedness [-Wpointer-sign]
   52 |     if (resource && resource[0] == '/' && !strcmp((resource + 1), in_uri)) {
      |                                                   ~~~~~~~~~~^~~~
      |                                                             |
      |                                                             uint8_t * {aka unsigned char *}
In file included from /usr/include/libavutil/common.h:40,
                 from /usr/include/libavutil/avutil.h:301,
                 from /usr/include/libavcodec/codec_par.h:26,
                 from /usr/include/libavformat/avformat.h:312,
                 from avio_http_serve_files.c:31:
/usr/include/string.h:137:32: note: expected ‘const char *’ but argument is of type ‘uint8_t *’ {aka ‘unsigned char *’}
  137 | extern int strcmp (const char *__s1, const char *__s2)
      |                    ~~~~~~~~~~~~^~~~
cc  -Wall -g   -c -o avio_list_dir.o avio_list_dir.c
cc  -Wall -g   -c -o avio_read_callback.o avio_read_callback.c
cc  -Wall -g   -c -o decode_audio.o decode_audio.c
cc  -Wall -g   -c -o decode_filter_audio.o decode_filter_audio.c
cc  -Wall -g   -c -o decode_filter_video.o decode_filter_video.c
cc  -Wall -g   -c -o decode_video.o decode_video.c
cc  -Wall -g   -c -o demux_decode.o demux_decode.c
cc  -Wall -g   -c -o encode_audio.o encode_audio.c
cc  -Wall -g   -c -o encode_video.o encode_video.c
cc  -Wall -g   -c -o extract_mvs.o extract_mvs.c
cc  -Wall -g   -c -o hw_decode.o hw_decode.c
cc  -Wall -g   -c -o mux.o mux.c
cc  -Wall -g   -c -o remux.o remux.c
cc  -Wall -g   -c -o resample_audio.o resample_audio.c
cc  -Wall -g   -c -o scale_video.o scale_video.c
cc  -Wall -g   -c -o show_metadata.o show_metadata.c
cc  -Wall -g   -c -o transcode_aac.o transcode_aac.c
cc  -Wall -g   -c -o transcode.o transcode.c
cc   avio_http_serve_files.o  -lavdevice -lm -latomic -ldrm -lxcb -lasound -lavfilter -pthread -lm -latomic -lrga -lpostproc -lm -latomic -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lrga -llzma -lrockchip_mpp -lz -lswresample -lm -latomic -lswscale -lm -latomic -lavutil -pthread -lm -ldrm -lrockchip_mpp -latomic -lX11  -o avio_http_serve_files
cc   avio_list_dir.o  -lavdevice -lm -latomic -ldrm -lxcb -lasound -lavfilter -pthread -lm -latomic -lrga -lpostproc -lm -latomic -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lrga -llzma -lrockchip_mpp -lz -lswresample -lm -latomic -lswscale -lm -latomic -lavutil -pthread -lm -ldrm -lrockchip_mpp -latomic -lX11  -o avio_list_dir
cc   avio_read_callback.o  -lavdevice -lm -latomic -ldrm -lxcb -lasound -lavfilter -pthread -lm -latomic -lrga -lpostproc -lm -latomic -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lrga -llzma -lrockchip_mpp -lz -lswresample -lm -latomic -lswscale -lm -latomic -lavutil -pthread -lm -ldrm -lrockchip_mpp -latomic -lX11  -o avio_read_callback
cc   decode_audio.o  -lavdevice -lm -latomic -ldrm -lxcb -lasound -lavfilter -pthread -lm -latomic -lrga -lpostproc -lm -latomic -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lrga -llzma -lrockchip_mpp -lz -lswresample -lm -latomic -lswscale -lm -latomic -lavutil -pthread -lm -ldrm -lrockchip_mpp -latomic -lX11  -o decode_audio
cc   decode_filter_audio.o  -lavdevice -lm -latomic -ldrm -lxcb -lasound -lavfilter -pthread -lm -latomic -lrga -lpostproc -lm -latomic -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lrga -llzma -lrockchip_mpp -lz -lswresample -lm -latomic -lswscale -lm -latomic -lavutil -pthread -lm -ldrm -lrockchip_mpp -latomic -lX11  -o decode_filter_audio
cc   decode_filter_video.o  -lavdevice -lm -latomic -ldrm -lxcb -lasound -lavfilter -pthread -lm -latomic -lrga -lpostproc -lm -latomic -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lrga -llzma -lrockchip_mpp -lz -lswresample -lm -latomic -lswscale -lm -latomic -lavutil -pthread -lm -ldrm -lrockchip_mpp -latomic -lX11  -o decode_filter_video
collect2: error: ld returned 1 exit status
make: *** [<builtin>: decode_filter_video] Error 1

JianJingGao, 2024-11-29:
This problem is solved, because the default is compiled out of the static library, so this problem is dangerous

closed 2024-11-29 by JianJingGao