nyanmisaka/ffmpeg-rockchip #144: h264_rkmpp missing if compile with --enable-shared --enable-pic

PMGZED, 2024-12-15:
Hello
If i compile ffmpeg with --enable-shared --enable-pic  the ./configure outputs that is it compile with h264_rkmpp but if i try to use it i got 

[vost#0:0 @ 0xaaab08d10760] Unknown encoder 'h264_rkmpp'
[vost#0:0 @ 0xaaab08d10760] Error selecting an encoder
Error opening output file test.mp4.
Error opening output files: Encoder not found

is there a way to compile it shared?

PMGZED, 2024-12-15:
I am amazed 
i have compiled again with the start parameter ./configure  --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
and got
[vost#0:0 @ 0xaaaaca3dd760] Unknown encoder 'h264_rkmpp'
[vost#0:0 @ 0xaaaaca3dd760] Error selecting an encoder
Error opening output file test.mp4.

again?

nyanmisaka, 2024-12-15:
Are you sure you know the difference between running ./ffmpeg and ffmpeg?

PMGZED, 2024-12-15:
Yes system installation and local build
I have write a new image to the sd kart and i can reproduce it same error. now i flash again and try --enable-shared --enable-pic on the first run

nyanmisaka, 2024-12-15:
Uninstall `ffmpeg` and `libav*` packages from your system and try again.

PMGZED, 2024-12-15:
Ok i Have a fresh new SD Kart Image
ffmpeg is not installed

Output of ./configure --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-shared --enable-pic

```
root@radxa-zero3:/ffmpeg-rockchip# ./configure --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-shared --enable-pic
install prefix            /usr/local
source path               .
C compiler                gcc
C library                 glibc
ARCH                      aarch64 (generic)
big-endian                no
runtime cpu detection     yes
NEON enabled              yes
DOTPROD enabled           yes
I8MM enabled              yes
debug symbols             yes
strip symbols             yes
optimize for size         no
optimizations             yes
static                    yes
shared                    yes
postprocessing support    yes
network support           yes
threading support         pthreads
safe bitstream reader     yes
texi2html enabled         no
perl enabled              yes
pod2man enabled           yes
makeinfo enabled          no
makeinfo supports HTML    no
xmllint enabled           no

External libraries:
iconv                   libdrm                  rkmpp                   rkrga                   zlib

External libraries providing hardware acceleration:
cuda_llvm               rkmpp                   v4l2_m2m

Libraries:
avcodec                 avfilter                avutil                  swresample
avdevice                avformat                postproc                swscale

Programs:
ffmpeg                  ffprobe

Enabled decoders:
aac                     aura                    flv                     mp3on4                  pgx                     truemotion2rt
aac_fixed               aura2                   fmvc                    mp3on4float             phm                     truespeech
aac_latm                av1                     fourxm                  mpc7                    photocd                 tscc
aasc                    av1_rkmpp               fraps                   mpc8                    pictor                  tscc2
ac3                     avrn                    frwu                    mpeg1_rkmpp             pixlet                  tta
ac3_fixed               avrp                    ftr                     mpeg1_v4l2m2m           pjs                     twinvq
acelp_kelvin            avs                     g2m                     mpeg1video              png                     txd
adpcm_4xm               avui                    g723_1                  mpeg2_rkmpp             ppm                     ulti
adpcm_adx               ayuv                    g729                    mpeg2_v4l2m2m           prores                  utvideo
adpcm_afc               bethsoftvid             gdv                     mpeg2video              prosumer                v210
adpcm_agm               bfi                     gem                     mpeg4                   psd                     v210x
adpcm_aica              bink                    gif                     mpeg4_rkmpp             ptx                     v308
adpcm_argo              binkaudio_dct           gremlin_dpcm            mpeg4_v4l2m2m           qcelp                   v408
adpcm_ct                binkaudio_rdft          gsm                     mpegvideo               qdm2                    v410
adpcm_dtk               bintext                 gsm_ms                  mpl2                    qdmc                    vb
adpcm_ea                bitpacked               h261                    msa1                    qdraw                   vble
adpcm_ea_maxis_xa       bmp                     h263                    mscc                    qoi                     vbn
adpcm_ea_r1             bmv_audio               h263_rkmpp              msmpeg4v1               qpeg                    vc1
adpcm_ea_r2             bmv_video               h263_v4l2m2m            msmpeg4v2               qtrle                   vc1_v4l2m2m
adpcm_ea_r3             bonk                    h263i                   msmpeg4v3               r10k                    vc1image
adpcm_ea_xas            brender_pix             h263p                   msnsiren                r210                    vcr1
adpcm_g722              c93                     h264                    msp2                    ra_144                  vmdaudio
[... 304 lines cut ...]
Enabled indevs:
fbdev                   kmsgrab                 lavfi                   oss                     v4l2

Enabled outdevs:
fbdev                   oss                     v4l2

License: GPL version 3 or later
root@radxa-zero3:/ffmpeg-rockchip#

```

output shell

```
root@radxa-zero3:/ffmpeg-rockchip# ffmpeg
-bash: ffmpeg: command not found
root@radxa-zero3:/ffmpeg-rockchip# ./ffmpeg   -hide_banner -f lavfi -i testsrc2=duration=300:size=1920x1080:rate=30  -c:v h264_rkmpp  test.mp4
Input #0, lavfi, from 'testsrc2=duration=300:size=1920x1080:rate=30':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 30 tbn
[vost#0:0 @ 0xaaaacd1f4760] Unknown encoder 'h264_rkmpp'
[vost#0:0 @ 0xaaaacd1f4760] Error selecting an encoder
Error opening output file test.mp4.
Error opening output files: Encoder not found
```

nyanmisaka, 2024-12-15:
`sudo apt remove libav*`

PMGZED, 2024-12-15:
ok maybe the armbianimage has a failure?
i never installed ffmpeg with apt or make install
if i try to uninstall
```
root@radxa-zero3:/ffmpeg-rockchip# apt remove libav*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libavcodec
E: Unable to locate package libavdevice
E: Unable to locate package libavfilter
E: Unable to locate package libavformat
E: Unable to locate package libavutil
```

PMGZED, 2024-12-15:
Ok
 find / | grep libav 
shows 

```
/usr/lib/aarch64-linux-gnu/libavahi-glib.so.1
/usr/lib/aarch64-linux-gnu/libavutil.so.58
/usr/lib/aarch64-linux-gnu/libavformat.so.60
/usr/lib/aarch64-linux-gnu/libavahi-client.so.3.2.9
/usr/lib/aarch64-linux-gnu/libavtp.so.0.2.0
/usr/lib/aarch64-linux-gnu/libavdevice.so.60
/usr/lib/aarch64-linux-gnu/libavc1394.so.0.3.0
/usr/lib/aarch64-linux-gnu/libavfilter.so.9.12.100
/usr/lib/aarch64-linux-gnu/libavahi-client.so.3
/usr/lib/aarch64-linux-gnu/libavtp.so.0
/usr/lib/aarch64-linux-gnu/libavformat.so.60.16.100
/usr/lib/aarch64-linux-gnu/libavcodec.so.60
/usr/lib/aarch64-linux-gnu/libavahi-glib.so.1.0.2
/usr/lib/aarch64-linux-gnu/libavfilter.so.9
/usr/lib/aarch64-linux-gnu/libavutil.so.58.29.100
/usr/lib/aarch64-linux-gnu/libavahi-ui-gtk3.so.0
/usr/lib/aarch64-linux-gnu/libavahi-ui-gtk3.so.0.1.4
/usr/lib/aarch64-linux-gnu/libavahi-common.so.3.5.4
...
```

but how can i remove it i never installed

nyanmisaka, 2024-12-15:
You can also run `ldd ./ffmpeg`.

Or run `./ffmpeg` with `LD_LIBRARY_PATH=./libavcodec ...`

PMGZED, 2024-12-15:
it goes forward

root@radxa-zero3:/ffmpeg-rockchip# LD_LIBRARY_PATH=./libavcodec ./ffmpeg   -hide_banner -f lavfi -i ```
testsrc2=duration=300:size=1920x1080:rate=30  -c:v h264_rkmpp  test.mp4
Input #0, lavfi, from 'testsrc2=duration=300:size=1920x1080:rate=30':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 30 tbn
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
[h264_rkmpp @ 0xaaaad32873b0] Failed to create a RKMPP hardware device: -12
[vost#0:0/h264_rkmpp @ 0xaaaad3287030] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
Error while filtering: Cannot allocate memory
[out#0/mp4 @ 0xaaaad3285d70] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=N/A bitrate=N/A speed=N/A
Conversion failed!

```

if i compiled it with --prefix=/usr and make install
should the new ffmpeg and avlib override the old version?

nyanmisaka, 2024-12-15:
> [h264_rkmpp @ 0xaaaad32873b0] Failed to create a RKMPP hardware device: -12

https://github.com/nyanmisaka/ffmpeg-rockchip#important

> should the new ffmpeg and avlib override the old version?

It will, but I'm not sure where your old libavcodec came from, and whether they will come back again via package updates.

PMGZED, 2024-12-15:
Ok i test 
very big thanks nyanmisaka  :-)

PMGZED, 2024-12-15:
Ok libavutil58 was installed after remove it works.
dont know why its not been removed with apt remove libav*
again big thanks :-)

closed 2024-12-15 by PMGZED