nyanmisaka/ffmpeg-rockchip #148: Ask for support to create a obs ecndoer

PMGZED, 2024-12-21:
Hi, you helped me to got running this ffmpeg on my radxa zero 3w
now i try to create a obs encoder. Obs work fine with the panfrost driver for send the stream obs need a encoder

i have copy the nvenc encode and change to use the rkmpp eincoder but if the stream try to init the encoder it failed.

the cops ffmpeg-video-encoder calls the avcodec_open2 function. this function return -542398533 but i don know why.
maybe you have a idea?
i have already start a thread in obs forum and discord https://obsproject.com/forum/threads/is-there-a-esay-way-to-add-ffmpeg-with-a-videocodec-as-a-streaming-encoder.182707/ 

the call of avcidec_open2 are in https://github.com/obsproject/obs-studio/blob/master/plugins/obs-ffmpeg/obs-ffmpeg-video-encoders.c line 13

PMGZED, 2024-12-21:
i think the parameter are wrong. can u tell me what the minimum parameter require the encoder need?

PMGZED, 2024-12-21:
Ok i had not permissions vor the rkmpp devices. (how can i set the device rights permanent after every restart i need to change with chmod ? )
[image]
so now it dont got a error but the fps brake down to 1 fps but dont know why

nyanmisaka, 2024-12-22:
> set the device rights permanent

https://jellyfin.org/docs/general/administration/hardware-acceleration/rockchip/#configure-on-linux-host

> but the fps brake down to 1 fps but dont know why

I also have no experience with ffmpeg-rockchip in obs-studio. You can turn on ffmpeg logging and track what is causing the 1fps.

PMGZED, 2024-12-24:
i have switch to the rock 5b with more power.
if i use the record it works but not if i want to use my stream encoder
[image]
[image]

maybe i forgot to set some information?
	int profile =0;
	int bitrate = 5000;
	int keyint_sec = 250;
	enc->ffve.context->height = 1080;
	enc->ffve.context->width = 1920;
	enc->ffve.context->framerate.num = 30;
	enc->ffve.context->time_base.num = 1;
	enc->ffve.context->pix_fmt = AV_PIX_FMT_NV12;
		info("settings:\n"
	     "\tencoder:      %s\n"
	     "\trate_control: %i\n"
	     "\tbitrate:      %d\n"
	     "\tkeyint:       %d\n"
	     "\tprofile:      %i\n"
	     "\twidth:        %d\n"
	     "\theight:       %d\n"
	     "\tb-frames:     %d\n"
	     "\tFramerate:    %d\n"
	     "\tTimebase:     %d\n"
	     "\tFormat:       %d\n",
	     enc->ffve.enc_name, rcmode, bitrate, enc->ffve.context->gop_size, profile, enc->ffve.context->width,
	     enc->ffve.context->height, enc->ffve.context->max_b_frames, enc->ffve.context->framerate.num,
	     enc->ffve.context->time_base.num, enc->ffve.context->pix_fmt);

nyanmisaka, 2024-12-24:
There are no special parameters that need to be set. Therefore I think the error is in obs-ffmpeg rather than ffmpeg-rockchip.

PMGZED, 2025-01-03:
basically it work now 😊
(but i have a high cpu usage. hope i can findout why)

closed 2025-01-03 by PMGZED