nyanmisaka/ffmpeg-rockchip #150: error: ‘AV_HWDEVICE_TYPE_RKMPP’ was not declared in this scope

AshleyRoth, 2024-12-28:
Hello!

I got error:
error: ‘AV_HWDEVICE_TYPE_RKMPP’ was not declared in this scope; did you mean ‘AV_HWDEVICE_TYPE_DRM’?
  164 |     if (av_hwdevice_ctx_create(&hw_device_ctx, AV_HWDEVICE_TYPE_RKMPP, nullptr, nullptr, 0) < 0)

why? ffmpeg configure and installed with rkmpp.

my code:
AVBufferRef *hw_device_ctx = nullptr;
    if (av_hwdevice_ctx_create(&hw_device_ctx, AV_HWDEVICE_TYPE_RKMPP, nullptr, nullptr, 0) < 0)
    {
        std::cerr << "Error: Could not create hardware device context for RKMpp" << std::endl;
        std::cerr << "Ensure that Rockchip MPP is installed and configured correctly." << std::endl;
        return -1;
    }

nyanmisaka, 2024-12-29:
https://github.com/nyanmisaka/ffmpeg-rockchip/blob/b81c3bf1c40bbc34de3950451287fd4f5213bdd3/libavutil/hwcontext.h#L40

You have to include `ffmpeg-rockchip/libavutil/hwcontext.h`

closed 2024-12-29 by nyanmisaka

AshleyRoth, 2024-12-29:
@nyanmisaka I can't make a working code for capturing rtsp video stream using ffmpeg-rockchip, can you help? I got error "std::cerr << "Error: No video stream found" << std::endl;"

const char* input_url = argv[1];

    avformat_network_init();

    AVFormatContext* format_ctx = nullptr;
    if (avformat_open_input(&format_ctx, input_url, nullptr, nullptr) != 0) {
        std::cerr << "Error: Could not open input " << input_url << std::endl;
        return -1;
    }

    if (avformat_find_stream_info(format_ctx, nullptr) < 0) {
        std::cerr << "Error: Could not retrieve stream info" << std::endl;
        avformat_close_input(&format_ctx);
        return -1;
    }

    int video_stream_index = -1;
    for (unsigned int i = 0; i < format_ctx->nb_streams; i++) {
        if (format_ctx->streams[i]->codecpar->codec_type == AVMEDIA_TYPE_VIDEO) {
            video_stream_index = i;
            break;
        }
    }
    if (video_stream_index == -1) {
        std::cerr << "Error: No video stream found" << std::endl;
        avformat_close_input(&format_ctx);
        return -1;
    }

    AVCodecParameters* codec_par = format_ctx->streams[video_stream_index]->codecpar;
    const AVCodec* codec = avcodec_find_decoder(codec_par->codec_id);
    if (!codec) {
        std::cerr << "Error: Unsupported codec" << std::endl;
        avformat_close_input(&format_ctx);
        return -1;
    }