nyanmisaka/ffmpeg-rockchip #152: 关于bgr3支持4k 60fps的问题

huochen1, 2025-01-08:
您好请问一下，就是我这边根据这个命令./ffmpeg -f v4l2 -i /dev/video0 -c:v h264_rkmpp -flags +low_delay -b:v 4M -qp_init 26 -minrate
2M -maxrate 5M -profile:v high -level 4.1 -g:v 100 -vframes 1200 -y bgr3_4k60fps_264.mp4  测试了一下，我的输入源是4k 60fps的bgr3源，发现1200帧平均下来的帧率是54帧（保存下来的视频可以使用ffmpeg查看属性的），而且这个保存下来的视频播放会随着时间的推移越来越卡顿，可能后面都没有54fps了。但是如果是NV12的颜色输入的话，就不会有卡顿的现象，而且也是60fps来的。请问一下这个我要怎么去优化这个以支持BGR3的颜色输入。

nyanmisaka, 2025-01-08:
Remove `-flags +low_delay` and try again?

This flag enfores single-core encoder and disable async encoding to reduce latency.

closed 2025-01-12 by nyanmisaka

huochen1, 2025-01-14:
不要意思现在才看到，根据您的提示输入的还是bgr3 4k 60fps的，我使用了./ffmpeg -f v4l2 -i /dev/video0 -c:v h264_rkmpp -qp_init 20 -profile:v main   -vframes 2400 -y bgr3_4k_60fps_264.mp4来进行编码，发现编码的速度是60fps的，我的/dev/video0 设备输入就是60fps，保存下来的视频文件也是60fps的，但是播放的时候缺发现会有明显的卡顿现象，这个是rga的问题吗？

huochen1, 2025-01-14:
同样的我也测试了一下NV16格式的，发现也不会卡顿，bgr3的4k 30fps不会卡顿，但是4k 60fps会卡顿。

nyanmisaka, 2025-01-14:
Please upload a video showing the lag.

huochen1, 2025-01-14:
https://github.com/user-attachments/assets/3530ea56-5fcc-4301-a6c1-cd3d449ee17f

上面的是有延迟的，下面这个是没有延迟

https://github.com/user-attachments/assets/674e1de7-c843-4bd1-bc3f-11ede1fab04b

的

huochen1, 2025-01-14:
是卡顿

nyanmisaka, 2025-01-14:
```
 V....D libx264              libx264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (codec h264)
 V....D libx264rgb           libx264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 RGB (codec h264)
```

Have you ever tried using software encoder `-c:v libx264rgb` to rule out the problem?

nyanmisaka, 2025-01-14:
If the software encoder is also causing the lagging video, then check your video input.

sathishNexion, 2025-01-14:
[image]
[image]
[image]

I exactly Followed this --> https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation
But i encountered some errors, Friend

I am using rv1126

huochen1, 2025-01-15:
输入源我可以排除掉了，因为我试过很多输入源，都是存在这个问题，我猜想是不是由于BGR3的数据量比NV12大很多，编码处理不过来导致的，虽然他结果也是显示4K 60FPS，就还是会显示卡顿

sathishNexion, 2025-01-15:
Okay

huochen1, 2025-01-15:
> Okay
> […](#)

我这个不是回复您的，我是回复作者的

huochen1, 2025-01-15:
> If the software encoder is also causing the lagging video, then check your video input.
不好意思又打扰您了，就是我现在使用这个命令，小球的速度是可以跟上原始视频的了，但是还是存在一个问题就是，播放的时候看到小球跟不上预测的小球行走的路线的，导致小球边缘会有卡顿的现象，请问您这个需要怎么优化这个编码的参数，我尝试了很多参数的选择和修改最终达到的效果都是跟这个视频播放的差不多的
./ffmpeg -f  v4l2 -i /dev/video0 -c:v h264_rkmpp -b:v 15M -minrate 10M -maxrate 20M -profile:v high  -g:v 60 -vframes 2400 -y bgr3_4k_60fps.mp4

https://github.com/user-attachments/assets/9e78022b-f0fa-4af6-97de-dfdf1b0d2fe2

nyanmisaka, 2025-01-17:
Make sure you used the MPP library from https://github.com/nyanmisaka/mpp/tree/jellyfin-mpp

The upstream mpp/develop tree is always broken.

huochen1, 2025-01-17:
> Make sure you used the MPP library from https://github.com/nyanmisaka/mpp/tree/jellyfin-mpp
> 
> The upstream mpp/develop tree is always broken.

是的，我还再重新编译了一下还是这样的效果，我使用的是Orange Pi 5 Plus的rk3588，HDMI IN的进入来测试的。电脑作为输入的源。我换了一个播放的视频更加明显，好像是丢帧的现象
测试命令：./ffmpeg -f v4l2 -i /dev/video0 -c:v hevc_rkmpp -qp_init 20 -profile:v main   -vframes 2400 -y bgr3_4k_60fps_264.mp4
https://github.com/user-attachments/assets/e920af5d-901b-4f9c-a87a-bd5835b7899a

huochen1, 2025-01-17:
[image]
我发现重复帧过多，在2400帧下的重复帧达到了差不多1/3，可能是这个原因

nyanmisaka, 2025-01-17:
The actual input frame rate on the encoder side is only about ~47fps, which should explain why the lagging is observed. And hdmirx-related cma alloc failures are observed in `dmesg`, so this is not an encoder error.

```
[ 1008.921085] cma: cma_alloc: reserved: alloc failed, req-size: 6075 pages, ret: -12
[ 1008.921113] rk_hdmirx fdee0000.hdmirx-controller: dma alloc of size 24883200 failed
```

```
# v4l2-ctl -d /dev/video0 -V -D
Driver Info:
        Driver name      : rk_hdmirx
        Card type        : rk_hdmirx
        Bus info         : fdee0000.hdmirx-controller
        Driver version   : 6.1.84
        Capabilities     : 0x84201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format
Format Video Capture Multiplanar:
        Width/Height      : 3840/2160
        Pixel Format      : 'BGR3' (24-bit BGR 8-8-8)
        Field             : None
        Number of planes  : 1
        Flags             : 0x00000080
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr/HSV Encoding: Unknown (0x000000ff)
        Quantization      : Full Range
        Plane 0           :
           Bytes per Line : 11520
           Size Image     : 24883200
```

```
# v4l2-ctl --set-edid type=hdmi-4k-600mhz,audio,ycbcr444,ycbcr422,fl-fr

# ./ffmpeg -f v4l2 -i /dev/video0 -f null -
ffmpeg version dc1e883921 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --disable-shared --enable-static --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-libx264 --enable-libx265 --enable-libdav1d
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 1306.802138, bitrate: 11930001 kb/s
  Stream #0:0: Video: rawvideo (BGR[24] / 0x18524742), bgr24, 3840x2160, 11930001 kb/s, 59.93 fps, 59.93 tbr, 1000k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf61.7.100
  Stream #0:0: Video: wrapped_avframe, bgr24(pc, gbr/unknown/unknown, progressive), 3840x2160, q=2-31, 200 kb/s, 59.93 fps, 59.93 tbn
      Metadata:
        encoder         : Lavc61.19.101 wrapped_avframe
frame= 2110 fps= 47 q=-0.0 size=N/A time=00:00:44.66 bitrate=N/A speed=0.992x
```

huochen1, 2025-01-17:
我这边特意找了一个电脑可以支持HDMI IN输入的4k 60fps的，使用这个命令可以查看当前HDMI IN的帧率，查询帧率： v4l2-ctl -d /dev/video0 --set-fmt-video=width=3840,height=2160, --stream-mmap=3 --stream-skip=1 --stream-poll，

[image]  可以看到他是60fps的，并且我们可以进入sudo systemctl set-default multi-user 下执行gst-launch-1.0 v4l2src device=/dev/video0 ! videoconvert ! autovideosink sync=false命令可以查看HDMI IN的实时情况，可以判断输入源卡不卡顿。我是查看了这个播放的视频没有滞后的现象的。不过请问作者这个v4l2-ctl --set-edid type=hdmi-4k-600mhz 这个命令还可以禁止BGR3颜色的吗，我查看HDMI IN的话是支持4种颜色格式的YUV420 ,YUV422,YUV444,以及BGR3的。

huochen1, 2025-01-17:
还有一个发现就是使用potplayer播放器来查看这个滞后的视频，一帧一帧的查看就是会有每四帧会有一个重叠帧，这个重叠帧是不是有可能造成这种视觉上的滞后，我也尝试去调整这几个参数，发现并没有很大的改善。

[image]

nyanmisaka, 2025-01-17:
Your gstreamer command should be zero-copy, no copying from hdmirx->display. FFmpeg's libavdevice/v4l2 does not have this optimization.

The chroma subsample of the video encoded by the rkmpp encoder is 4:2:0, so BGR3/BGR24 is not very useful here. Setting the EDID to only support 4k60 ycbcr420 should be able to disable BGR3/BGR24.

huochen1, 2025-01-18:
感谢您的回复，我这边尝试了不知道是不是设置的方式不对，好像v4l2-ctl --set-edid 的方式我看了只能将NV24禁掉，对于BGR3还是无能为力。想请问作者有什么建议可以将HMDI IN的这个BGR3给禁止掉。