nyanmisaka/ffmpeg-rockchip #155: problems with ffpmeg in frigate 0.15

Vladd1980, 2025-01-27:
I'm using frigate 0.13 on orange pi5. it has been working stably for a long time. But for a long time now I have been trying to switch to frigate 0.15 and I have a problem with two cameras. in frigate 0.13 they work like this: 
go2rtc:
  streams:
    Facade3: 
      ffmpeg:rtsp://192.168.201.13:8554/profile0#video=h265/rkmpp#audio=copy#audio=aac#audio=opus
# - rtsp://admin:GKVn@192.168.200.11:30060/Fasad3
    Fasad3_sub: 
      ffmpeg:rtsp://192.168.201.13:8554/profile1#video=h265/rkmpp#audio=copy#audio=aac#audio=opus

but in frigate 0.15 it doesn't work. Only works:
Facade3: 
      ffmpeg:rtsp://192.168.201.13:8554/profile0#video=h264#audio=copy#audio=aac#audio=opus
which leads to excessive processor load. I also tried other options, all this is described in my topic here:
https://github.com/blakeblackshear/frigate/discussions/12713#discussioncomment-11947403
MarcA711 advised me to write about it here. Please help!

MarcA711, 2025-01-27:
Hey, additional contex:
He tries to convert the hevc rtsp stream from his camera to h264 using this command:
```
ffmpeg -hide_banner -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel -hwaccel_output_format drm_prime -afbc rga -i rtsp://192.168.201.13:8554/profile0 -c:v h264_rkmpp -g 50 -bf 0 -an -rtsp_transport tcp -f rtsp {{output}}
```

However, this results in the following error:
```
[rtsp] error=streams: exec/rtsp
[hevc @ 0x559254e990] SPS 0 does not exist.
Last message repeated 1 times
[rtsp @ 0x5592541250] max delay reached. need to consume packet
[rtsp @ 0x5592541250] RTP: missed 2 packets
Input #0, rtsp, from 'rtsp://192.168.201.13:8554/profile1':
Metadata:
title: session
Duration: N/A, start: 0.000000, bitrate: N/A
Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 960x480, 15 fps, 29.92 tbr, 90k tbn
Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, s16, 64 kb/s
```

Apperently, it was working with hbiyik's fork https://github.com/hbiyik/FFmpeg.
dmesg output can be found [here](https://github.com/blakeblackshear/frigate/discussions/12713#discussioncomment-11970433).

nyanmisaka, 2025-01-27:
The logs are not helpful. Add `-v trace` to ffmpeg command to get a more detailed log.

Vladd1980, 2025-01-27:
> Логи бесполезны. Добавьте `-v trace`в команду ffmpeg, чтобы получить более подробный лог.

Good afternoon Unfortunately my knowledge is not great. If you write in detail what needs to be done, I will provide all the necessary information

MarcA711, 2025-01-27:
Run these commands on your OPi5 (not in frigate):
```
mkdir ~/ffmpeg_test
cd ~/ffmpeg_test
wget https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/download/6.1-7/ffmpeg
chmod +x ffmpeg
sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel -hwaccel_output_format drm_prime -afbc rga -i rtsp://192.168.201.13:8554/profile0 -c:v h264_rkmpp -g 50 -bf 0 -an test.mp4 -y
```

Vladd1980, 2025-01-27:
> Выполните эти команды на вашем OPi5 (не на фрегате):
> 
> ```
> mkdir ~/ffmpeg_test
> cd ~/ffmpeg_test
> wget https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/download/6.1-7/ffmpeg
> chmod +x ffmpeg
> sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel -hwaccel_output_format drm_prime -afbc rga -i rtsp://192.168.201.13:8554/profile0 -c:v h264_rkmpp -g 50 -bf 0 -an test.mp4 -y
> ```

root@orangepi5:~/ffmpeg_test# sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel -hwaccel_output_format drm_prime -afbc rga -i rtsp://192.168.201.13:8554/profile0 -c:v h264_rkmpp -g 50 -bf 0 -an test.mp4 -y
ffmpeg version b81c3bf1c4-20250117 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 14.2.0 (crosstool-NG 1.26.0.120_4d36f27)
  configuration: --prefix=/ffbuild/prefix --pkg-config-flags=--static --pkg-config=pkg-config --cross-prefix=aarch64-ffbuild-linux-gnu- --arch=aarch64 --target-os=linux --enable-gpl --enable-version3 --disable-debug --enable-iconv --enable-zlib --enable-libfreetype --enable-libfribidi --enable-gmp --enable-libxml2 --enable-openssl --enable-lzma --enable-fontconfig --enable-libharfbuzz --enable-libvorbis --enable-opencl --enable-libpulse --enable-libvmaf --enable-libxcb --enable-xlib --enable-amf --enable-libaom --enable-libaribb24 --enable-avisynth --enable-chromaprint --enable-libdav1d --disable-libdavs2 --disable-libfdk-aac --enable-ffnvcodec --enable-cuda-llvm --enable-frei0r --enable-libgme --enable-libkvazaar --enable-libaribcaption --enable-libass --enable-libbluray --enable-libjxl --enable-libmp3lame --enable-libopus --enable-librist --enable-libssh --enable-libtheora --enable-libvpx --enable-libwebp --enable-libzmq --enable-lv2 --disable-libvpl --enable-openal --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-librav1e --enable-rkmpp --enable-rkrga --enable-librubberband --disable-schannel --enable-sdl2 --enable-libsnappy --enable-libsoxr --enable-libsrt --enable-libsvtav1 --enable-libtwolame --enable-libuavs3d --enable-libdrm --disable-vaapi --enable-libvidstab --enable-vulkan --enable-libshaderc --enable-libplacebo --enable-libx264 --enable-libx265 --disable-libxavs2 --enable-libxvid --enable-libzimg --enable-libzvbi --extra-cflags=-DLIBTWOLAME_STATIC --extra-cxxflags= --extra-libs='-ldl -lstdc++ -lstdc++ -lgomp' --extra-ldflags=-pthread --extra-ldexeflags=-pie --cc=aarch64-ffbuild-linux-gnu-gcc --cxx=aarch64-ffbuild-linux-gnu-g++ --ar=aarch64-ffbuild-linux-gnu-gcc-ar --ranlib=aarch64-ffbuild-linux-gnu-gcc-ranlib --nm=aarch64-ffbuild-linux-gnu-gcc-nm --extra-version=20250117
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Splitting the commandline.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'trace'.
Reading option '-allowed_media_types' ... matched as AVOption 'allowed_media_types' with argument 'video+audio'.
Reading option '-fflags' ... matched as AVOption 'fflags' with argument 'nobuffer'.
Reading option '-flags' ... matched as AVOption 'flags' with argument 'low_delay'.
Reading option '-timeout' ... matched as AVOption 'timeout' with argument '5000000'.
Reading option '-rtsp_flags' ... matched as AVOption 'rtsp_flags' with argument 'prefer_tcp'.
Reading option '-hwaccel' ... matched as option 'hwaccel' (use HW accelerated decoding) with argument '-hwaccel_output_format'.
Reading option 'drm_prime' ... matched as output url.
Reading option '-afbc' ... matched as AVOption 'afbc' with argument 'rga'.
Reading option '-i' ... matched as output url with argument 'rtsp://192.168.201.13:8554/profile0'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'h264_rkmpp'.
Reading option '-g' ... matched as AVOption 'g' with argument '50'.
Reading option '-bf' ... matched as AVOption 'bf' with argument '0'.
Reading option '-an' ... matched as option 'an' (disable audio) with argument '1'.
Reading option 'test.mp4' ... matched as output url.
Reading option '-y' ... matched as option 'y' (overwrite output files) with argument '1'.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option v (set logging level) with argument trace.
Applying option y (overwrite output files) with argument 1.
Successfully parsed a group of options.
Parsing a group of options: input url rtsp://192.168.201.13:8554/profile0.
Successfully parsed a group of options.
Opening an input file: rtsp://192.168.201.13:8554/profile0.
Probing rtsp score:100 size:0
[tcp @ 0xaaaade909860] No default whitelist set
[tcp @ 0xaaaade909860] Original list of addresses:
[tcp @ 0xaaaade909860] Address 192.168.201.13 port 8554
[tcp @ 0xaaaade909860] Interleaved list of addresses:
[tcp @ 0xaaaade909860] Address 192.168.201.13 port 8554
[tcp @ 0xaaaade909860] Starting connection attempt to 192.168.201.13 port 8554
[tcp @ 0xaaaade909860] Successfully connected to 192.168.201.13 port 8554
[rtsp @ 0xaaaade9091e0] Sending:
OPTIONS rtsp://192.168.201.13:8554/profile0 RTSP/1.0
CSeq: 1
User-Agent: Lavf60.16.100

--
[... cut ...]
Successfully opened the file.
Parsing a group of options: output url drm_prime.
Option hwaccel (use HW accelerated decoding) cannot be applied to output url drm_prime -- you are trying to apply an input option to an output file or vice versa. Move this option before the file it belongs to.
Error parsing options for output file drm_prime.
Error opening output files: Invalid argument
[rtsp @ 0xaaaade9091e0] Sending:
TEARDOWN rtsp://192.168.201.13:8554/profile0/ RTSP/1.0
CSeq: 6
User-Agent: Lavf60.16.100
Session: 1210792470

nyanmisaka, 2025-01-27:
That command is not correct.
```
[in#0/rtsp @ 0xaaaade9090e0] Codec AVOption afbc (Enable AFBC (Arm Frame Buffer Compression) to save bandwidth) has not been used for any stream. The most likely reason is either wrong type (e.g. a video option with no video streams) or that it is a private option of some decoder which was not actually used for any stream.
Successfully opened the file.
Parsing a group of options: output url drm_prime.
Option hwaccel (use HW accelerated decoding) cannot be applied to output url drm_prime -- you are trying to apply an input option to an output file or vice versa. Move this option before the file it belongs to.
Error parsing options for output file drm_prime.
Error opening output files: Invalid argument
```

Vladd1980, 2025-01-27:
> Эта команда неверна.
> 
> ```
> [in#0/rtsp @ 0xaaaade9090e0] Codec AVOption afbc (Enable AFBC (Arm Frame Buffer Compression) to save bandwidth) has not been used for any stream. The most likely reason is either wrong type (e.g. a video option with no video streams) or that it is a private option of some decoder which was not actually used for any stream.
> Successfully opened the file.
> Parsing a group of options: output url drm_prime.
> Option hwaccel (use HW accelerated decoding) cannot be applied to output url drm_prime -- you are trying to apply an input option to an output file or vice versa. Move this option before the file it belongs to.
> Error parsing options for output file drm_prime.
> Error opening output files: Invalid argument
> ```

Is there anything else I can do?

nyanmisaka, 2025-01-27:
Try this
```
sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -i rtsp://192.168.201.13:8554/profile0 -f null -
```

Vladd1980, 2025-01-27:
> Попробуйте это
> 
> ```
> sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -i rtsp://192.168.201.13:8554/profile0 -f null -
> ```
sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -i rtsp://192.168.201.13:8554/profile0 -f null -
ffmpeg version b81c3bf1c4-20250117 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 14.2.0 (crosstool-NG 1.26.0.120_4d36f27)
  configuration: --prefix=/ffbuild/prefix --pkg-config-flags=--static --pkg-config=pkg-config --cross-prefix=aarch64-ffbuild-linux-gnu- --arch=aarch64 --target-os=linux --enable-gpl --enable-version3 --disable-debug --enable-iconv --enable-zlib --enable-libfreetype --enable-libfribidi --enable-gmp --enable-libxml2 --enable-openssl --enable-lzma --enable-fontconfig --enable-libharfbuzz --enable-libvorbis --enable-opencl --enable-libpulse --enable-libvmaf --enable-libxcb --enable-xlib --enable-amf --enable-libaom --enable-libaribb24 --enable-avisynth --enable-chromaprint --enable-libdav1d --disable-libdavs2 --disable-libfdk-aac --enable-ffnvcodec --enable-cuda-llvm --enable-frei0r --enable-libgme --enable-libkvazaar --enable-libaribcaption --enable-libass --enable-libbluray --enable-libjxl --enable-libmp3lame --enable-libopus --enable-librist --enable-libssh --enable-libtheora --enable-libvpx --enable-libwebp --enable-libzmq --enable-lv2 --disable-libvpl --enable-openal --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-librav1e --enable-rkmpp --enable-rkrga --enable-librubberband --disable-schannel --enable-sdl2 --enable-libsnappy --enable-libsoxr --enable-libsrt --enable-libsvtav1 --enable-libtwolame --enable-libuavs3d --enable-libdrm --disable-vaapi --enable-libvidstab --enable-vulkan --enable-libshaderc --enable-libplacebo --enable-libx264 --enable-libx265 --disable-libxavs2 --enable-libxvid --enable-libzimg --enable-libzvbi --extra-cflags=-DLIBTWOLAME_STATIC --extra-cxxflags= --extra-libs='-ldl -lstdc++ -lstdc++ -lgomp' --extra-ldflags=-pthread --extra-ldexeflags=-pie --cc=aarch64-ffbuild-linux-gnu-gcc --cxx=aarch64-ffbuild-linux-gnu-g++ --ar=aarch64-ffbuild-linux-gnu-gcc-ar --ranlib=aarch64-ffbuild-linux-gnu-gcc-ranlib --nm=aarch64-ffbuild-linux-gnu-gcc-nm --extra-version=20250117
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Splitting the commandline.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'trace'.
Reading option '-allowed_media_types' ... matched as AVOption 'allowed_media_types' with argument 'video+audio'.
Reading option '-fflags' ... matched as AVOption 'fflags' with argument 'nobuffer'.
Reading option '-flags' ... matched as AVOption 'flags' with argument 'low_delay'.
Reading option '-timeout' ... matched as AVOption 'timeout' with argument '5000000'.
Reading option '-rtsp_flags' ... matched as AVOption 'rtsp_flags' with argument 'prefer_tcp'.
Reading option '-hwaccel' ... matched as option 'hwaccel' (use HW accelerated decoding) with argument 'rkmpp'.
Reading option '-hwaccel_output_format' ... matched as option 'hwaccel_output_format' (select output format used with HW accelerated decoding) with argument 'drm_prime'.
Reading option '-i' ... matched as output url with argument 'rtsp://192.168.201.13:8554/profile0'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'null'.
Reading option '-' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option v (set logging level) with argument trace.
Successfully parsed a group of options.
Parsing a group of options: input url rtsp://192.168.201.13:8554/profile0.
Applying option hwaccel (use HW accelerated decoding) with argument rkmpp.
Applying option hwaccel_output_format (select output format used with HW accelerated decoding) with argument drm_prime.
Successfully parsed a group of options.
Opening an input file: rtsp://192.168.201.13:8554/profile0.
Probing rtsp score:100 size:0
[tcp @ 0xaaaacd6529b0] No default whitelist set
[tcp @ 0xaaaacd6529b0] Original list of addresses:
[tcp @ 0xaaaacd6529b0] Address 192.168.201.13 port 8554
[tcp @ 0xaaaacd6529b0] Interleaved list of addresses:
[tcp @ 0xaaaacd6529b0] Address 192.168.201.13 port 8554
[tcp @ 0xaaaacd6529b0] Starting connection attempt to 192.168.201.13 port 8554
[tcp @ 0xaaaacd6529b0] Successfully connected to 192.168.201.13 port 8554
[rtsp @ 0xaaaacd652090] Sending:
OPTIONS rtsp://192.168.201.13:8554/profile0 RTSP/1.0
CSeq: 1
User-Agent: Lavf60.16.100

--
[rtsp @ 0xaaaacd652090] ret=1 c=52 [R]
[rtsp @ 0xaaaacd652090] ret=1 c=54 [T]
[rtsp @ 0xaaaacd652090] ret=1 c=53 [S]
[rtsp @ 0xaaaacd652090] ret=1 c=50 [P]
[rtsp @ 0xaaaacd652090] ret=1 c=2f [/]
[rtsp @ 0xaaaacd652090] ret=1 c=31 [1]
[rtsp @ 0xaaaacd652090] ret=1 c=2e [.]
[rtsp @ 0xaaaacd652090] ret=1 c=30 [0]
[rtsp @ 0xaaaacd652090] ret=1 c=20 [ ]
[rtsp @ 0xaaaacd652090] ret=1 c=32 [2]
[rtsp @ 0xaaaacd652090] ret=1 c=30 [0]
[... cut ...]
[in#0/rtsp @ 0xaaaacd651f70]   Input stream #0:0 (video): 202 packets read (192340 bytes); 0 frames decoded; 0 decode errors; 
[in#0/rtsp @ 0xaaaacd651f70]   Input stream #0:1 (audio): 427 packets read (136640 bytes); 426 frames decoded; 0 decode errors (136320 samples); 
[in#0/rtsp @ 0xaaaacd651f70]   Total: 629 packets (328980 bytes) demuxed
[rtsp @ 0xaaaacd652090] Sending:
TEARDOWN rtsp://192.168.201.13:8554/profile0/ RTSP/1.0
CSeq: 6
User-Agent: Lavf60.16.100
Session: 1851587828

--
Exiting normally, received signal 2.
root@orangepi5:~/ffmpeg_test#

nyanmisaka, 2025-01-27:
Recompile ffmpeg with `sed -i "s/,dump_extra//g" libavcodec/rkmppdec.c` and try again.

Vladd1980, 2025-01-27:
> sed -i "s/,dump_extra//g" libavcodec/rkmppdec.c

root@orangepi5:~/ffmpeg_test# sed -i "s/,dump_extra//g" libavcodec/rkmppdec.c
sed: can't read libavcodec/rkmppdec.c: No such file or directory
sorry, I don't know how to do this

Vladd1980, 2025-01-27:
> [@Vladd1980](https://github.com/Vladd1980)Я перекомпилирую FFmpeg, но это займет пару часов. Я напишу вам, как только у меня будет готова новая версия.

Thank you!

MarcA711, 2025-01-27:
Hey, it's finished:

```
cd ~/ffmpeg_test
rm ffmpeg
wget https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/download/6.1-7-no_dump_extra/ffmpeg
chmod +x ffmpeg
sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -i rtsp://192.168.201.13:8554/profile0 -f null -
```

Vladd1980, 2025-01-28:
> Эй, готово:
> 
> ```
> cd ~/ffmpeg_test
> rm ffmpeg
> wget https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/download/6.1-7-no_dump_extra/ffmpeg
> chmod +x ffmpeg
> sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -i rtsp://192.168.201.13:8554/profile0 -f null -
> ```

root@orangepi5:~/ffmpeg_test# sudo ./ffmpeg -v trace -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -i rtsp://192.168.201.13:8554/profile0 -f null -
ffmpeg version 4b0e666711-20250127 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 14.2.0 (crosstool-NG 1.26.0.120_4d36f27)
  configuration: --prefix=/ffbuild/prefix --pkg-config-flags=--static --pkg-config=pkg-config --cross-prefix=aarch64-ffbuild-linux-gnu- --arch=aarch64 --target-os=linux --enable-gpl --enable-version3 --disable-debug --enable-iconv --enable-zlib --enable-libfreetype --enable-libfribidi --enable-gmp --enable-libxml2 --enable-openssl --enable-lzma --enable-fontconfig --enable-libharfbuzz --enable-libvorbis --enable-opencl --enable-libpulse --enable-libvmaf --enable-libxcb --enable-xlib --enable-amf --enable-libaom --enable-libaribb24 --enable-avisynth --enable-chromaprint --enable-libdav1d --disable-libdavs2 --disable-libfdk-aac --enable-ffnvcodec --enable-cuda-llvm --enable-frei0r --enable-libgme --enable-libkvazaar --enable-libaribcaption --enable-libass --enable-libbluray --enable-libjxl --enable-libmp3lame --enable-libopus --enable-librist --enable-libssh --enable-libtheora --enable-libvpx --enable-libwebp --enable-libzmq --enable-lv2 --disable-libvpl --enable-openal --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-librav1e --enable-rkmpp --enable-rkrga --enable-librubberband --disable-schannel --enable-sdl2 --enable-libsnappy --enable-libsoxr --enable-libsrt --enable-libsvtav1 --enable-libtwolame --enable-libuavs3d --enable-libdrm --disable-vaapi --enable-libvidstab --enable-vulkan --enable-libshaderc --enable-libplacebo --enable-libx264 --enable-libx265 --disable-libxavs2 --enable-libxvid --enable-libzimg --enable-libzvbi --extra-cflags=-DLIBTWOLAME_STATIC --extra-cxxflags= --extra-libs='-ldl -lstdc++ -lstdc++ -lgomp' --extra-ldflags=-pthread --extra-ldexeflags=-pie --cc=aarch64-ffbuild-linux-gnu-gcc --cxx=aarch64-ffbuild-linux-gnu-g++ --ar=aarch64-ffbuild-linux-gnu-gcc-ar --ranlib=aarch64-ffbuild-linux-gnu-gcc-ranlib --nm=aarch64-ffbuild-linux-gnu-gcc-nm --extra-version=20250127
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Splitting the commandline.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'trace'.
Reading option '-allowed_media_types' ... matched as AVOption 'allowed_media_types' with argument 'video+audio'.
Reading option '-fflags' ... matched as AVOption 'fflags' with argument 'nobuffer'.
Reading option '-flags' ... matched as AVOption 'flags' with argument 'low_delay'.
Reading option '-timeout' ... matched as AVOption 'timeout' with argument '5000000'.
Reading option '-rtsp_flags' ... matched as AVOption 'rtsp_flags' with argument 'prefer_tcp'.
Reading option '-hwaccel' ... matched as option 'hwaccel' (use HW accelerated decoding) with argument 'rkmpp'.
Reading option '-hwaccel_output_format' ... matched as option 'hwaccel_output_format' (select output format used with HW accelerated decoding) with argument 'drm_prime'.
Reading option '-i' ... matched as output url with argument 'rtsp://192.168.201.13:8554/profile0'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'null'.
Reading option '-' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option v (set logging level) with argument trace.
Successfully parsed a group of options.
Parsing a group of options: input url rtsp://192.168.201.13:8554/profile0.
Applying option hwaccel (use HW accelerated decoding) with argument rkmpp.
Applying option hwaccel_output_format (select output format used with HW accelerated decoding) with argument drm_prime.
Successfully parsed a group of options.
Opening an input file: rtsp://192.168.201.13:8554/profile0.
Probing rtsp score:100 size:0
[tcp @ 0xaaab1c9ba9b0] No default whitelist set
[tcp @ 0xaaab1c9ba9b0] Original list of addresses:
[tcp @ 0xaaab1c9ba9b0] Address 192.168.201.13 port 8554
[tcp @ 0xaaab1c9ba9b0] Interleaved list of addresses:
[tcp @ 0xaaab1c9ba9b0] Address 192.168.201.13 port 8554
[tcp @ 0xaaab1c9ba9b0] Starting connection attempt to 192.168.201.13 port 8554
[tcp @ 0xaaab1c9ba9b0] Successfully connected to 192.168.201.13 port 8554
[rtsp @ 0xaaab1c9ba090] Sending:
OPTIONS rtsp://192.168.201.13:8554/profile0 RTSP/1.0
CSeq: 1
User-Agent: Lavf60.16.100

--
[rtsp @ 0xaaab1c9ba090] ret=1 c=52 [R]
[rtsp @ 0xaaab1c9ba090] ret=1 c=54 [T]
[rtsp @ 0xaaab1c9ba090] ret=1 c=53 [S]
[rtsp @ 0xaaab1c9ba090] ret=1 c=50 [P]
[rtsp @ 0xaaab1c9ba090] ret=1 c=2f [/]
[rtsp @ 0xaaab1c9ba090] ret=1 c=31 [1]
[rtsp @ 0xaaab1c9ba090] ret=1 c=2e [.]
[... cut ...]
[AVFilterGraph @ 0xaaab1c9ba680] query_formats: 4 queried, 6 merged, 3 already done, 0 delayed
[out#0/null @ 0xaaab1c9cd260] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=N/A bitrate=N/A speed=N/A    
[rtsp @ 0xaaab1c9ba090] Sending:
TEARDOWN rtsp://192.168.201.13:8554/profile0/ RTSP/1.0
CSeq: 6
User-Agent: Lavf60.16.100
Session: 403644160

--
Exiting normally, received signal 2.
root@orangepi5:~/ffmpeg_test#

Vladd1980, 2025-01-28:
Can I hope for a solution to the problem in the near future? I haven't been able to run frigate 0.15 for several months now and it's causing me a lot of problems. I would really not like to change equipment as it is not so simple, expensive and until recently I was completely satisfied

nyanmisaka, 2025-02-08:
I tried all locally accessible streams and was unable to reproduce. If someone could provide a publicly available RTSP stream that can be used to reproduce the problem, that would be very helpful to this issue.

Vladd1980, 2025-02-08:
> Я перепробовал все локально доступные потоки и не смог воспроизвести. Если бы кто-то мог предоставить общедоступный поток RTSP, который можно использовать для воспроизведения проблемы, это было бы очень полезно для решения этой проблемы.

Hello! thank you for your experiences :). I can give you access to the stream. I'll need a little time. where can I send you the link? I don't want to share it publicly

nyanmisaka, 2025-02-08:
> Hello! thank you for your experiences :). I can give you access to the stream. I'll need a little time. where can I send you the link? I don't want to share it publicly

You can send one to "nst799610810 at gmail dot com"

Vladd1980, 2025-02-09:
> > > Я перепробовал все локально доступные потоки и не смог воспроизвести. Если бы кто-то мог рассчитать общедоступный поток RTSP, который можно использовать для обнаружения проблем, это было бы очень полезно для решения этих проблем.
> > 
> > 
> > Привет! Спасибо за ваш опыт :). Я могу дать вам доступ к трансляции. Мне нужно немного времени. Куда я могу отправить вам ссылку? Я не хочу делиться ею публично
> 
> Вы можете отправить его на адрес "nst799610810 at gmail dot com"

I sent the stream address to your email address

nyanmisaka, 2025-02-10:
The first link you provided returned a 404 error.
```
[rtsp @ 0xaaaad54a9630] method DESCRIBE failed: 404 Not Found
[rtsp @ 0xaaaad54a9630] Cseq: 2

[in#0 @ 0xaaaad54a9440] Error opening input: Server returned 404 Not Found
Error opening input file rtsp://<IP_ADDR>:8554/profile0.
Error opening input files: Server returned 404 Not Found
```

Can both links be used to reproduce the problem?

nyanmisaka, 2025-02-10:
Can you fix this link `rtsp://<IP_ADDR>:8554/profile0`? So that I can verify that my fix worked.

nyanmisaka, 2025-02-10:
> Recompile ffmpeg with `sed -i "s/,dump_extra//g" libavcodec/rkmppdec.c` and try again.

I guess @MarcA711 may have mistakenly uploaded an unpatched ffmpeg which caused some misleading. Because when I manually modified and compiled ffmpeg locally, it can decode the rtsp stream you provided normally. But using [`6.1-7-no_dump_extra`](https://github.com/MarcA711/Rockchip-FFmpeg-Builds/tree/6.1-7-no_dump_extra) it still cannot decode.

```
~/workspace/ffmpeg-rockchip$ ./ffmpeg -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -fast_parse 1 -i rtsp://<IP_ADDR>/profile0 -an -sn -dn -f null -

ffmpeg version n7.1-196-gee68fb8521 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.3.0-6ubuntu2~24.04)
  configuration: --disable-ffplay --disable-doc --disable-shared --enable-gpl --enable-version3 --enable-static --enable-gnutls --enable-opencl --enable-libdrm --enable-libdav1d --enable-libx264 --enable-libx265 --enable-rkmpp --enable-rkrga
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
[hevc @ 0xaaaade282980] SPS 0 does not exist.
    Last message repeated 1 times
Input #0, rtsp, from 'rtsp://<IP_ADDR>/profile0':
  Metadata:
    title           : session
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 2560x1440, 12 fps, 12 tbr, 90k tbn
  Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, s16, 64 kb/s
rga_api version 1.10.1_[3]
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    title           : session
    encoder         : Lavf61.7.100
  Stream #0:0: Video: wrapped_avframe, drm_prime(pc, bt709, progressive), 2560x1440, q=2-31, 200 kb/s, 12 fps, 12 tbn
      Metadata:
        encoder         : Lavc61.19.101 wrapped_avframe
frame= 1099 fps= 10 q=-0.0 size=N/A time=00:01:39.91 bitrate=N/A speed=0.937x
```

Vladd1980, 2025-02-10:
> > Перекомпилируйте ffmpeg `sed -i "s/,dump_extra//g" libavcodec/rkmppdec.c`и попробуйте еще раз.
> 
> Наверное[@MarcA711](https://github.com/MarcA711)возможно, ошибочно загрузил непропатченный ffmpeg, что вызвало некоторое введение в заблуждение. Потому что когда я вручную модифицировал и скомпилировал ffmpeg локально, он может декодировать rtsp-поток, который вы предоставили обычным образом. Но при [`6.1-7-no_dump_extra`](https://github.com/MarcA711/Rockchip-FFmpeg-Builds/tree/6.1-7-no_dump_extra)его использовании декодирование все равно не удается.
> 
> ```
> ~/workspace/ffmpeg-rockchip$ ./ffmpeg -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -fast_parse 1 -i rtsp://<IP_ADDR>/profile0 -an -sn -dn -f null -
> 
> ffmpeg version n7.1-196-gee68fb8521 Copyright (c) 2000-2024 the FFmpeg developers
>   built with gcc 13 (Ubuntu 13.3.0-6ubuntu2~24.04)
>   configuration: --disable-ffplay --disable-doc --disable-shared --enable-gpl --enable-version3 --enable-static --enable-gnutls --enable-opencl --enable-libdrm --enable-libdav1d --enable-libx264 --enable-libx265 --enable-rkmpp --enable-rkrga
>   libavutil      59. 39.100 / 59. 39.100
>   libavcodec     61. 19.101 / 61. 19.101
>   libavformat    61.  7.100 / 61.  7.100
>   libavdevice    61.  3.100 / 61.  3.100
>   libavfilter    10.  4.100 / 10.  4.100
>   libswscale      8.  3.100 /  8.  3.100
>   libswresample   5.  3.100 /  5.  3.100
>   libpostproc    58.  3.100 / 58.  3.100
> [hevc @ 0xaaaade282980] SPS 0 does not exist.
>     Last message repeated 1 times
> Input #0, rtsp, from 'rtsp://<IP_ADDR>/profile0':
>   Metadata:
>     title           : session
>   Duration: N/A, start: 0.000000, bitrate: N/A
>   Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 2560x1440, 12 fps, 12 tbr, 90k tbn
>   Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, s16, 64 kb/s
> rga_api version 1.10.1_[3]
> Stream mapping:
>   Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> wrapped_avframe (native))
> Press [q] to stop, [?] for help
> Output #0, null, to 'pipe:':
>   Metadata:
>     title           : session
>     encoder         : Lavf61.7.100
>   Stream #0:0: Video: wrapped_avframe, drm_prime(pc, bt709, progressive), 2560x1440, q=2-31, 200 kb/s, 12 fps, 12 tbn
>       Metadata:
>         encoder         : Lavc61.19.101 wrapped_avframe
> frame= 1099 fps= 10 q=-0.0 size=N/A time=00:01:39.91 bitrate=N/A speed=0.937x
> ```

I'm glad that we managed to localize the problem. I hope Mark fixes it soon! Are threads no longer needed? can I close them?

nyanmisaka, 2025-02-10:
```
When decode a live video, sequence header is passed through sdp and video stream may not have sequence header. In this case qsv_decoder cannot decode header and keep waiting for more data.

For example:
push stream:
ffmpeg -v verbose -re -stream_loop -1 -i input.mp4 -c:v hevc_qsv
-bsf:v extract_extradata=remove=1 -f rtsp
rtsp://localhost:8554/mystream

receive stream:
ffmpeg -v verbose -hwaccel rkmpp -i rtsp://localhost:8554/mystream
-f null -

Header data is stored in extradata, so add "dump_extra" bsf into rkmppdec, and then rkmpp decoder works on video stream without sequence header.
```

It has nothing to do with performance. The ffmpeg fix https://github.com/nyanmisaka/ffmpeg-rockchip/commit/04f5eaa11c713577e41f201552ff8183af6508bf has been pushed to the repo, you can try again and then we will close the issue.

If anyone encounters incompatible rtsp streams after this change, please update to FFmpeg 7.0+ branch and let the user add `-bsf:v dump_extra` before `-i INPUT`. This option is user specific, it will not apply to everyone's rtsp server.

MarcA711, 2025-02-10:
Thank you very much for the explanation.

@Vladd1980 Please test this release: https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/download/6.1-8-no_extra_dump/ffmpeg

MarcA711, 2025-02-11:
Download the binary using:
```
wget https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/download/6.1-8-no_extra_dump/ffmpeg
chmod +x ffmpeg
```

and mount it to your docker container by adjusting this line and adding it to your compose.yaml:
```
volumes:
    ...
    - /path/to/new/ffmpeg:/usr/lib/ffmpeg/6.0/bin/ffmpeg
```

In the next official frigate release (0.15.1 or 0.16.0) I will probably include the fixed version. You can remove the mount and the ffmpeg binary in that case.

Vladd1980, 2025-02-11:
> Загрузите двоичный файл с помощью:
> 
> ```
> wget https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/download/6.1-8-no_extra_dump/ffmpeg
> chmod +x ffmpeg
> ```
> 
> и смонтируйте его в свой Docker-контейнер, изменив эту строку и добавив ее в свой compose.yaml:
> 
> ```
> volumes:
>     ...
>     - /path/to/new/ffmpeg:/usr/lib/ffmpeg/6.0/bin/ffmpeg
> ```
> 
> В следующем официальном релизе frigate (0.15.1 или 0.16.0) я, вероятно, включу исправленную версию. В этом случае вы можете удалить монтирование и бинарник ffmpeg.

So. I tested ffmpeg above. The configuration from 0.13 ffmpeg:rtsp://192.168.201.13:8554/profile0#video=h265#audio=copy#audio=aac#audio=opus still results in maximum CPU load. the ffmpeg:rtsp://192.168.201.13:8554/profile0#video=h264#audio=copy#audio=aac#audio=opus configuration works, the CPU load is normal, but the stream is unstable. every 10 seconds the discharge stops and passes after 5-7 seconds

MarcA711, 2025-02-11:
You will still need to manually call FFmpeg. The built in presets for hardware acceleration on rockchip boards from go2rtc is broken after all and it's unlikely that this will get fixed.

Try something like this in `go2rtc: streams:`:
```
Fasad3: exec:ffmpeg -hide_banner -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i rtsp://192.168.201.13:8554/profile0 -c:v h264_rkmpp -g 50 -bf 0 -an -rtsp_transport tcp -f rtsp {{output}}
```

Vladd1980, 2025-02-11:
> Вам все равно придется вручную вызывать FFmpeg. Встроенные пресеты для аппаратного ускорения на платах Rockchip от go2rtc все-таки сломаны, и маловероятно, что это будет исправлено.
> 
> Попробуйте что-то вроде этого `go2rtc: streams:`:
> 
> ```
> Fasad3: exec:ffmpeg -hide_banner -allowed_media_types video+audio -fflags nobuffer -flags low_delay -timeout 5000000 -rtsp_flags prefer_tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i rtsp://192.168.201.13:8554/profile0 -c:v h264_rkmpp -g 50 -bf 0 -an -rtsp_transport tcp -f rtsp {{output}}
> ```

It's better that way. But in essence the situation is like this. Detection appears, preview appears. But in 90% of cases there is no recording. When trying to export, it says that there are no files to export. At the same time, when I scroll through the scale in history, I see these objects, that is, it will turn out to be

MarcA711, 2025-02-11:
Since this is probably a configuration issue, we will discuss this here https://github.com/blakeblackshear/frigate/discussions/12713

closed 2025-02-14 by nyanmisaka