nyanmisaka/ffmpeg-rockchip #156: 在正点原子ATK-RK3588上编译项目完成后使用时无法初始化mpp

QBHCWQ, 2025-02-07:
lrl@ATK-DLRK3588-Ubuntu:~/dev$ ffmpeg -f lavfi -i testsrc2=s=1920x1080,format=nv12 -c:v hevc_rkmpp -qp_init 26 -profile:v main -level 4.1 -g:v 100 -vframes 5000 -y /tmp/tmp.mp4
ffmpeg version 4b0e666 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, lavfi, from 'testsrc2=s=1920x1080,format=nv12':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, nv12, 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 25 tbn
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> hevc (hevc_rkmpp))
Press [q] to stop, [?] for help
[hevc_rkmpp @ 0x55afa51160] Failed to init MPP context: -1
[vost#0:0/hevc_rkmpp @ 0x55afa50e00] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
Error while filtering: Generic error in an external library
[out#0/mp4 @ 0x55afa4fa90] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=N/A bitrate=N/A speed=N/A    
Conversion failed!

QBHCWQ, 2025-02-07:
开发板上我移植了ubuntu 版本如下：
lrl@ATK-DLRK3588-Ubuntu:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal

QBHCWQ, 2025-02-07:
具体参考链接：
https://blog.csdn.net/Leon_Chenl/article/details/144839093#comments_36171084

QBHCWQ, 2025-02-07:
编译mpp过程如下：
lrl@ATK-DLRK3588-Ubuntu:~/dev$ cd rkmpp
lrl@ATK-DLRK3588-Ubuntu:~/dev/rkmpp$ ls
CHANGELOG.md  CMakeLists.txt  LICENSES  build  debian  doc  inc  merge_static_lib.sh  mpp  osal  pkgconfig  readme.txt  rkmpp_build  test  tools  utils
lrl@ATK-DLRK3588-Ubuntu:~/dev/rkmpp$ vi merge_static_lib.sh 
lrl@ATK-DLRK3588-Ubuntu:~/dev/rkmpp$ cd rkmpp_build/
lrl@ATK-DLRK3588-Ubuntu:~/dev/rkmpp/rkmpp_build$ ls
CMakeCache.txt  CMakeFiles  Makefile  cmake_install.cmake  compile_commands.json  install_manifest.txt  mpp  osal  rockchip_mpp.pc  rockchip_vpu.pc  test  utils
lrl@ATK-DLRK3588-Ubuntu:~/dev/rkmpp/rkmpp_build$ cmake \
>     -DCMAKE_INSTALL_PREFIX=/usr \
>     -DCMAKE_BUILD_TYPE=Release \
>     -DBUILD_SHARED_LIBS=ON \
>     -DBUILD_TEST=OFF \
>     ..
-- cmake version 3.16.3
-- Detected ARMv8 system processor
-- current version:
-- fda5a02 author: nyanmisaka     2025-01-17 Revert "chore[cmake]: compile both dynamic and static libraries"
-- git version history:
-- fda5a02 author: nyanmisaka     2025-01-17 Revert \"chore[cmake]: compile both dynamic and static libraries\"  (grafted, HEAD -> jellyfin-mpp, origin/jellyfin-mpp)
-- total 1 git version recorded
-- Install git hooks done
-- Checking for one of the modules 'pthread'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lrl/dev/rkmpp/rkmpp_build

lrl@ATK-DLRK3588-Ubuntu:~/dev/rkmpp/rkmpp_build$ make
[ 11%] Built target osal
[ 18%] Built target mpp_base
[ 21%] Built target utils
[ 22%] Built target hal_av1d_com
[ 23%] Built target mpp_rc
[ 25%] Built target vproc_iep2
[ 25%] Built target codec_dummy_dec
[ 26%] Built target dec_common
[ 27%] Built target codec_avsd
[ 28%] Built target codec_avs2d
[ 29%] Built target codec_h263d
[ 32%] Built target codec_h264d
[ 34%] Built target codec_h265d
[ 34%] Built target codec_mpeg2d
[ 35%] Built target codec_mpeg4d
[ 36%] Built target codec_vp8d
[ 37%] Built target codec_vp9d
[ 38%] Built target codec_jpegd
[ 40%] Built target codec_av1d
[ 41%] Built target codec_dummy_enc
[ 45%] Built target enc_rc
[ 47%] Built target codec_h264e
[ 48%] Built target codec_jpege
[ 50%] Built target codec_h265e
[ 51%] Built target codec_vp8e
[ 52%] Built target vproc_rga
[ 52%] Built target vproc_iep
[ 53%] Built target mpp_vproc
[ 55%] Built target mpp_codec
[ 55%] Built target hal_vepu_common
[ 56%] Built target hal_h263d
[ 57%] Built target hal_mpeg2d
[ 59%] Built target hal_mpeg4d
[ 60%] Built target hal_vp8d
[ 62%] Built target hal_jpegd
[ 63%] Built target hal_av1d_vpu
[ 64%] Built target hal_jpege_vpu
[ 65%] Built target hal_h264e_com
[ 67%] Built target hal_h264e_vpu
[ 70%] Built target hal_vp8e
[ 71%] Built target vdpu383_com
[ 72%] Built target vdpu34x_com
[... 110 lines cut ...]
-- Up-to-date: /usr/include/rockchip/vpu_api.h
-- Up-to-date: /usr/include/rockchip/mpp_rc_api.h
-- Up-to-date: /usr/include/rockchip/rk_venc_rc.h
-- Up-to-date: /usr/include/rockchip/rk_mpi.h
-- Up-to-date: /usr/include/rockchip/mpp_err.h
-- Up-to-date: /usr/include/rockchip/mpp_log_def.h
-- Up-to-date: /usr/include/rockchip/mpp_frame.h
-- Up-to-date: /usr/include/rockchip/mpp_log.h
-- Up-to-date: /usr/include/rockchip/vpu.h
-- Up-to-date: /usr/include/rockchip/mpp_meta.h
-- Up-to-date: /usr/include/rockchip/rk_vdec_cmd.h
-- Up-to-date: /usr/include/rockchip/rk_venc_cmd.h
-- Up-to-date: /usr/include/rockchip/mpp_packet.h
-- Up-to-date: /usr/include/rockchip/rk_hdr_meta_com.h
-- Up-to-date: /usr/include/rockchip/rk_venc_cfg.h
-- Up-to-date: /usr/include/rockchip/mpp_buffer.h
-- Up-to-date: /usr/include/rockchip/rk_venc_ref.h
-- Up-to-date: /usr/lib/aarch64-linux-gnu/pkgconfig/rockchip_mpp.pc
-- Up-to-date: /usr/lib/aarch64-linux-gnu/pkgconfig/rockchip_vpu.pc
-- Up-to-date: /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.0
-- Up-to-date: /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
-- Up-to-date: /usr/lib/aarch64-linux-gnu/librockchip_mpp.so
-- Up-to-date: /usr/lib/aarch64-linux-gnu/librockchip_vpu.so.0
-- Up-to-date: /usr/lib/aarch64-linux-gnu/librockchip_vpu.so.1
-- Up-to-date: /usr/lib/aarch64-linux-gnu/librockchip_vpu.so

nyanmisaka, 2025-02-07:
> Failed to init MPP context: -1

https://github.com/nyanmisaka/ffmpeg-rockchip#important

## Important
* User MUST be granted permission to access these device files.
```
# DRM allocator
/dev/dri

# DMA_HEAP allocator
/dev/dma_heap

# RGA filters
/dev/rga

# MPP codecs
/dev/mpp_service

closed 2025-02-07 by nyanmisaka