nyanmisaka/ffmpeg-rockchip #157: Ffmpeg stops to work with hdmi input

ErezAlster, 2025-02-09:
Hi All

I compiled ffmpeg (thanks for the great and easy documentation).
I'm working with orange pi 5 ultra (which has Rockchip 3588).

I'm able to record a video but every several seconds ffmpeg stops with the following error:

this is the ffmpeg exec:

> /home/orangepi/dev/ffmpeg/ffmpeg -f v4l2 -i /dev/video0 -vf "fps=30" -s 1920x1080 -c:v h264_rkmpp -rc_mode AVBR -b:v 6M -minrate 4M -maxrate 8M -profile:v main -level 5.1 -f rtsp rtsp://localhost:8554/starium
ffmpeg version ea9247a Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
[video4linux2,v4l2 @ 0x55b1c8c650] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x55b1c8c650] ioctl(VIDIOC_G_PARM): Inappropriate ioctl for device, estimated framerate 30/1 from dv timings.
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 5218.560637, bitrate: 995328 kb/s
  Stream #0:0: Video: rawvideo, nv16, 1920x1080, 995328 kb/s, 30 fps, 30 tbr, 1000k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
Output #0, rtsp, to 'rtsp://localhost:8554/starium':
  Metadata:
    encoder         : Lavf60.16.100
  Stream #0:0: Video: h264 (Main), nv16(progressive), 1920x1080, q=2-31, 6000 kb/s, 30 fps, 90k tbn
    Metadata:
      encoder         : Lavc60.31.102 h264_rkmpp
[video4linux2,v4l2 @ 0x55b1c8c650] ioctl(VIDIOC_DQBUF): Invalid argument.952x    
[in#0/video4linux2,v4l2 @ 0x55b1c8c530] Error during demuxing: Invalid argument
[in#0/video4linux2,v4l2 @ 0x55b1c8c530] Error retrieving a packet from demuxer: Invalid argument
[out#0/rtsp @ 0x55b1c8e400] video:1252kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
frame=   55 fps= 30 q=-0.0 Lsize=N/A time=00:00:01.80 bitrate=N/A speed=0.982x  

v4l2-ctl -d /dev/video0 -D
Driver Info:
        Driver name      : rk_hdmirx
        Card type        : rk_hdmirx
        Bus info         : fdee0000.hdmirx-controller
        Driver version   : 6.1.43
        Capabilities     : 0x84201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format

v4l2-ctl -d /dev/video0 --get-fmt-video
Format Video Capture Multiplanar:
        Width/Height      : 1920/1080
        Pixel Format      : 'NV16' (Y/UV 4:2:2)
        Field             : None
        Number of planes  : 1
        Flags             : premultiplied-alpha, 0x000000fe
        Colorspace        : Default
        Transfer Function : Unknown (0x000000b8)
        YCbCr/HSV Encoding: xvYCC 601
        Quantization      : Limited Range
        Plane 0           :
           Bytes per Line : 1920
           Size Image     : 4147200

v4l2-ctl -d /dev/video0 --get-dv-timings
DV timings:
        Active width: 1920
        Active height: 1080
        Total width: 2200
        Total height: 1125
        Frame format: progressive
        Polarities: -vsync -hsync
        Pixelclock: 74248000 Hz (30.00 frames per second)
        Horizontal frontporch: 88
        Horizontal sync: 44
        Horizontal backporch: 148
        Vertical frontporch: 4
        Vertical sync: 5
        Vertical backporch: 36
        Standards: 
        Flags: 

In the dmesg i'm getting the following error:

> [ 4866.179834] fdee0000.hdmirx-controller: Vertical Sync threshold reached interrupt 0x2
[ 4866.260829] fdee0000.hdmirx-controller: hdmirx_wait_lock_and_get_timing signal lock ok, i:23!
[ 4866.897145] rk_hdmirx fdee0000.hdmirx-controller: hdmirx_delayed_work_audio: enable audio
[ 4870.556651] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x32315559, cur_fmt:0x3631564e!
[ 4870.556667] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x32315559, cur_fmt:0x3631564e!
[ 4870.556669] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x32315659, cur_fmt:0x3631564e!
[ 4870.556672] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x50323234, cur_fmt:0x3631564e!
[ 4870.556674] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x56595559, cur_fmt:0x3631564e!
[ 4870.556676] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x59565955, cur_fmt:0x3631564e!
[ 4870.556679] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x50313134, cur_fmt:0x3631564e!
[ 4870.556681] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x39565559, cur_fmt:0x3631564e!
[ 4870.556683] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x39555659, cur_fmt:0x3631564e!
[ 4870.556684] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x4f424752, cur_fmt:0x3631564e!
[ 4870.556686] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x51424752, cur_fmt:0x3631564e!
[ 4870.556688] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x50424752, cur_fmt:0x3631564e!
[ 4870.556690] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x52424752, cur_fmt:0x3631564e!
[ 4870.556693] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x33524742, cur_fmt:0x3631564e!
[ 4870.556695] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x33424752, cur_fmt:0x3631564e!
[ 4870.556697] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x34325258, cur_fmt:0x3631564e!
[ 4870.556699] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x34325842, cur_fmt:0x3631564e!
[ 4870.556701] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x34325241, cur_fmt:0x3631564e!
[ 4870.556703] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x34324142, cur_fmt:0x3631564e!
[ 4870.556705] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x34524742, cur_fmt:0x3631564e!
[ 4870.556707] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x34424752, cur_fmt:0x3631564e!
[ 4870.556709] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x59455247, cur_fmt:0x3631564e!
[ 4870.556711] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x20363159, cur_fmt:0x3631564e!
[ 4870.556712] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x2036315a, cur_fmt:0x3631564e!
[ 4870.556714] fdee0000.hdmirx-controller: hdmirx_s_fmt_vid_cap_mplane: err, set_fmt:0x3432564e, cur_fmt:0x3631564e!
[ 4870.583506] fdee0000.hdmirx-controller: hdmirx_start_streaming: delay_line:720
[ 4871.779810] fdee0000.hdmirx-controller: rcv frames
[ 4876.512192] fdee0000.hdmirx-controller: stream start stopping
[ 4876.563926] fdee0000.hdmirx-controller: hdmirx_stop_streaming wait last irq timeout, return bufs!
[ 4876.563972] fdee0000.hdmirx-controller: stream stopping finished
[ 4877.580032] fdee0000.hdmirx-controller: Vertical Sync threshold reached interrupt 0x2
[ 4877.679342] fdee0000.hdmirx-controller: hdmirx_wait_lock_and_get_timing signal lock ok, i:23!
[ 4878.043928] rk_hdmirx fdee0000.hdmirx-controller: hdmirx_delayed_work_audio: enable audio
[ 4878.273579] fdee0000.hdmirx-controller: hdmirx_format_change: New format: 1920x1080p30.00 (2200x1125)
[ 4879.413396] fdee0000.hdmirx-controller: Vertical Sync threshold reached interrupt 0x2
[ 4879.514266] fdee0000.hdmirx-controller: hdmirx_wait_lock_and_get_timing signal lock ok, i:23!

Any reason what can be the problem?
Thanks
Erez

nyanmisaka, 2025-02-09:
This is a kernel issue rather than an ffmpeg issue. 6.1.43 is also an outdated rockchip kernel version.

Required patches for hdmirx kernel driver: https://github.com/armbian/linux-rockchip/commits/rk-6.1-rkr5/drivers/media/platform/rockchip/hdmirx

closed 2025-02-09 by nyanmisaka

ErezAlster, 2025-02-10:
Thanks a lot. I opened a ticket for them in their forums
https://forum.armbian.com/topic/49741-orange-pi-5-ultra-support/