nyanmisaka/ffmpeg-rockchip #160: Low starting frame rate when output contains audio and video streams

medvedev, 2025-02-19:
When the output contains both, audio and video streams, ffmpeg starts with a really low FPS (like 0.1) and slowly increases it over time.  
I have done multiple experiments locally. Video-only or audio-only output works as expected, but once I attempt to combine them - it takes minutes to get expected FPS.  
Attached is a sample log file from my attempt to publish a video stream to youtube.  
The outcome does not depend on the sink type. I.e. it behaves the same if I write streams to a local file.  
I experimented with different audio bitrates and sources (wav, mp3), with or without re-encoding audio stream - it's always the same.  
Log file: [ffmpeg-20250219-114738.log](https://github.com/user-attachments/files/18867464/ffmpeg-20250219-114738.log)

nyanmisaka, 2025-02-19:
I can't reproduce it.
```
./ffmpeg -thread_queue_size 64 -f lavfi -i "testsrc2=s=1920x1080,format=yuv420p" -f lavfi -i "anullsrc=r=44100:cl=stereo" -s 1280x720 -c:v h264_rkmpp -rc_mode AVBR -b:v 2M -minrate 1500k -maxrate 4M -r 25 -g 50 -pix_fmt yuv420p -c:a aac -b:a 64k -profile:v main -bufsize 512k -vf vflip -level 5.1 -vframes 5000 -v verbose -f flv -y /tmp/tmp.flv

ffmpeg version n7.1-203-gf5c7e0d63b Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.3.0-6ubuntu2~24.04)
  configuration: --disable-ffplay --disable-doc --disable-shared --enable-gpl --enable-version3 --enable-static --enable-gnutls --enable-opencl --enable-libdrm --enable-libdav1d --enable-libx264 --enable-libx265 --enable-rkmpp --enable-rkrga
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
[Parsed_testsrc2_0 @ 0xaaaacd1c6120] size:1920x1080 rate:25/1 duration:-1.000000 sar:1/1
Input #0, lavfi, from 'testsrc2=s=1920x1080,format=yuv420p':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, 1 reference frame, yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 25 tbn
Input #1, lavfi, from 'anullsrc=r=44100:cl=stereo':
  Duration: N/A, start: 0.000000, bitrate: 705 kb/s
  Stream #1:0: Audio: pcm_u8, 44100 Hz, stereo, u8, 705 kb/s
[out#0/flv @ 0xaaaacd1ccec0] No explicit maps, mapping streams automatically...
[vost#0:0/h264_rkmpp @ 0xaaaacd1cd6b0] Created video stream from input stream 0:0
[aost#0:1/aac @ 0xaaaacd1d02b0] Created audio stream from input stream 1:0
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> h264 (h264_rkmpp))
  Stream #1:0 -> #0:1 (pcm_u8 (native) -> aac (native))
[vost#0:0/h264_rkmpp @ 0xaaaacd1cd6b0] Starting thread...
[aost#0:1/aac @ 0xaaaacd1d02b0] Starting thread...
[vf#0:0 @ 0xaaaacd1ce1e0] Starting thread...
[af#0:1 @ 0xaaaacd1cd9e0] Starting thread...
[vist#0:0/wrapped_avframe @ 0xaaaacd1c8ce0] [dec:wrapped_avframe @ 0xaaaacd1ced80] Starting thread...
[aist#1:0/pcm_u8 @ 0xaaaacd1ccb80] [dec:pcm_u8 @ 0xaaaacd1d2090] Starting thread...
[in#0/lavfi @ 0xaaaacd1c4a80] Starting thread...
[in#1/lavfi @ 0xaaaacd1c8e60] Starting thread...
Press [q] to stop, [?] for help
[graph_-1_in_1:0 @ 0xffff88002450] tb:1/44100 samplefmt:u8 samplerate:44100 chlayout:stereo
[graph -1 input from stream 0:0 @ 0xffff940024d0] w:1920 h:1080 pixfmt:yuv420p tb:1/25 fr:25/1 sar:1/1 csp:unknown range:unknown
[scaler_out_#0:0 @ 0xffff94002e40] w:1280 h:720 flags:'' interl:0
[format_out_#0:1 @ 0xffff88002c50] auto-inserting filter 'auto_aresample_0' between the filter 'Parsed_anull_0' and the filter 'format_out_#0:1'
[auto_aresample_0 @ 0xffff88003850] ch:2 chl:stereo fmt:u8 r:44100Hz -> ch:2 chl:stereo fmt:fltp r:44100Hz
[scaler_out_#0:0 @ 0xffff94002e40] w:1920 h:1080 fmt:yuv420p csp:unknown range:unknown sar:1/1 -> w:1280 h:720 fmt:yuv420p csp:unknown range:tv sar:1/1 flags:0x00000004
[scaler_out_#0:0 @ 0xffff94002e40] [framesync @ 0xffff94002f18] Selected 1/25 time base
[scaler_out_#0:0 @ 0xffff94002e40] [framesync @ 0xffff94002f18] Sync level 1
[graph -1 input from stream 0:0 @ 0xffff940024d0] video frame properties congruent with link at pts_time: 0
[h264_rkmpp @ 0xaaaacd1c6c90] Rate Control mode is set to AVBR
[h264_rkmpp @ 0xaaaacd1c6c90] Bitrate Target/Min/Max is set to 2000000/1500000/4000000
[h264_rkmpp @ 0xaaaacd1c6c90] QP Init/Max/Min/Max_I/Min_I is set to 26/48/0/48/0
[h264_rkmpp @ 0xaaaacd1c6c90] Profile is set to MAIN
[h264_rkmpp @ 0xaaaacd1c6c90] Level is set to 51
[h264_rkmpp @ 0xaaaacd1c6c90] Coder is set to CABAC
[h264_rkmpp @ 0xaaaacd1c6c90] Created a RKMPP hardware device
Output #0, flv, to '/tmp/tmp.flv':
  Metadata:
    encoder         : Lavf61.7.100
  Stream #0:0: Video: h264 (Main), 1 reference frame ([7][0][0][0] / 0x0007), yuv420p(tv, progressive), 1280x720 [SAR 1:1 DAR 16:9], q=2-31, 2000 kb/s, 25 fps, 1k tbn
      Metadata:
        encoder         : Lavc61.19.101 h264_rkmpp
  Stream #0:1: Audio: aac (LC) ([10][0][0][0] / 0x000A), 44100 Hz, stereo, fltp, delay 1024, 64 kb/s
      Metadata:
        encoder         : Lavc61.19.101 aac
[out#0/flv @ 0xaaaacd1ccec0] Starting thread...
[h264_rkmpp @ 0xaaaacd1c6c90] Configured with size: 1280x720 | pix_fmt: yuv420p | sw_pix_fmt: yuv420p
[vf#0:0 @ 0xaaaacd1ce1e0] All consumers returned EOF03:15.84 bitrate=2505.8kbits/s speed=11.9x
[aost#0:1/aac @ 0xaaaacd1d02b0] Encoder thread received EOF
[vost#0:0/h264_rkmpp @ 0xaaaacd1cd6b0] Encoder thread received EOF
[vf#0:0 @ 0xaaaacd1ce1e0] Terminating thread with return code 0 (success)
[vist#0:0/wrapped_avframe @ 0xaaaacd1c8ce0] [dec:wrapped_avframe @ 0xaaaacd1ced80] Decoder returned EOF, finishing
[af#0:1 @ 0xaaaacd1cd9e0] All consumers returned EOF
[... 4 lines cut ...]
[aost#0:1/aac @ 0xaaaacd1d02b0] Terminating thread with return code 0 (success)
[af#0:1 @ 0xaaaacd1cd9e0] Terminating thread with return code 0 (success)
[aist#1:0/pcm_u8 @ 0xaaaacd1ccb80] [dec:pcm_u8 @ 0xaaaacd1d2090] Decoder returned EOF, finishing
[aist#1:0/pcm_u8 @ 0xaaaacd1ccb80] [dec:pcm_u8 @ 0xaaaacd1d2090] Terminating thread with return code 0 (success)
[aist#1:0/pcm_u8 @ 0xaaaacd1ccb80] All consumers of this stream are done
[in#1/lavfi @ 0xaaaacd1c8e60] All consumers are done
[... cut ...]
[in#0/lavfi @ 0xaaaacd1c4a80]   Input stream #0:0 (video): 5018 packets read (2207920 bytes); 5009 frames decoded; 0 decode errors;
[in#0/lavfi @ 0xaaaacd1c4a80]   Total: 5018 packets (2207920 bytes) demuxed
[in#1/lavfi @ 0xaaaacd1c8e60] Input file #1 (anullsrc=r=44100:cl=stereo):
[in#1/lavfi @ 0xaaaacd1c8e60]   Input stream #1:0 (audio): 9075 packets read (18585600 bytes); 9066 frames decoded; 0 decode errors (9283584 samples);
[in#1/lavfi @ 0xaaaacd1c8e60]   Total: 9075 packets (18585600 bytes) demuxed
```

nyanmisaka, 2025-02-19:
This repo is not touched for v4l2+camera use case in ffmpeg. You can download vanilla ffmpeg and use libx264 encoder (with superfast preset) for testing. I guess the problem will still exist and it is an upstream issue.

https://github.com/BtbN/FFmpeg-Builds/releases/tag/latest
https://github.com/BtbN/FFmpeg-Builds/releases/download/latest/ffmpeg-master-latest-linuxarm64-gpl.tar.xz

closed 2025-03-02 by nyanmisaka