nyanmisaka/ffmpeg-rockchip #164: `kmsgrab` with `-vf` Rockchip Filters

defencedog, 2025-02-28:
I am able to use `ffmpeg` for screen recording however I am not able to cook up a syntax where I can use `-vf vpp_rkmpp` with `-f kmsgrab` option

@nyanmisaka I have collected my thoughts here after reading your posts at RADXA, can you please review

> https://github.com/defencedog/orangepi3b_v2.1/blob/main/ffmpeg_hwaccel_screen_recording.md

>

nyanmisaka, 2025-03-02:
@defencedog Try this patch and see if it works.
Make sure you tell ffmpeg to use rkmpp as hw filter device: `ffmpeg -init_hw_device rkmpp=rk -filter_hw_device rk ...`

[0001-allow-rkrga-io-on-different-hwctx.patch](https://github.com/user-attachments/files/19042951/0001-allow-rkrga-io-on-different-hwctx.patch)

```diff
From a0e94dfe96bf21ad302d5795488f7e9a36c4c291 Mon Sep 17 00:00:00 2001
From: nyanmisaka <nst799610810@gmail.com>
Date: Sun, 2 Mar 2025 20:19:56 +0800
Subject: [PATCH] allow rkrga io on different hwctx

---
 libavfilter/rkrga_common.c     | 36 ++++++++++++++++++++++++++--------
 libavfilter/vf_overlay_rkrga.c |  1 +
 libavfilter/vf_vpp_rkrga.c     |  2 ++
 3 files changed, 31 insertions(+), 8 deletions(-)

diff --git a/libavfilter/rkrga_common.c b/libavfilter/rkrga_common.c
index 0019eb8114..7ca88ccae9 100644
--- a/libavfilter/rkrga_common.c
+++ b/libavfilter/rkrga_common.c
@@ -694,6 +694,7 @@ static av_cold int init_hwframes_ctx(AVFilterContext *avctx)
     AVHWFramesContext *hwfc_out;
     AVBufferRef       *hwfc_out_ref;
     AVHWDeviceContext *device_ctx;
+    AVBufferRef       *device_ref;
     AVRKMPPFramesContext *rkmpp_fc;
     int                ret;
 
@@ -701,12 +702,21 @@ static av_cold int init_hwframes_ctx(AVFilterContext *avctx)
         return AVERROR(EINVAL);
 
     hwfc_in = (AVHWFramesContext *)inlink->hw_frames_ctx->data;
-    device_ctx = (AVHWDeviceContext *)hwfc_in->device_ref->data;
+    device_ref = hwfc_in->device_ref;
+    device_ctx = (AVHWDeviceContext *)device_ref->data;
 
-    if (!device_ctx || device_ctx->type != AV_HWDEVICE_TYPE_RKMPP)
-        return AVERROR(EINVAL);
+    if (!device_ctx || device_ctx->type != AV_HWDEVICE_TYPE_RKMPP) {
+        if (avctx->hw_device_ctx) {
+            device_ref = avctx->hw_device_ctx;
+            device_ctx = (AVHWDeviceContext *)device_ref->data;
+        }
+        if (!device_ctx || device_ctx->type != AV_HWDEVICE_TYPE_RKMPP) {
+            av_log(avctx, AV_LOG_ERROR, "No RKMPP hardware context provided\n");
+            return AVERROR(EINVAL);
+        }
+    }
 
-    hwfc_out_ref = av_hwframe_ctx_alloc(hwfc_in->device_ref);
+    hwfc_out_ref = av_hwframe_ctx_alloc(device_ref);
     if (!hwfc_out_ref)
         return AVERROR(ENOMEM);
 
@@ -741,6 +751,7 @@ static av_cold int init_pat_preproc_hwframes_ctx(AVFilterContext *avctx)
     AVHWFramesContext *hwfc_pat;
     AVBufferRef       *hwfc_pat_ref;
     AVHWDeviceContext *device_ctx0;
+    AVBufferRef       *device_ref0;
     int                ret;
 
     if (!inlink0->hw_frames_ctx || !inlink1->hw_frames_ctx)
@@ -748,12 +759,21 @@ static av_cold int init_pat_preproc_hwframes_ctx(AVFilterContext *avctx)
 
     hwfc_in0 = (AVHWFramesContext *)inlink0->hw_frames_ctx->data;
     hwfc_in1 = (AVHWFramesContext *)inlink1->hw_frames_ctx->data;
-    device_ctx0 = (AVHWDeviceContext *)hwfc_in0->device_ref->data;
+    device_ref0 = hwfc_in0->device_ref;
+    device_ctx0 = (AVHWDeviceContext *)device_ref0->data;
 
-    if (!device_ctx0 || device_ctx0->type != AV_HWDEVICE_TYPE_RKMPP)
-        return AVERROR(EINVAL);
+    if (!device_ctx0 || device_ctx0->type != AV_HWDEVICE_TYPE_RKMPP) {
+        if (avctx->hw_device_ctx) {
+            device_ref0 = avctx->hw_device_ctx;
+            device_ctx0 = (AVHWDeviceContext *)device_ref0->data;
+        }
+        if (!device_ctx0 || device_ctx0->type != AV_HWDEVICE_TYPE_RKMPP) {
+            av_log(avctx, AV_LOG_ERROR, "No RKMPP hardware context provided\n");
+            return AVERROR(EINVAL);
+        }
+    }
 
-    hwfc_pat_ref = av_hwframe_ctx_alloc(hwfc_in0->device_ref);
+    hwfc_pat_ref = av_hwframe_ctx_alloc(device_ref0);
     if (!hwfc_pat_ref)
         return AVERROR(ENOMEM);
 
diff --git a/libavfilter/vf_overlay_rkrga.c b/libavfilter/vf_overlay_rkrga.c
index 14659a32cc..8f36af5cb2 100644
--- a/libavfilter/vf_overlay_rkrga.c
+++ b/libavfilter/vf_overlay_rkrga.c
@@ -366,4 +366,5 @@ const AVFilter ff_vf_overlay_rkrga = {
     FILTER_SINGLE_PIXFMT(AV_PIX_FMT_DRM_PRIME),
     .preinit        = rgaoverlay_framesync_preinit,
     .flags_internal = FF_FILTER_FLAG_HWFRAME_AWARE,
+    .flags          = AVFILTER_FLAG_HWDEVICE,
 };
diff --git a/libavfilter/vf_vpp_rkrga.c b/libavfilter/vf_vpp_rkrga.c
index b38226329a..9c6515876b 100644
--- a/libavfilter/vf_vpp_rkrga.c
+++ b/libavfilter/vf_vpp_rkrga.c
@@ -534,6 +534,7 @@ const AVFilter ff_vf_scale_rkrga = {
     FILTER_SINGLE_PIXFMT(AV_PIX_FMT_DRM_PRIME),
     .activate       = rgavpp_activate,
     .flags_internal = FF_FILTER_FLAG_HWFRAME_AWARE,
+    .flags          = AVFILTER_FLAG_HWDEVICE,
 };
 
 #endif
@@ -574,6 +575,7 @@ const AVFilter ff_vf_vpp_rkrga = {
     FILTER_SINGLE_PIXFMT(AV_PIX_FMT_DRM_PRIME),
     .activate       = rgavpp_activate,
     .flags_internal = FF_FILTER_FLAG_HWFRAME_AWARE,
+    .flags          = AVFILTER_FLAG_HWDEVICE,
 };
 
 #endif
-- 
2.43.0
```

nyanmisaka, 2025-04-22:
Close in https://github.com/nyanmisaka/ffmpeg-rockchip/commit/a7817eed0ba530ad724aee452949dd4b7ae03fc2