nyanmisaka/ffmpeg-rockchip #165: rk3588上运行4个转码线程，工作一分钟左右，程序异常退出

changliwu, 2025-03-05:
您好：
        rk3588上运行2个转码线程，工作正常，但是增加到4个转码，程序就会在decode时，异常退出，请问是什么原因？

Thread 17 "TranscodeD" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7cd7171ce0 (LWP 2086)]
0x0000007ff57f278c in mpp_mem_pool_get_f () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
(gdb) bt
#0  0x0000007ff57f278c in mpp_mem_pool_get_f () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#1  0x0000007ff56d180c in mpp_packet_new () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#2  0x0000007ff56d2a28 in mpp_packet_copy_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#3  0x0000007ff56a3bfc in Mpp::put_packet(void*) () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#4  0x0000007ff56a696c in mpi_decode_put_packet(void*, void*) () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#5  0x0000007ff74f6694 in ?? () from /lib/libavcodec.so.60
#6  0x0000007ff716aa1c in ?? () from /lib/libavcodec.so.60
#7  0x0000007ff716b0f8 in avcodec_send_packet () from /lib/libavcodec.so.60
#8  0x000000555555484c in CTranscode::decode (this=0x7fffffe2f0, pVideoPkt=0x7d64000b70, pFrame=0x7cc8000be0) at transcode.cpp:127
#9  0x0000005555555074 in CTranscode::RunDoThread (this=0x7fffffe2f0) at transcode.cpp:300
#10 0x00000055555545e4 in CTranscode::DoThread (pArg=0x7fffffe2f0) at transcode.cpp:86
#11 0x0000007ff5a8d5c8 in ?? () from /lib/aarch64-linux-gnu/libc.so.6
#12 0x0000007ff5af5edc in ?? () from /lib/aarch64-linux-gnu/libc.so.6

changliwu, 2025-03-05:
链接库版本信息如下：
pi@NanoPi-M6:~/RK3588S/transcode$ ldd TranscodeD
	linux-vdso.so.1 (0x0000007f9b83e000)
	libavcodec.so.60 => /lib/libavcodec.so.60 (0x0000007f9a720000)
	libavutil.so.58 => /lib/libavutil.so.58 (0x0000007f99640000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f99410000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f993e0000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f99230000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f9b805000)
	libswresample.so.4 => /lib/libswresample.so.4 (0x0000007f99200000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f99160000)
	librga.so.2 => /lib/aarch64-linux-gnu/librga.so.2 (0x0000007f99110000)
	liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007f990d0000)
	librockchip_mpp.so.1 => /lib/aarch64-linux-gnu/librockchip_mpp.so.1 (0x0000007f98e90000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f98e60000)
	libdrm.so.2 => /lib/aarch64-linux-gnu/libdrm.so.2 (0x0000007f98e30000)

nyanmisaka, 2025-03-05:
https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation

> Make sure you have built and installed the latest MPP runtime and headers. Otherwise undefined behavior may occur, such as segmentation fault. Be careful if you use MPP deb packages from a vendor/BSP image, these packages are likely to be out of date.

 `mpp_mem_pool_get_f()` is a func from MPP - https://github.com/rockchip-linux/mpp/blob/413e74bfe661c06078f48aae4e52e64d6e40773d/osal/mpp_mem_pool.cpp#L191

Make sure you built and installed MPP from 
```
git clone -b jellyfin-mpp --depth=1 https://github.com/nyanmisaka/mpp.git rkmpp
```
instead of the preinstalled MPP from the vendor.

changliwu, 2025-03-05:
我下载的是git clone -b jellyfin-mpp --depth=1 https://github.com/nyanmisaka/mpp.git rkmpp这个版本

请问ffmpeg-rockchip/doc/examples/中的transcode.c,vaapi_transcode.c,qsv_transcode.c 三个转码demo中，我在rk3588上运行应该参考哪个呢？

nyanmisaka, 2025-03-05:
Check dmesg and see why it crashed, mem pool errors usually mean out of memory or outdated kernel version (uname -a).

qsv_transcode is the most similar one, except for the codec suffix and pixel format.

changliwu, 2025-03-05:
qsv_transcode.c 中修改如下：
1）av_hwdevice_ctx_create(&hw_device_ctx, AV_HWDEVICE_TYPE_QSV, NULL, NULL, 0) 为av_hwdevice_ctx_create(&hw_device_ctx, AV_HWDEVICE_TYPE_RKMPP, NULL, NULL, 0);
2）encoder_ctx->pix_fmt   = AV_PIX_FMT_QSV为encoder_ctx->pix_fmt   = AV_PIX_FMT_NV12
3）decoder = avcodec_find_decoder_by_name("h264_qsv") 为decoder = avcodec_find_decoder_by_name("h264_rkmpp")

然后编译运行./qsv_transcode ../../../春天.mp4 h264_rkmpp 1.h264 0 出现如下错误：
Thread 1 "qsv_transcode" received signal SIGSEGV, Segmentation fault.
0x0000007ff5bd02bc in av_buffer_ref () from /lib/libavutil.so.58
(gdb) bt
#0  0x0000007ff5bd02bc in av_buffer_ref () from /lib/libavutil.so.58
#1  0x0000005555552528 in dec_enc (pkt=0x5555577370, enc_codec=0x7ff77a4520, optstr=0x7ffffff64f "0")
    at qsv_transcode.c:277
#2  0x0000005555552d40 in main (argc=5, argv=0x7ffffff328) at qsv_transcode.c:408

请问哪里修改有问题？

changliwu, 2025-03-05:
pi@NanoPi-M6:~$ uname -a
Linux NanoPi-M6 6.1.99 #32 SMP Mon Jan 20 15:50:32 CST 2025 aarch64 aarch64 aarch64 GNU/Linux

changliwu, 2025-03-06:
> Check dmesg and see why it crashed, mem pool errors usually mean out of memory or outdated kernel version (uname -a).
> 
> qsv_transcode is the most similar one, except for the codec suffix and pixel format.

dmesg 消息打印如下：
[60657.440714] rk_vcodec: mpp_wait_result_default:989: wait task break by signal

nyanmisaka, 2025-03-07:
There is no support for modifying `qsv_transcode`. You'll have to do it on your own.

Regarding decoding with ffmpeg-rockchip, I tested transcoding 16x 1080p videos simultaneously on RK3588 without any problems.

[image]

closed 2025-03-07 by nyanmisaka

changliwu, 2025-03-07:
谢谢你的回复，请问有没有demo提供一个呢?