nyanmisaka/ffmpeg-rockchip #167: error while Build MPP

sqf147, 2025-03-11:
`make[2]: *** [mpp/vproc/rga/test/CMakeFiles/rga_test.dir/build.make:169: mpp/vproc/rga/test/rga_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:2449: mpp/vproc/rga/test/CMakeFiles/rga_test.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
`
what to do?

nyanmisaka, 2025-03-11:
It's MPP's own issue and has nothing to do with this repo. Disable TEST `-DBUILD_TEST=OFF`.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation

closed 2025-03-11 by nyanmisaka

sqf147, 2025-03-11:
`
root@radxa-zero3:~/dev/rkmpp/rkmpp_build# cmake \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_BUILD_TYPE=Release \
    -DBUILD_SHARED_LIBS=ON \
    -DBUILD_TEST=OFF \
    ..
-- cmake version 3.18.4
-- Detected ARMv8 system processor
-- current version:
-- fda5a02 author: nyanmisaka     2025-01-17 Revert "chore[cmake]: compile both dynamic and static libraries"
-- git version history:
-- fda5a02 author: nyanmisaka     2025-01-17 Revert \"chore[cmake]: compile both dynamic and static libraries\"  (grafted, HEAD -> jellyfin-mpp, origin/jellyfin-mpp)
-- total 1 git version recorded
-- Install git hooks done
-- Checking for one of the modules 'pthread'
-- Configuring done
-- Generating done
-- Build files have been written to: /root/dev/rkmpp/rkmpp_build
root@radxa-zero3:~/dev/rkmpp/rkmpp_build# make -j $(nproc)
[  1%] Built target mpp_rc
[  3%] Built target vproc_iep2
[ 14%] Built target osal
[ 14%] Built target codec_dummy_dec
[ 15%] Built target vproc_iep
[ 15%] Built target codec_dummy_enc
[ 19%] Built target enc_rc
[ 20%] Built target vdpu383_com
[ 21%] Built target vdpu34x_com
[ 22%] Built target hal_dummy
[ 24%] Built target vproc_vdpp
[ 31%] Built target mpp_base
[ 32%] Built target hal_av1d_com
[ 35%] Built target utils
[ 36%] Built target dec_common
[ 37%] Built target codec_avsd
[ 39%] Built target codec_h263d
[ 39%] Built target codec_avs2d
[ 39%] Built target codec_mpeg2d
[ 40%] Built target codec_mpeg4d
[ 41%] Built target codec_vp8d
[ 42%] Built target codec_vp9d
[ 43%] Built target codec_jpegd
[ 45%] Built target codec_av1d
[ 48%] Built target codec_jpege
[ 48%] Built target codec_h264e
[ 49%] Built target codec_vp8e
[ 51%] Built target codec_h265e
[ 52%] Built target vproc_rga
[ 52%] Built target hal_vepu_common
[ 53%] Built target hal_h263d
[ 54%] Built target hal_mpeg2d
[ 55%] Built target hal_vp8d
[ 57%] Built target hal_mpeg4d
[ 59%] Built target hal_jpegd
[ 60%] Built target hal_av1d_vpu
[ 61%] Built target hal_jpege_vpu
[ 62%] Built target hal_h264e_com
[ 65%] Built target hal_vp8e
[ 67%] Built target hal_avsd
[ 68%] Built target hal_avs2d
[ 70%] Built target hal_h265d
[ 72%] Built target hal_vp9d
[ 72%] Built target hal_av1d_rkv
[ 74%] Built target hal_vepu541_common
[ 75%] Built target hal_common
[ 75%] Built target hal_jpege_com
[ 76%] Built target vdpp
[ 78%] Built target codec_h265d
[ 81%] Built target codec_h264d
[... 41 lines cut ...]
make[1]: *** [CMakeFiles/Makefile2:2562: mpp/vproc/iep/test/CMakeFiles/iep_test.dir/all] Error 2
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_jpege_vepu_rc'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `qp_reorder_table'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `get_msb_lsb_at_pos'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_jpege_vepu_deinit_rc'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_api_jpegd'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_jpege_vepu_init_rc'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `jpege_restart_marker'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_api_vp8d'
collect2: error: ld returned 1 exit status
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_jpege_vepu_rc'
make[2]: *** [mpp/vproc/vdpp/test/CMakeFiles/vdpp_test.dir/build.make:170: mpp/vproc/vdpp/test/vdpp_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:2848: mpp/vproc/vdpp/test/CMakeFiles/vdpp_test.dir/all] Error 2
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `qp_reorder_table'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `get_msb_lsb_at_pos'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_jpege_vepu_deinit_rc'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_api_jpegd'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_jpege_vepu_init_rc'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `jpege_restart_marker'
/usr/bin/ld: ../../../librockchip_mpp.so.0: undefined reference to `hal_api_vp8d'
collect2: error: ld returned 1 exit status
make[2]: *** [mpp/vproc/iep2/test/CMakeFiles/iep2_test.dir/build.make:169: mpp/vproc/iep2/test/iep2_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:2675: mpp/vproc/iep2/test/CMakeFiles/iep2_test.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
`

nyanmisaka, 2025-03-11:
Not sure but you might have messed up the environment variables. Please build in a clean environment.

For reference: [build-mpp-log.txt](https://github.com/user-attachments/files/19186229/build-mpp-log.txt)

sqf147, 2025-03-11:
Will try to build in a clean env.
Thank you for your reply.