nyanmisaka/ffmpeg-rockchip #169: Occasionally Segmentation fault used on rk3562 platform

cnjswwxbtcc, 2025-03-13:
```bash
# ./fvtest
视频码率: 1066
视频编码: 27
视频像素格式: 179
视频总帧数: 0
视频最大B帧数量: 0
原始视频宽高: 640 , 480
输出视频宽高: 720 , 480
解码硬件格式: drm_prime
filter string is: scale_rkrga=w=720:h=480:format=rgb24,hwdownload,format=rgb24
RGA supported formats: drm_prime, rgb24
XOpenIM() failed
[/root/serious.mp4] Playback started
[h264_rkmpp @ 0x38648a30] An invalid frame was output by a decoder. This is a bug, please report it.
Error during decoding
rga_api version 1.9.2_[1]
filter config success
RingBuffer Size: 1
is key frame: 0, frame pts: 0, frame dts: -9223372036854775808
get frame: 0
Segmentation fault
```

mark3303, 2025-03-27:
I met maybe the same problem when I use the encoder "hevc_rkmpp". It broken with "Segmentation fault" before avcodec_send_frame, the gdb info show down here.

#0  0x0000007f9877a968 in mpp_buffer_create () from /oem/usr/lib/librockchip_mpp.so.1
#1  0x0000007f9877e33c in mpp_buffer_get_with_tag () from /oem/usr/lib/librockchip_mpp.so.1
#2  0x0000000000d12bb4 in rkmpp_drm_pool_alloc (opaque=0x346585f0, size=<optimized out>) at libavutil/hwcontext_rkmpp.c:225
#3  0x0000000000cffd6c in pool_alloc_buffer (pool=0x34687f00) at libavutil/buffer.c:362
#4  av_buffer_pool_get (pool=0x34687f00) at libavutil/buffer.c:401
#5  0x0000000000d125c8 in rkmpp_get_buffer (hwfc=0x346585f0, frame=0x3466f7a0) at libavutil/hwcontext_rkmpp.c:337
#6  0x0000000000d110c0 in av_hwframe_get_buffer (hwframe_ref=<optimized out>, frame=0x3466f7a0, flags=flags@entry=0) at libavutil/hwcontext.c:563
#7  0x000000000075bc1c in rkmpp_submit_frame (frame=0x346670b0, avctx=0x34674600) at libavcodec/rkmppenc.c:607
#8  rkmpp_encode_frame (avctx=0x34674600, packet=0x34675420, frame=0x346670b0, got_packet=0x7ff974600c) at libavcodec/rkmppenc.c:875
#9  0x000000000056235c in ff_encode_encode_cb (avctx=avctx@entry=0x34674600, avpkt=avpkt@entry=0x34675420, frame=0x346670b0, got_packet=got_packet@entry=0x7ff974600c) at libavcodec/encode.c:264
#10 0x00000000005626c4 in encode_simple_internal (avpkt=0x34675420, avctx=0x34674600) at libavcodec/encode.c:350
#11 encode_simple_receive_packet (avpkt=<optimized out>, avctx=<optimized out>) at libavcodec/encode.c:364
#12 encode_receive_packet_internal (avctx=avctx@entry=0x34674600, avpkt=0x34675420) at libavcodec/encode.c:398
#13 0x0000000000562994 in avcodec_send_frame (avctx=0x34674600, frame=0x34674ff0) at libavcodec/encode.c:541
#14 0x0000000000462db0 in Storage::Push(cv::Mat) ()
#15 0x0000000000461e78 in main ()

It could worked with the same code by using encoder "AV_CODEC_ID_H265".

nyanmisaka, 2025-03-27:
https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation#attention

> Make sure you have built and installed the latest MPP runtime and headers. Otherwise undefined behavior may occur, such as segmentation fault. Be careful if you use MPP deb packages from a vendor/BSP image, these packages are likely to be out of date.

No it’s not related.

nyanmisaka, 2025-04-08:
Closing as I couldn't get a sample.

closed 2025-04-08 by nyanmisaka