nyanmisaka/ffmpeg-rockchip #170: 一些视频流无法成功调用硬加速

yuege969, 2025-03-18:
我在使用 `ffmpeg-rockchip` 的硬解码过程中遇到了一些问题，我不知道哪里出了问题，还请不吝赐教。 
板端环境：`rk3568`芯片，内核版本`5.10.110`
`ffmpeg`是基于[`Wiki`](https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation) 文档改的交叉编译版本，编译脚本如下：
```bash
#!/bin/bash

# 设置交叉编译工具链和安装路径（根据实际硬件调整）
export CROSS_COMPILE=/opt/toolchains-10.2.0/bin/aarch64-none-linux-gnu
export INSTALL_DIR=/home/zmy/install

# Build libdrm
mkdir -p ~/dev && cd ~/dev
# git clone --depth=1 https://gitlab.freedesktop.org/mesa/drm.git libdrm
git clone --depth=1 https://github.com/grate-driver/libdrm.git libdrm
cd libdrm

# 创建meson交叉编译配置文件
cat > libdrm_cross_file.txt <<EOF
[binaries]
c = '$CROSS_COMPILE-gcc'
cpp = '$CROSS_COMPILE-g++'
ar = '$CROSS_COMPILE-ar'
strip = '$CROSS_COMPILE-strip'
pkgconfig = 'pkg-config'

[properties]
needs_exe_wrapper = true

[host_machine]
system = 'linux'
cpu_family = 'aarch64'
cpu = 'cortex-a53'
endian = 'little'

[paths]
prefix = '$INSTALL_DIR'
EOF

# 配置和编译libdrm
meson setup build \
    --prefix=$INSTALL_DIR \
    --libdir=lib \
    --buildtype=release \
    --default-library=shared \
    -Dudev=false \
    -Dvalgrind=false \
    -Dcairo-tests=false \
    -Dman-pages=false \
    -Detnaviv=false \
    -Dfreedreno=false \
    -Dintel=false \
    -Dnouveau=false \
    -Domap=false \
    -Dradeon=false \
    -Dvc4=false \
    -Dvmwgfx=false \
    -Damdgpu=false \
    -Dexynos=false \
    -Dtegra=false \
    -Dfreedreno-kgsl=false \
    --cross-file=libdrm_cross_file.txt

ninja -C build
ninja -C build install

# Build MPP
mkdir -p ~/dev && cd ~/dev
git clone -b jellyfin-mpp --depth=1 https://github.com/nyanmisaka/mpp.git rkmpp
pushd rkmpp
mkdir rkmpp_build
pushd rkmpp_build
cmake \
    -DCMAKE_INSTALL_PREFIX=$INSTALL_DIR \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_SYSTEM_NAME=Linux \
    -DCMAKE_C_COMPILER=$CROSS_COMPILE-gcc \
    -DCMAKE_CXX_COMPILER=$CROSS_COMPILE-g++ \
    -DBUILD_SHARED_LIBS=ON \
    -DBUILD_TEST=OFF \
    ..
make -j $(nproc)
make install
popd
popd

# Build RGA
mkdir -p ~/dev && cd ~/dev
git clone -b jellyfin-rga --depth=1 https://github.com/nyanmisaka/rk-mirrors.git rkrga

# First, let's fix the cross file
cd ~/dev/rkrga
cat > rkrga_cross_file.txt <<EOF
[binaries]
c = '$CROSS_COMPILE-gcc'
cpp = '$CROSS_COMPILE-g++'
ar = '$CROSS_COMPILE-ar'
strip = '$CROSS_COMPILE-strip'
pkgconfig = 'pkg-config'

[properties]
needs_exe_wrapper = true

[host_machine]
system = 'linux'
cpu_family = 'aarch64'
cpu = 'cortex-a53'
endian = 'little'

[paths]
prefix = '$INSTALL_DIR'
EOF

# Now try building again, but with libdrm enabled
# Make sure CROSS_COMPILE and INSTALL_DIR are set correctly
meson setup rkrga/rkrga_build \
    --prefix=$INSTALL_DIR \
    --libdir=lib \
    --buildtype=release \
    --default-library=shared \
    -Dcpp_args=-fpermissive \
    -Dlibdrm=true \
    -Dlibrga_demo=false \
    --cross-file=rkrga_cross_file.txt
ninja -C rkrga/rkrga_build
ninja -C rkrga/rkrga_build install

# Build FFmpeg with cross compile
mkdir -p ~/dev && cd ~/dev
git clone --depth=1 https://github.com/nyanmisaka/ffmpeg-rockchip.git ffmpeg
cd ffmpeg

# 设置环境变量以便FFmpeg找到交叉编译的库
export PKG_CONFIG_PATH=$INSTALL_DIR/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=$INSTALL_DIR/lib:$LD_LIBRARY_PATH

./configure \
    --prefix=$INSTALL_DIR \
    --cross-prefix=$CROSS_COMPILE- \
    --target-os=linux \
    --arch=aarch64 \
    --enable-gpl \
    --enable-version3 \
    --enable-libdrm \
    --enable-rkmpp \
    --enable-rkrga \
    --extra-cflags="-I$INSTALL_DIR/include" \
    --extra-ldflags="-L$INSTALL_DIR/lib" \
    --extra-libs="-lpthread -lm" \
    --pkg-config=pkg-config

make -j $(nproc)

# 验证编译结果（需要交叉编译的环境支持）
# ./ffmpeg -decoders | grep rkmpp
# ./ffmpeg -encoders | grep rkmpp
# ./ffmpeg -filters | grep rkrga

# 安装到指定目录
make install
```
### Case # 1. 使用 Blender 渲染出的视频。（使用手机拍摄的一张图片拷贝120份，形成图片序列导入Blender渲染出的h264编码视频）[视频获取地址](https://www.alipan.com/s/6dhz4oKLax6)

[image]

将该视频用 `ffserver` 推流，使用 `ffplay` 验证拉流没问题.

[image]

- 硬解码：获取不到解码后的数据帧
```bash
./ffmpeg -stream_loop -1 -hwaccel rkmpp -hwaccel_output_format drm_prime -rtsp_transport tcp -i rtsp://192.168.0.55:8554/test1 -an -sn -vframes 5000 -f null -
```

[image]

- 软解码：正常
```bash
./ffmpeg -stream_loop -1 -rtsp_transport tcp -i rtsp://192.168.0.55:8554/test1 -an -sn -vframes 5000 -f null -
```

[image]

### Case # 2. 在 `EasyDarwin` 流媒体平台配置真实相机，由 `EasyDarwin` 直接拉取相机流地址，转出新的 `rtsp` 流地址：`rtsp://192.168.0.14:25544/live/stream_3`
直接拉取相机的流地址是正常的，但是由中间平台中转之后就异常了，情况和Case # 1一致，软解正常，硬解取不到解码后的数据帧。

nyanmisaka, 2025-03-18:
Switch to FFmpeg 7.x branch and add `-bsf:v dump_extra` before the `-i INPUT`.

https://github.com/nyanmisaka/ffmpeg-rockchip/branches

yuege969, 2025-03-18:
> Switch to FFmpeg 7.x branch and add `-bsf:v dump_extra` before the `-i INPUT`.
> 
> https://github.com/nyanmisaka/ffmpeg-rockchip/branches

我切换至branch 7.1 并且重新交叉编译了一份之后，按您所述增添参数，执行如下指令出现了新的问题
```bash
LD_LIBRARY_PATH=../lib ./ffmpeg -stream_loop -1 -hwaccel rkmpp -hwaccel_output_format drm_prime -rtsp_transport tcp -bsf:v dump_extra -i rtsp://192.168.0.55:8554/test1 -an -sn -vframes 5000 -f null -
```

[image]

并且，如果我不增加这个参数，按原Case # 1中的指令执行，问题依然存在，且现象与之前一致。

nyanmisaka, 2025-03-18:
Check your version numbers, they are not from 7.1

```
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
```

yuege969, 2025-03-18:
> Check your version numbers, they are not from 7.1
>
> ```
>   libavutil      59. 39.100 / 59. 39.100
>   libavcodec     61. 19.101 / 61. 19.101
>   libavformat    61.  7.100 / 61.  7.100
>   libavdevice [...]

问题已解决，谢谢。

closed 2025-03-18 by nyanmisaka