nyanmisaka/ffmpeg-rockchip #171: rk3588上解码时，输入1080i 50，输出的pts间隔不对

changliwu, 2025-03-19:
你好：
       解码输入dts间隔是1800， 输出pts间隔大部分是1800，突然会跳一个19800的间隔，如下所示：
输入dts间隔       输出pts间隔
1800
1800
1800
1800
                           1800
                           1800
1800
1800
1800
1800
1800
1800
1800
1800
1800
1800
                          19800
                          1800
                          1800
                          1800

当输入1080p 25，解码输出pts间隔正常，输入dts间隔3600，输出pts间隔也是3600
输入dts间隔       输出pts间隔
3602
3600
3600
                        3600
                        3598
3600
3600
3600
                        3602
                        3600
                        3600
                        3600
                        3600

nyanmisaka, 2025-03-19:
FFmpeg is NOT responsible for deinterlace, instead it feeds the PTS generated by FFmpeg to MPP and gets the deinterlaced frame along with the updated PTS from MPP. So you have to track down where that peak value comes from yourself.

changliwu, 2025-03-26:
问题找到了，谢谢！
解决办法：
从github.com/rockchip-linux/mpp下载mpp，然后编译，生成新的librockchip_mpp.so,替换ffmpeg-rockchip编译生成的librockchip_mpp.so

closed 2025-03-26 by changliwu