nyanmisaka/ffmpeg-rockchip #174: Orange Pi 3B - "Failed to init MPP context"

AlexxIT, 2025-03-21:
Device: [Orange Pi 3B](https://www.armbian.com/orangepi3b/)
Core: RK3566
OS: Armbian Linux v6.12
Kernel: Linux orangepi3b 6.12.17-current-rockchip64 ... aarch64 GNU/Linux

I've created a file `/etc/udev/rules.d/99-rk-device-permissions.rules`.

I have make and install `https://github.com/rockchip-linux/mpp` project. But I can't see `/dev/mpp_service` and `/dev/rga`:
```
alexey@orangepi3b:~$ sudo ls -l /dev | grep -E "mpp|rga|dri|dma_heap"
drwxr-xr-x 2 root root          80 jan  1  1970 dma_heap
drwxr-xr-x 3 root root         120 mar 21 11:04 dri
```

Can run `./mpp_info_test`:
```
alexey@orangepi3b:~/mpp/build/linux/aarch64/test$ ./mpp_info_test
mpp[24014]: mpp_info_test: normal version log:
mpp[24014]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[24014]: mpp_info_test: history version log:
mpp[24014]: mpp_info: mpp version history 10:
...
```

Can't run `/usr/local/bin/mpp_info_test`:
```
alexey@orangepi3b:~/mpp/build/linux/aarch64/test$ /usr/local/bin/mpp_info_test
/usr/local/bin/mpp_info_test: error while loading shared libraries: librockchip_mpp.so.1: cannot open shared object file: No such file or directory
```

Have ffmpeg issues:
```
[h264_rkmpp @ 0xaaaaf5832f70] Failed to init MPP context: -1
[vist#0:1/h264 @ 0xaaaaf58329f0] Error while opening decoder: Generic error in an external library
[vost#0:0/hevc_rkmpp @ 0xaaaaf5755b30] Error initializing a simple filtergraph
```

Do I need to install `mpp_service` and `rga` separately somehow?
Or maybe it's just that my kernel doesn't support RKMPP like you said here: https://github.com/nyanmisaka/ffmpeg-rockchip/issues/133#issuecomment-2488002940 ?

nyanmisaka, 2025-03-21:
Only the Rockchip 5.10 and 6.1 BSP kernel are supported.

https://github.com/nyanmisaka/ffmpeg-rockchip#important

closed 2025-03-21 by nyanmisaka

AlexxIT, 2025-03-21:
Thanks!

For those who are interested, first I downloaded Arabian Linux v6.12 (Home Assistant edition).
After that, I had to download Arabian Linux 6.1 (Minimal / IOT edition). And fix the access rights to `/dev/mpp_service` and /dev/rga`.
Everything is working now.