nyanmisaka/ffmpeg-rockchip #176: use filter to trim the video error

zyKrip, 2025-03-28:
/lib/jellyfin-ffmpeg/ffmpeg -init_hw_device rkmpp=rk        -hwaccel rkmpp  -c:v hevc_rkmpp      -hwaccel_output_format drm_prime        -afbc rga      -i "/P1103141.MOV"    -filter_complex "[0:v]trim=start=6.53:end=29.93,setpts=PTS-STARTPTS[v0]; [0:a]atrim=start=6.53:end=29.93,asetpts=PTS-STARTPTS[a0]; [v0][a0]concat=n=1:v=1:a=1[outv][outa]"        -map "[outa]"        -map "[outv]"        -c:v hevc_rkmpp  "/P1103141-202503251207282.MOV"
ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto=auto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libxml2 --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libharfbuzz --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=arm64 --cross-prefix=/usr/bin/aarch64-linux-gnu- --toolchain=hardened --enable-cross-compile --enable-rkmpp --enable-rkrga --enable-libwebp --enable-neon
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[aist#0:1/pcm_s24be @ 0x55b2a685d0] Guessed Channel Layout: mono
[aist#0:2/pcm_s24be @ 0x55b2ade510] Guessed Channel Layout: mono
[aist#0:3/pcm_s24be @ 0x55b2b53c90] Guessed Channel Layout: mono
[aist#0:4/pcm_s24be @ 0x55b2b544a0] Guessed Channel Layout: mono
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/P1103141.MOV':
  Metadata:
    major_brand     : qt  
    minor_version   : 537986816
    compatible_brands: qt  pana
    creation_time   : 2025-01-03T03:34:33.000000Z
    com.panasonic.Semi-Pro.metadata.xml: <?xml version="1.0" encoding="UTF-8" standalone="no" ?>
                    : <ClipMain xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="urn:schemas-Professional-Plug-in:Semi-Pro:ClipMetadata:v1.0">
                    :   <ClipContent>
                    :     <GlobalClipID>060A2B340101010501010D211300000087370E2F0B000004718D146121120038</GlobalClipID>
                    :     <Duration>3012</Duration>
                    :     <EditUnit>1/25</EditUnit>
                    :     <EssenceList>
                    :       <Video>
                    :         <Codec BitRate="150">H265_420_LongGOP</Codec>
                    :         <ActiveLine>2160</ActiveLine>
                    :         <ActivePixel>3840</ActivePixel>
                    :         <BitDepth>10</BitDepth>
                    :         <FrameRate>25p</FrameRate>
                    :         <TimecodeType>NonDrop</TimecodeType>
                    :         <StartTimecode>18:36:57:16</StartTimecode>
                    :       </Video>
                    :       <Audio>
                    :         <Channel>4</Channel>
                    :         <SamplingRate>96000</SamplingRate>
                    :         <BitsPerSample>24</BitsPerSample>
                    :       </Audio>
                    :     </EssenceList>
                    :     <ClipMetadata>
                    :       <Rating>0</Rating>
                    :       <Access>
                    :         <CreationDate>2025-01-03T11:34:33+08:00</CreationDate>
                    :         <LastUpdateDate>2025-01-03T11:34:33+08:00</LastUpdateDate>
                    :       </Access>
                    :       <Device>
                    :         <Manufacturer>Panasonic</Manufacturer>
                    :         <ModelName>DC-S5M2</ModelName>
                    :       </Device>
                    :       <Shoot>
                    :         <StartDate>2025-01-03T11:34:33+08:00</StartDate>
                    :       </Shoot>
                    :     </ClipMetadata>
                    :   </ClipContent>
                    :   <UserArea>
                    :     <AcquisitionMetadata xmlns="urn:schemas-Professional-Plug-in:P2:CameraMetadata:v1.2">
                    :       <CameraUnitMetadata>
                    :         <ISOSensitivity>640</ISOSensitivity>
                    :         <Gamma>
                    :           <CaptureGamma>V-Log</CaptureGamma>
                    :         </Gamma>
                    :         <Gamut>
                    :           <CaptureGamut>V-Gamut</CaptureGamut>
                    :         </Gamut>
                    :       </CameraUnitMetadata>
                    :     </AcquisitionMetadata>
                    :   </UserArea>
                    : </ClipMain>
                    : 
  Duration: 00:02:00.48, start: 0.000000, bitrate: 151436 kb/s
  Stream #0:0[0x1](und): Video: hevc (Main 10) (hvc1 / 0x31637668), yuv420p10le(pc, bt709), 3840x2160 [SAR 1:1 DAR 16:9], 142053 kb/s, 25 fps, 25 tbr, 90k tbn (default)
      Metadata:
        creation_time   : 2025-01-03T03:34:33.000000Z
        vendor_id       : [0][0][0][0]
        timecode        : 18:36:57:16
  Stream #0:1[0x2](und): Audio: pcm_s24be (lpcm / 0x6D63706C), 96000 Hz, mono, s32 (24 bit), 2304 kb/s (default)
      Metadata:
        creation_time   : 2025-01-03T03:34:33.000000Z
        vendor_id       : [0][0][0][0]
        timecode        : 18:36:57:16
  Stream #0:2[0x3](und): Audio: pcm_s24be (lpcm / 0x6D63706C), 96000 Hz, mono, s32 (24 bit), 2304 kb/s (default)
      Metadata:
        creation_time   : 2025-01-03T03:34:33.000000Z
        vendor_id       : [0][0][0][0]
        timecode        : 18:36:57:16
  Stream #0:3[0x4](und): Audio: pcm_s24be (lpcm / 0x6D63706C), 96000 Hz, mono, s32 (24 bit), 2304 kb/s (default)
      Metadata:
        creation_time   : 2025-01-03T03:34:33.000000Z
        vendor_id       : [0][0][0][0]
        timecode        : 18:36:57:16
  Stream #0:4[0x5](und): Audio: pcm_s24be (lpcm / 0x6D63706C), 96000 Hz, mono, s32 (24 bit), 2304 kb/s (default)
      Metadata:
        creation_time   : 2025-01-03T03:34:33.000000Z
        vendor_id       : [0][0][0][0]
        timecode        : 18:36:57:16
  Stream #0:5[0x6](und): Data: none (tmcd / 0x64636D74), 0 kb/s (default)
      Metadata:
        creation_time   : 2025-01-03T03:34:33.000000Z
        timecode        : 18:36:57:16
rga_api version 1.10.1_[3]
Stream mapping:
  Stream #0:0 (hevc_rkmpp) -> trim:default
  Stream #0:1 (pcm_s24be) -> atrim:default
  concat -> Stream #0:0 (aac)
  concat -> Stream #0:1 (hevc_rkmpp)
Press [q] to stop, [?] for help
[vost#0:1/hevc_rkmpp @ 0x55b2acd180] No information about the input framerate is available. Falling back to a default value of 25fps. Use the -r option if you want a different framerate.
[hevc_rkmpp @ 0x55b2acd5f0] Unsupported input pixel format 'nv15'
[vost#0:1/hevc_rkmpp @ 0x55b2acd180] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
[fc#0 @ 0x55b2a62210] Error sending frames to consumers: Function not implemented
[fc#0 @ 0x55b2a62210] Task finished with error code: -38 (Function not implemented)
[fc#0 @ 0x55b2a62210] Terminating thread with return code -38 (Function not implemented)
[aost#0:0/aac @ 0x55b2acc900] Could not open encoder before EOF
[aost#0:0/aac @ 0x55b2acc900] Task finished with error code: -22 (Invalid argument)
[aost#0:0/aac @ 0x55b2acc900] Terminating thread with return code -22 (Invalid argument)
[vost#0:1/hevc_rkmpp @ 0x55b2acd180] Could not open encoder before EOF
[vost#0:1/hevc_rkmpp @ 0x55b2acd180] Task finished with error code: -22 (Invalid argument)
[vost#0:1/hevc_rkmpp @ 0x55b2acd180] Terminating thread with return code -22 (Invalid argument)
[out#0/mov @ 0x55b2acbee0] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0KiB time=N/A bitrate=N/A speed=N/A    
Conversion failed!

nyanmisaka, 2025-03-28:
```
[hevc_rkmpp @ 0x55b2acd5f0] Unsupported input pixel format 'nv15'
```

The encoder does not support NV15 output after decoding. You need to use the `scale_rkrga=format=nv12` filter to convert the pixel format to NV12 before feeding it to the encoder.

See also https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Video-Transcode

closed 2025-03-28 by nyanmisaka

zyKrip, 2025-03-28:
great @nyanmisaka  /lib/jellyfin-ffmpeg/ffmpeg -init_hw_device rkmpp=rk -hwaccel rkmpp -c:v hevc_rkmpp \
    -hwaccel_output_format drm_prime -afbc rga -i "/P1103141.MOV" \
    -filter_complex "[0:v]trim=start=6.53:end=29.93,setpts=PTS-STARTPTS[v0]; \
                     [v0]scale_rkrga=format=nv12:afbc=1[v0_scaled]; \
                     [0:a]atrim=start=6.53:end=29.93,asetpts=PTS-STARTPTS[a0]; \
                     [v0_scaled][a0]concat=n=1:v=1:a=1[outv][outa]" \
    -map "[outa]" -map "[outv]" -c:v hevc_rkmpp "/P1103141-202503251207282.MOV"