nyanmisaka/ffmpeg-rockchip #177: subtitles with scale_rkrga

zyKrip, 2025-04-02:
ffmpeg \
-init_hw_device rkmpp=rk \
-hwaccel rkmpp \
-hwaccel_output_format drm_prime \
-afbc rga \
-i "P1103141.MOV" \
-filter_complex "[0:v]trim=start=6.53:end=29.93,setpts=PTS-STARTPTS[v0]; \
[v0]scale_rkrga=format=nv12:afbc=1[v0_scaled]; \
[0:a]atrim=start=6.53:end=29.93,asetpts=PTS-STARTPTS[a0]; \
[v0_scaled][a0]concat=n=1:v=1:a=1[outv][outa]; [outv]subtitles=P1103141-202504020636110.MOV.srt,format=nv12[outv]" \
-map "[outa]" \
-map "[outv]" \
-c:v hevc_rkmpp \
"P1103141-202504020634470.MOV"

Impossible to convert between the formats supported by the filter 'Parsed_concat_5' and the filter 'auto_scale_0'
[fc#0 @ 0x559b3552a0] Error reinitializing filters!
[fc#0 @ 0x559b3552a0] Task finished with error code: -38 (Function not implemented)
[fc#0 @ 0x559b3552a0] Terminating thread with return code -38 (Function not implemented)

How can I implement this feature

nyanmisaka, 2025-04-02:
Change `scale_rkrga=format=nv12:afbc=1` to `scale_rkrga=format=nv12,hwmap,format=nv12`

zyKrip, 2025-04-02:
@nyanmisaka great！

zyKrip, 2025-04-02:
Can you explain the principle to me?

zyKrip, 2025-04-02:
Is there a similar method to directly use hardware filters to achieve the desired effect？

nyanmisaka, 2025-04-02:
Burning SRT/SSA/ASS subtitles only has software filters and cannot be done in hardware.