nyanmisaka/ffmpeg-rockchip #182: ffmpeg works unstably during inputs camera streams and outputs mosaic.

liangtaikonaut, 2025-04-17:
Dear ffmpeg-rockchip developers,

The command is:
ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -f lavfi -i color=c=cyan:s=1920x1080,format=nv12 \
-rtsp_transport tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -re  -i rtsp://admin:taiji2022@10.0.24.39:554 \
-rtsp_transport tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -re  -i rtsp://admin:taiji2022@10.0.24.39:554 \
-rtsp_transport tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -re  -i rtsp://admin:taiji2022@10.0.24.39:554 \
-rtsp_transport tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -re  -i rtsp://admin:taiji2022@10.0.24.39:554 \
-filter_complex "[0:v]hwupload[main];[1:v]scale_rkrga=w=960:h=540:format=bgra:afbc=1[overlay1];[2:v]scale_rkrga=w=960:h=540:format=bgra:afbc=1[overlay2];[3:v]scale_rkrga=w=960:h=540:format=bgra:afbc=1[overlay3];[4:v]scale_rkrga=w=960:h=540:format=bgra:afbc=1[overlay4];\
[main][overlay1]overlay_rkrga=eof_action=pass:repeatlast=0:format=nv12:x=0:y=0[temp1];\
[temp1][overlay2]overlay_rkrga=eof_action=pass:repeatlast=0:format=nv12:x=0:y=H/2[temp2];\
[temp2][overlay3]overlay_rkrga=eof_action=pass:repeatlast=0:format=nv12:x=W/2:y=0[temp3];\
[temp3][overlay4]overlay_rkrga=eof_action=pass:repeatlast=0:format=nv12:x=W/2:y=H/2" \
-c:v h264_rkmpp -b:v 2M -maxrate 4M -an -f flv rtmp://127.0.0.1:10050/etest/mosaic_04z

it works fine at the beginning, and after several minutes it looks like:

[image]

after 2 hours, there's only background(cyan) left.

nyanmisaka, 2025-04-17:
Using four local video files as input data, a short test did not reproduce the problem of layer disappearance. A longer test may be required.

Also, I don't see any mosaics. The point is that **2M / 2000K** bitrate is **TOO LOW** for encoding **1080p** using h264_rkmpp **hardware** encoder.

liangtaikonaut, 2025-04-18:
Thanks for your comments, it's much better now with "-b:v 4M -maxrate 8M":) I'm doing more tests.

BTW: I mean "mosaic" to be something like"split-screen"  , sorry for confusing you.

closed 2025-07-03 by nyanmisaka