nyanmisaka/ffmpeg-rockchip #183: Failure to run make

AlejandroHuerta, 2025-04-22:
I'm trying to follow the instructions for compilation but ffmpeg is failing to build. Here's the output of `make -j$(nproc)` command.

```LD      ffmpeg_g
LD      ffplay_g
LD      ffprobe_g
/usr/bin/ld: /lib/aarch64-linux-gnu/libtirpc.so.3: warning: common of `rpc_createerr@@GLIBC_2.17' overridden by definition from /lib/aarch64-linux-gnu/libc.so.6
/usr/bin/ld: /lib/aarch64-linux-gnu/libtirpc.so.3: warning: common of `rpc_createerr@@GLIBC_2.17' overridden by definition from /lib/aarch64-linux-gnu/libc.so.6
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
make: *** [Makefile:133: ffmpeg_g] Error 1
make: *** Waiting for unfinished jobs....
/usr/bin/ld: /lib/aarch64-linux-gnu/libtirpc.so.3: warning: common of `rpc_createerr@@GLIBC_2.17' overridden by definition from /lib/aarch64-linux-gnu/libc.so.6
radxa@radxa-zero3:~/dev/ffmpeg$ make -j$(nproc)
STRIP   ffmpeg
STRIP   ffplay
STRIP   ffprobe
make: *** [Makefile:125: ffmpeg] Error 1
make: *** Waiting for unfinished jobs....

I'm not versed in using make so apologies if this is something obvious.

nyanmisaka, 2025-04-22:
> radxa@radxa-zero3

I guess the compilation process was aborted because the compilation platform had insufficient memory and other resources.

It is recommended to use a platform with larger memory for compilation, or use cross-compilation.

AlejandroHuerta, 2025-05-09:
That's definitely possible. I was trying to compile on a zero 3w with 1GB of ram. I'll close this for now. Thank you for the suggestion!

closed 2025-05-09 by AlejandroHuerta