nyanmisaka/ffmpeg-rockchip #184: RKRGA filter fail to work with output as rawvideo

cbalint13, 2025-04-27:
Trying to use rkrga filter to scale down and convert to bgr24 pixel format as raw image via "pipe:" output fails.
Trying to write a pure c++ flow also falls into errors like "Impossible to convert between the formats supported by the filter".

---

* Here is the CLI command (HW_decoding -> HW_rga-scaling+formatting -> pipe:):
```
ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -vcodec hevc_rkmpp -i /data/video/1/2025/04/27/H134858.asf -vf hwupload,scale_rkrga=w=1280:h=720:format=bgr24 -f rawvideo -pix_fmt bgr24 -s 1280x720 pipe:
```

* Or with syntetic data (based on the [wiki](https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Filter) example):
```
ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -f lavfi -i testsrc2=s=1920x1080,format=bgr24 -vf hwupload,scale_rkrga=w=1280:h=720:format=bgr24 -f rawvideo -s 1280x720 pipe:
```

```
ffmpeg version git-snapshot-rockchip-20250422-7.1 Copyright (c) 2000-2025 the FFmpeg developers
  built with gcc 15 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=aarch64 --optflags='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wno-complain-wrong-lang -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -mbranch-protection=standard -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer' --extra-ldflags='-Wl,-z,relro -Wl,--as-needed -Wl,-z,pack-relative-relocs -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -Wl,--build-id=sha1 ' --extra-cflags=' -I/usr/include/rav1e' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --enable-chromaprint --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-lcms2 --enable-libaom --enable-libaribb24 --enable-libaribcaption --enable-libdav1d --enable-libass --enable-libbluray --enable-libbs2b --enable-libcodec2 --enable-libcdio --enable-libdrm --enable-libjack --enable-libjxl --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libharfbuzz --enable-libilbc --enable-liblc3 --enable-libmp3lame --enable-libmysofa --enable-nvenc --enable-openal --enable-opencl --enable-opengl --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-libplacebo --enable-librsvg --enable-librav1e --enable-librubberband --enable-libqrencode --enable-libsmbclient --enable-version3 --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libv4l2 --enable-libvidstab --enable-libvpx --disable-vulkan --disable-libshaderc --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libxml2 --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-avfilter --enable-libmodplug --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --enable-rkmpp --enable-rkrga --shlibdir=/usr/lib64 --enable-lto
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
[aist#0:1/pcm_alaw @ 0x557ea0dcc0] Guessed Channel Layout: mono
Input #0, asf, from '/data/video/1/2025/04/27/H134858.asf':
  Duration: 00:02:59.97, start: 0.005000, bitrate: 1118 kb/s
  Stream #0:0: Video: hevc (Main) (HEVC / 0x43564548), yuvj420p(pc, bt709), 2880x1616, 5 fps, 5 tbr, 1k tbn
  Stream #0:1: Audio: pcm_alaw ([6][0][0][0] / 0x0006), 8000 Hz, mono, s16, 64 kb/s
mpp[51180]: mpp_info: mpp version: ff3ae5c author: Xiaobo Liu     2025-04-11 fix[mpi]: Fix typo
mpp[51180]: mpp_platform: client 4 driver is not ready!
mpp[51180]: mpp_platform: client 12 driver is not ready!
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> rawvideo (native))
Press [q] to stop, [?] for help
Impossible to convert between the formats supported by the filter 'Parsed_scale_rkrga_1' and the filter 'auto_scale_0'
[vf#0:0 @ 0x557ea0d5e0] Error reinitializing filters!
[vf#0:0 @ 0x557ea0d5e0] Task finished with error code: -38 (Function not implemented)
[vf#0:0 @ 0x557ea0d5e0] Terminating thread with return code -38 (Function not implemented)
[vost#0:0/rawvideo @ 0x557e8c7590] Could not open encoder before EOF
[vost#0:0/rawvideo @ 0x557e8c7590] Task finished with error code: -22 (Invalid argument)
[vost#0:0/rawvideo @ 0x557e8c7590] Terminating thread with return code -22 (Invalid argument)
[out#0/rawvideo @ 0x557ea13720] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0KiB time=N/A bitrate=N/A speed=N/A    
Conversion failed!
```

With a dummy ```-f null /dev/null``` output all works fine.

GermanGeraskin, 2025-04-28:
Hi.

Just for info:
I have similar issue on Debian 9.5 stretch (the original OS image for Asus Tinker board - RK3288).
I understand that probably it is not expected that ffmpeg-rockchip can provide rkmpp encoding on such old OS version.

I managed to compile rkmpp, rkrga, ffmpeg-rockchip.
FFmpeg list decoders, encoders and filters exactly like in your manual.
But example commands do not work.

Thank you

cbalint13, 2025-04-28:
> Hi.
>

Hi @GermanGeraskin ,

> Just for info: I have similar issue on Debian 9.5 stretch (the original OS image for Asus Tinker board - RK3288). I understand that probably it is not expected that ffmpeg-rockchip can provide rkmpp encoding on such old OS version.
> 
> I managed to compile rkmpp, rkrga, ffmpeg-rockchip. FFmpeg lists decoders, encoders and filters exactly like in your manual. But example commands do not work.
> 
> Thank you
> 
> P.S.: If you think that it worth to investigate (and fix it), I can provide detailed information about the OS, installed software, ffmpeg output (using commands with -loglevel debug or trace).

I use git heads of [mpp](https://github.com/rockchip-linux/mpp) (20250411) & [ffmpeg](https://github.com/nyanmisaka/ffmpeg-rockchip/tree/7.1) (git-snapshot-rockchip-20250422-7.1) @ 7.1 branch.
The configuration of ffmpeg and other details are visible in the top post of this very PR.

Here are the fedora automated builds (COPR, self maintained), I use fedora-42 (latest) on a rk3588 platform.
* https://copr.fedorainfracloud.org/coprs/rezso/ML/package/libmpp-rockchip
* https://copr.fedorainfracloud.org/coprs/rezso/ML/package/ffmpeg

You can see the build logs if interested in this.

---

To sum up, I try to write a performance (HW driven) c++ implementation that fetches the video frames & apply scaling .
Such flow works fine without inserting the filter step as sink, but with filter inserted gives error as in top of this PR.

I posted here a simple reproducible case using ffmpeg CLI, when the final sink is ```-f null``` or is a encoding target (like in wiki sample) all works fine.

nyanmisaka, 2025-04-28:
`hwupload,...,hwdownload` must be used between hardware filters and software filters.

-vf hwupload,scale_rkrga=w=1280:h=720:format=bgr24 **,hwdownload,format=bgr24**

cbalint13, 2025-04-28:
Hi @nyanmisaka ,

> `hwupload,...,hwdownload` must be used between hardware filters and software filters.
> 
> -vf hwupload,scale_rkrga=w=1280:h=720:format=bgr24 **,hwdownload,format=bgr24**
> 
> Alternatively **,hwmap=mode=read,format=bgr24**

@nyanmisaka 

Thank you for the help !
 
* I was aware of ```hwdownload``` but not that an explicit following ```format=bgr24``` once more is needed.

It is a progress, but still doesn't work, using this flow:
```
# ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -vcodec hevc_rkmpp -i /data/video/1/2025/04/27/H134858.asf -vf hwupload,scale_rkrga=w=1280:h=720:format=bgr24,hwdownload,format=bgr24 -f rawvideo pipe:
```

Nothing is coming on the output (empty):
```
ffmpeg version git-snapshot-rockchip-20250422-rpmfusion Copyright (c) 2000-2025 the FFmpeg developers
  built with gcc 15 (GCC)
  {....}
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
[aist#0:1/pcm_alaw @ 0x558a19fc90] Guessed Channel Layout: mono
Input #0, asf, from '/data/video/1/2025/04/27/H134858.asf':
  Duration: 00:02:59.97, start: 0.005000, bitrate: 1118 kb/s
  Stream #0:0: Video: hevc (Main) (HEVC / 0x43564548), yuvj420p(pc, bt709), 2880x1616, 5 fps, 5 tbr, 1k tbn
  Stream #0:1: Audio: pcm_alaw ([6][0][0][0] / 0x0006), 8000 Hz, mono, s16, 64 kb/s
mpp[62018]: mpp_info: mpp version: ff3ae5c author: Xiaobo Liu     2025-04-11 fix[mpi]: Fix typo
mpp[62018]: mpp_platform: client 4 driver is not ready!
mpp[62018]: mpp_platform: client 12 driver is not ready!
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> rawvideo (native))
Press [q] to stop, [?] for help
rga_api version 1.10.3_[2]
[vf#0:0 @ 0x558a0580d0] No filtered frames for output stream, trying to initialize anyway.
Output #0, rawvideo, to 'pipe:':
  Metadata:
    encoder         : Lavf61.7.100
  Stream #0:0: Video: rawvideo (BGR[24] / 0x18524742), bgr24(progressive), 1280x718, q=2-31, 110284 kb/s, 5 fps, 5 tbn
      Metadata:
        encoder         : Lavc61.19.101 rawvideo
[out#0/rawvideo @ 0x558a1a56f0] video:0KiB audio:0KiB subtitle:0KiB other streams:0KiB global headers:0KiB muxing overhead: unknown
[out#0/rawvideo @ 0x558a1a56f0] Output file is empty, nothing was encoded(check -ss / -t / -frames parameters if used)
# 
```

I attach the fulll ```-v debug``` variant here: [ffmpeg-cli.log.gz](https://github.com/user-attachments/files/19939815/ffmpeg-cli.log.gz)
To my look, the decoder feeds the rga filter, but then nothing is on the output, some log highlights:
* "No filtered frames for output stream, trying to initialize anyway."
* "Output file is empty, nothing was encoded(check -ss / -t / -frames parameters if used)"

nyanmisaka, 2025-04-30:
> [graph -1 input from stream 0:0 @ 0x7f60003e30] video frame properties congruent with link at pts_time: -9223372036854.775391

It looks like that ASF container doesn't contain valid PTS timestamps.

Add a flag to let FFmpeg make some: `ffmpeg -fflags +genpts ...`

Also, to enable end-to-end hwaccel:
```
ffmpeg -fflags +genpts -hwaccel rkmpp -hwaccel_output_format drm_prime \
-vcodec hevc_rkmpp -afbc rga -i /data/video/1/2025/04/27/H134858.asf
-vf hwupload,scale_rkrga=w=1280:h=720:format=bgr24,hwdownload,format=bgr24 \
-f rawvideo pipe:
```

cbalint13, 2025-04-30:
> It looks like that ASF container doesn't contain valid PTS timestamps.
>
> Add a flag to let FFmpeg make some: `ffmpeg -fflags +genpts ...`
>
> Also, to enable end-to-end hwaccel:
>
> ```
> ffmpeg -fflags [...]

@nyanmisaka 

I confirm now that really works fine.
Noticed here the ```-afbc rga``` addition (guessing for optimized DMA transfer DEC->RGA).

Thank you very much !

closed 2025-04-30 by cbalint13

nyanmisaka, 2025-04-30:
Yes, it makes the hardware decoder's DPB (Decoded Picture Buffer) stored in an Arm-specific compression format, thereby improving random access efficiency and saving memory bandwidth. RGA can directly utilize this compression format.

To reduce memory copying once more, you can change **hwdownload** to **hwmap=mode=read**.

cbalint13, 2025-06-27:
@nyanmisaka 

Thank you again for the help you provided here.

 In exchange I provide (beside CLI) examples for C++ and python: https://github.com/cbalint13/ffmpeg-bindings

* It is super useful to feed your favourite vision model at highiest frame rates, removing all the burdens from CPU side.
* The hw rescale and bgr24 step yields very much and is highly useful inside OpenCV or any other vision models (w/o RKNPU).

nyanmisaka, 2025-06-28:
> [@nyanmisaka](https://github.com/nyanmisaka)
> 
> Thank you again for the help you provided here.
> 
> In exchange I provide (beside CLI) native examples for C++ and python: [cbalint13/ffmpeg-bindings](https://github.com/cbalint13/ffmpeg-bindings?rgh-link-date=2025-06-27T15%3A38%3A51.000Z)
> 
> * It is super useful to feed your favourite vision model at highest frame rates, removing all the burdens from CPU side.
> * The hw rescale and bgr24 steps also yields very much and is useful inside OpenCV or any other vision models (w/o RKNPU).

Thanks for the demo showing how to use rkmpp hwaccel and rkrga hardware filters via ffmpeg libav interface. I will forward the repo to anyone who needs it as a reference.