nyanmisaka/ffmpeg-rockchip #187: RK3576硬件转码性能不符合预期，转换速率0.075x，比软件转码还弱

SCUT-chenjia, 2025-05-07:
本项目是否可以直接在RK3576上使用?或者有没有适配过RK3576？

nyanmisaka, 2025-05-07:
Support for RK3576 was added last year. I have not experienced any performance issues.

closed 2025-05-07 by nyanmisaka

SCUT-chenjia, 2025-05-07:
> Support for RK3576 was added last year. I have not experienced any performance issues.

root@rk3576-ubuntu:/home/twdz/rk3576/3rdparty/ffmpeg/bin# ./ffmpeg -hide_banner -filters | grep rkrga
 ... overlay_rkrga     VV->V      Rockchip RGA (2D Raster Graphic Acceleration) video compositor
 ... scale_rkrga       V->V       Rockchip RGA (2D Raster Graphic Acceleration) video resizer and format converter
 ... vpp_rkrga         V->V       Rockchip RGA (2D Raster Graphic Acceleration) video post-process (scale/crop/transpose)
root@rk3576-ubuntu:/home/twdz/rk3576/3rdparty/ffmpeg/bin# ./ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -f lavfi -i testsrc2=s=1920x1080,format=nv12 -vf hwupload,scale_rkrga=w=1280:h=720:format=nv12 -c:v h
264_rkmpp -b:v 4M -maxrate 4M -vframes 1000 -y /home/twdz/test.mp4
ffmpeg version e8ec55a5e8 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
  configuration: --enable-cross-compile --cross-prefix=aarch64-linux-gnu- --arch=aarch64 --cpu=cortex-a53 --target-os=linux --prefix=/home/twdz/rk3576/install/../3rdparty/ffmpeg --pkg-config-flags=--static --extra-cflags=-I/home/twdz/rk3576/install/../3rdparty/ffmpeg/include --extra-ldflags='-L/home/twdz/rk3576/install/../3rdparty/ffmpeg/lib -Wl,-rpath,/home/twdz/rk3576/install/../3rdparty/ffmpeg/lib' --extra-libs='-lpthread -lm' --bindir=/home/twdz/rk3576/install/../3rdparty/ffmpeg/bin --enable-gpl --enable-version3 --enable-debug --disable-optimizations --disable-stripping --enable-libfdk-aac --enable-libopus --enable-libx264 --enable-libx265 --enable-nonfree --disable-sdl2 --disable-asm --enable-static --enable-shared --enable-libdrm --enable-rkrga --enable-rkmpp
  libavutil      58.  2.100 / 58.  2.100
  libavcodec     60.  3.100 / 60.  3.100
  libavformat    60.  3.100 / 60.  3.100
  libavdevice    60.  1.100 / 60.  1.100
  libavfilter     9.  3.100 /  9.  3.100
  libswscale      7.  1.100 /  7.  1.100
  libswresample   4. 10.100 /  4. 10.100
  libpostproc    57.  1.100 / 57.  1.100
Input #0, lavfi, from 'testsrc2=s=1920x1080,format=nv12':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, nv12, 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 25 tbn
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
rga_api version 1.10.1_[9]
Output #0, mp4, to '/home/twdz/test.mp4':
  Metadata:
    encoder         : Lavf60.3.100
  Stream #0:0: Video: h264 (High) (avc1 / 0x31637661), drm_prime(progressive), 1280x720 [SAR 1:1 DAR 16:9], q=2-31, 4000 kb/s, 25 fps, 12800 tbn
    Metadata:
      encoder         : Lavc60.3.100 h264_rkmpp
frame=  115 fps=1.9 q=-0.0 Lsize=       1kB time=00:00:04.56 bitrate=   1.7kbits/s speed=0.075x    
video:0kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 2312.820557%
Exiting normally, received signal 2.   我在RK3588上编译的，跑这个正常，同样编译方式在RK3576上结果就是这么慢了。

nyanmisaka, 2025-05-07:
I have no such issue. Contact your board vendor.

```
$ uname -a
Linux armsom-cm5-io 6.1.99-vendor-rk35xx #1 SMP Sun Mar  9 02:59:50 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux

$ lscpu
Architecture:             aarch64
  CPU op-mode(s):         32-bit, 64-bit
  Byte Order:             Little Endian
CPU(s):                   8
  On-line CPU(s) list:    0-7
Vendor ID:                ARM
  Model name:             Cortex-A53
    Model:                4
    Thread(s) per core:   1
    Core(s) per socket:   4
    Socket(s):            1
    Stepping:             r0p4
    CPU(s) scaling MHz:   100%
    CPU max MHz:          2016.0000
    CPU min MHz:          408.0000
    BogoMIPS:             48.00
    Flags:                fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
  Model name:             Cortex-A72
    Model:                0
    Thread(s) per core:   1
    Core(s) per socket:   4
    Socket(s):            1
    Stepping:             r1p0
    CPU(s) scaling MHz:   100%
    CPU max MHz:          2208.0000
    CPU min MHz:          408.0000
    BogoMIPS:             48.00
    Flags:                fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
Vulnerabilities:
  Gather data sampling:   Not affected
  Itlb multihit:          Not affected
  L1tf:                   Not affected
  Mds:                    Not affected
  Meltdown:               Not affected
  Mmio stale data:        Not affected
  Reg file data sampling: Not affected
  Retbleed:               Not affected
  Spec rstack overflow:   Not affected
  Spec store bypass:      Not affected
  Spectre v1:             Mitigation; __user pointer sanitization
  Spectre v2:             Vulnerable: Unprivileged eBPF enabled
  Srbds:                  Not affected
  Tsx async abort:        Not affected

$ ./ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -f lavfi -i testsrc2=s=1920x1080,format=nv12 -vf hwupload,scale_rkrga=w=1280:h=720:format=nv12 -c:v h264_rkmpp -b:v 4M -maxrate 4M -vframes 1000 -y /tmp/test.mp4
ffmpeg version 7.1.1-Jellyfin Copyright (c) 2000-2025 the FFmpeg developers
  built with gcc 14.2.0 (crosstool-NG 1.27.0.18_7458341)
  configuration: --prefix=/ffbuild/prefix --pkg-config=pkg-config --pkg-config-flags=--static --cross-prefix=aarch64-ffbuild-linux-gnu- --arch=aarch64 --cpu=armv8-a --target-os=linux --extra-version=Jellyfin --extra-cflags= --extra-cxxflags= --extra-ldflags= --extra-ldexeflags=-pie --extra-libs=-ldl --enable-gpl --enable-version3 --disable-ffplay --disable-debug --disable-doc --disable-sdl2 --disable-libxcb --disable-xlib --enable-lto=auto --enable-iconv --enable-zlib --enable-libfreetype --enable-libfribidi --enable-gmp --enable-libxml2 --enable-openssl --enable-lzma --enable-fontconfig --enable-libharfbuzz --enable-libvorbis --enable-opencl --enable-chromaprint --enable-libdav1d --enable-libfdk-aac --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc --enable-libass --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvpx --enable-libwebp --enable-libopenmpt --enable-rkmpp --enable-rkrga --enable-libsrt --enable-libsvtav1 --enable-libdrm --enable-vulkan --enable-libshaderc --enable-libplacebo --enable-libx264 --enable-libx265 --enable-libzimg --enable-libzvbi
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
Input #0, lavfi, from 'testsrc2=s=1920x1080,format=nv12':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, nv12, 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 25 tbn
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
rga_api version 1.10.1_[9]
Output #0, mp4, to '/tmp/test.mp4':
  Metadata:
    encoder         : Lavf61.7.100
  Stream #0:0: Video: h264 (High) (avc1 / 0x31637661), drm_prime(tv, progressive), 1280x720 [SAR 1:1 DAR 16:9], q=2-31, 4000 kb/s, 25 fps, 12800 tbn
      Metadata:
        encoder         : Lavc61.19.101 h264_rkmpp
[out#0/mp4 @ 0xaaaadfc40930] video:19218KiB audio:0KiB subtitle:0KiB other streams:0KiB global headers:0KiB muxing overhead: 0.026474%
frame= 1000 fps=243 q=-0.0 Lsize=   19223KiB time=00:00:39.96 bitrate=3940.9kbits/s speed=9.71x
```

SCUT-chenjia, 2025-05-07:
> Contact your board vendor.

好的，我确认手上的RK3576硬件底层是否有问题，导致硬件转码不符合预期。