nyanmisaka/ffmpeg-rockchip #216: mjpeg_rkmpp硬件编码缓存延时问题

William-Li01, 2025-08-16:
在使用mjpeg_rkmpp硬件编码NV12数据时，前两帧数据通过avcodec_send_frame函数发送给编码器，使用avcodec_receive_packet函数获取编码后的数据会报错返回-11，然后第三帧数据通过avcodec_send_frame函数发送给编码器，再调用avcodec_receive_packet获取可以获取到数据，但是获取的是第一帧的数据，这样会导致延迟两帧时间，如何配置可以立即获取编码后的数据啊？
我测试RKMPP的例程mpi_enc_test，没有出现编码缓存问题，不知道是哪里的配置问题，下面是我的测试代码。
`#include <iostream>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <sys/time.h>

extern "C"
{
#include <libavcodec/avcodec.h>
#include <libavfilter/avfilter.h>
#include <libavfilter/buffersink.h>
#include <libavfilter/buffersrc.h>
#include <libavformat/avformat.h>
#include <libavutil/hwcontext.h>
#include <libavutil/opt.h>
#include <libavutil/pixfmt.h>

#include <libavdevice/avdevice.h> // FFmpeg 设备库
#include <libswscale/swscale.h>   // FFmpeg 图像缩放库
#include <libavutil/mem.h>        // FFmpeg 内存管理工具
#include <libavutil/imgutils.h>   // FFmpeg 图像工具
#include "libavutil/time.h"
}

#include <thread>    // 引入线程库
#include <chrono>    // 引入时间相关的库

#define ENCODER_NAME "mjpeg_rkmpp"

static AVFormatContext *fmt_ctx;
static AVCodecContext *decodec_ctx;
static AVCodecContext *encodec_ctx;
static int video_stream_index = -1;
static AVStream *video_stream;
static const AVCodec *decodec;
static const AVCodec *encodec;

int64_t get_cur_us() {
    struct timeval tv;
    gettimeofday(&tv,NULL);
    return tv.tv_sec * 1000000 + tv.tv_usec;
}

void rescale(AVFormatContext *in_format_ctx, AVFormatContext *out_format_ctx, AVPacket *packet)
{
    AVStream *in_stream = in_format_ctx->streams[packet->stream_index];
    AVStream *out_stream = out_format_ctx->streams[packet->stream_index];

    if (in_stream->time_base.num == out_stream->time_base.num && in_stream->time_base.den == out_stream->time_base.den)
    {
        packet->pos = -1;
        return;
    }

    packet->pts = av_rescale_q(packet->pts, in_stream->time_base, out_stream->time_base);
    packet->dts = av_rescale_q(packet->dts, in_stream->time_base, out_stream->time_base);
    packet->duration = av_rescale_q(packet->duration, in_stream->time_base, out_stream->time_base);
    packet->pos = -1;
}

int main(int argc, char **argv)
{
    avdevice_register_all();                                               // 注册所有设备（如摄像头）
    avformat_network_init();

    int ret;
    AVPacket *packet;
    AVFrame *frame;
[... 278 lines cut ...]
decode running success! cost 11353 us
frame->pts: 23901604446
Send frame 23901604446
encode running success! cost 4511 us
Write packet 23901452445 (size=230571)
avcodec_receive_packet return -11
***ret: -11
times: 4, avcodec_receive_frame: 0
decode running success! cost 5623 us
frame->pts: 23901636446
Send frame 23901636446
encode running success! cost 1723 us
Write packet 23901568421 (size=102466)
avcodec_receive_packet return -11
***ret: -11
times: 5, avcodec_receive_frame: 0
decode running success! cost 5726 us
frame->pts: 23901668449
Send frame 23901668449
encode running success! cost 1716 us
Write packet 23901604446 (size=114413)
avcodec_receive_packet return -11
***ret: -11
`
可以看到第三帧才打印encode running success!信息，成功编码。

nyanmisaka, 2025-08-16:
The default is async encoding. To enable sync encoding, set the low-delay flag `-flags +low_delay` for the encoder.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders

William-Li01, 2025-08-16:
> 默认值为异步编码。若要启用同步编码，请为编码器设置低延迟标志。`-flags +low_delay`
> 
> https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders

好的，感谢，我试试

William-Li01, 2025-08-16:
`encodec_ctx->width = 1920;
encodec_ctx->height = 1080;
encodec_ctx->pix_fmt = AV_PIX_FMT_NV12;
encodec_ctx->time_base = video_stream->time_base;
encodec_ctx->framerate = video_stream->r_frame_rate;

av_opt_set(encodec_ctx->priv_data, "low_delay", "1", 0);

if (avcodec_open2(encodec_ctx, encodec, NULL) < 0)
{
    std::cout << "avcodec_open2 failed" << std::endl;
    return -1;
}`
您好，我试着在打开编码器时，添加av_opt_set(encodec_ctx->priv_data, "low_delay", "1", 0);命令，测试依然有两帧缓存

> 默认值为异步编码。若要启用同步编码，请为编码器设置低延迟标志。`-flags +low_delay`
> 
> https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders