nyanmisaka/ffmpeg-rockchip #222: rk3576 mpp error，video dropped frame

fys18648950197, 2025-08-27:
mpp[3966]: mpp_info: mpp version: 331ee68f autor: nyanmisaka 2024-12-11 fix[mpp]: Fix input_task_count for a sync enc.
mpp[3966]: mpp: unable to create enc vp8 for rk3576 unsupported

1. GST_DEBUG=fps*:7 gst-launch-1.0 uridecodebin uri=$URI ! fpsdisplaysink video-sink="rkximagesink" text-overlay=false signal-fps-measurements=true
2. gst-play-1.0 test.mp4 --flags=3 videosink="rkximagesink"

When I play videos using gst, the frame rate drops severely, going from 60 frames per second to 22.I don't know how to play videos offline without frame drops.
Can you  give me some guidance?

Debian GNV/Linux 12 (bookworm)
Xfce:4.18
GTK:3.24.38
core: 6.1.99

gstreamer version:1.22.9

fys18648950197, 2025-08-27:
At first, there would be frame drops. Later, the frame rate gradually stabilized.

nyanmisaka, 2025-08-27:
First, the mpp version is out of date.
Second, this is ffmpeg-rockchip, so I can't answer questions about gstreamer.

closed 2025-08-27 by nyanmisaka