nyanmisaka/ffmpeg-rockchip #224: High-bitrate video can't be scale with scale_rkrga?

cnjswwxbtcc, 2025-09-02:
### My hw filter is like:
scale_rkrga=w=480:h=360:force_original_aspect_ratio=disable:format=rgb24,hwdownload,format=rgb24

### The format of Video is like:
```bash
# ./ffprobe -v quiet -print_format json -show_format ./SNOW_CAPPED_MOUNTAINS.mp4
mpp[10372]: mpp_platform: client 16 driver is not ready!
mpp[10372]: kmpp_meta: kmpp_meta_init kmpp meta get objdef failed
{
mpp[10372]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[10372]: mpp_info: mpp version: unknown mpp version for missing VCS info
    "format": {
        "filename": "./SNOW_CAPPED_MOUNTAINS.mp4",
        "nb_streams": 2,
        "nb_programs": 0,
        "format_name": "mov,mp4,m4a,3gp,3g2,mj2",
        "format_long_name": "QuickTime / MOV",
        "start_time": "0.000000",
        "duration": "7.436944",
        "size": "30156476",
        "bit_rate": "32439642",
        "probe_score": 100,
        "tags": {
            "major_brand": "isom",
            "minor_version": "512",
            "compatible_brands": "isomiso2mp41",
            "creation_time": "2025-05-04T01:30:57.000000Z",
            "RecordSystemTime": "275816223355,275823625983",
            "com.android.manufacturer": "HUAWEI",
            "com.android.marketing_name": "HUAWEI Pura 70",
            "com.android.model": "ADY-AL10",
            "com.android.version": "12",
            "com.huawei.hdr format": "HDR Vivid, Version 1.0, HLG compatible"
        }
    }
}
```

nyanmisaka, 2025-09-02:
Video filters have nothing to do with video bitrate. The input and output resolutions supported by RGA hardware can be found in the RGA documentation.

https://github.com/airockchip/librga/blob/main/docs/Rockchip_Developer_Guide_RGA_EN.md#design-index

cnjswwxbtcc, 2025-09-03:
> There is simply no relationship between them. Video filters process raw video frames, which are YUV or RGA data, and their size is determined by the video resolution rather than the bit rate.

好的, 我可能得把方向放在pix_fmt;
还有个疑问, 我现在构造滤镜链的时候, 组成的字符串:
`video_size=1920x1080:pix_fmt=179:time_base=1/1:pixel_aspect=1/1`
pix_fmt是填枚举值, 还是说要写pix_fmt的name的字符串?